MarlinFirmware/Marlin #21109: [BUG] Auto Fan Glitch Exists When Rotary Encoder Button Pushed

borland1, 2021-02-16:
### Bug Description
When the `Configuration_adv` file is setup for Auto Fan ( `E0_AUTO_FAN` ), when fan is operational due to extruder temperature being greater than `EXTRUDER_AUTO_FAN_TEMPERATURE` setting, there is a glitch in fan operation when pressing the rotary encoder knob button.

This button press causes the fan to stop running momentarily, but the extruder fan will resume running when within a second or two, provided the LCD menu selections are not being subsequently selected with press of the rotary encoder button.

It is normal for PWM fan operation to be glitch free.

This glitch was reported earlier on unrelated issue #19808, but confirmed to still exist with latest bugfix-2.0x.

### Configuration Files
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/5991890/ConfigFiles.zip)

thinkyhead, 2021-02-16:
Please enable `PINS_DEBUGGING` and post the output of `M43` as a new comment so we can see if there's some kind of pin coincidence. You can format it as plain text by putting three backticks before and after the text ` ``` ` on lines by themselves.

borland1, 2021-02-16:
Here's the M43 response..

```
PIN:   0   Port: E0        RXD0                                   protected 
PIN:   1   Port: E1        TXD0                                   protected 
PIN:   2   Port: E4        <unused/unknown>                       Input  = 1                           TIMER3B   PWM:     0    WGM: 1    COM3B: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0  
PIN:   3   Port: E5        X_MIN_PIN                              protected 
.                          X_STOP_PIN                             protected 
PIN:   4   Port: G5        SERVO3_PIN                             Input  = 0    TIMER0B   PWM:   128    WGM: 3    COM0B: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   compare interrupt enabled   overflow interrupt enabled  
PIN:   5   Port: E3        SERVO2_PIN                             Input  = 0    TIMER3A   PWM:     0    WGM: 1    COM3A: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0  
PIN:   6   Port: H3        SERVO1_PIN                             Input  = 0    TIMER4A   PWM:     0    WGM: 1    COM4A: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0  
PIN:   7   Port: H4        MOSFET_D_PIN                           Input  = 0    TIMER4B   PWM:     0    WGM: 1    COM4B: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0  
PIN:   8   Port: H5        HEATER_BED_PIN                         protected 
.                          RAMPS_D8_PIN                           protected 
PIN:   9   Port: H6        E0_AUTO_FAN_PIN                        protected 
.                          RAMPS_D9_PIN                           protected 
PIN:  10   Port: B4        HEATER_0_PIN                           protected 
.                          RAMPS_D10_PIN                          protected 
PIN:  11   Port: B5        SERVO0_PIN                             Input  = 0    TIMER1A   PWM:  2000    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled  
PIN:  12   Port: B6        <unused/unknown>                       Input  = 0                           TIMER1B   PWM:     0    WGM: 4    COM1B: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode  
PIN:  13   Port: B7        LED_PIN                                Input  = 0    TIMER0A   PWM:     0    WGM: 3    COM0A: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled  
 .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  14   Port: J1        Y_MIN_PIN                              protected 
.                          Y_STOP_PIN                             protected 
PIN:  15   Port: J0        Z_MIN_PIN                              protected 
.                          Z_STOP_PIN                             protected 
PIN:  16   Port: H1        DOGLCD_CS                              Output = 1  
PIN:  17   Port: H0        DOGLCD_MOSI                            Output = 0  
PIN:  18   Port: D3        TXD1                                   Input  = 1  
PIN:  19   Port: D2        RXD1                                   Input  = 1  
PIN:  20   Port: D1        <unused/unknown>                       Input  = 0                         
PIN:  21   Port: D0        <unused/unknown>                       Input  = 0                         
PIN:  22   Port: A0        <unused/unknown>                       Input  = 0                         
PIN:  23   Port: A1        DOGLCD_SCK                             Output = 0  
PIN:  24   Port: A2        E0_ENABLE_PIN                          protected 
PIN:  25   Port: A3        DOGLCD_A0                              Output = 1  
PIN:  26   Port: A4        E0_STEP_PIN                            protected 
PIN:  27   Port: A5        LCD_PINS_RS                            Output = 1  
PIN:  28   Port: A6        E0_DIR_PIN                             protected 
PIN:  29   Port: A7        LCD_PINS_D7                            Input  = 0  
PIN:  30   Port: C7        E1_ENABLE_PIN                          Output = 1  
PIN:  31   Port: C6        BTN_EN1                                Input  = 1  
PIN:  32   Port: C5        <unused/unknown>                       Input  = 0                         
PIN:  33   Port: C4        BTN_EN2                                Input  = 1  
PIN:  34   Port: C3        E1_DIR_PIN                             Output = 0  
PIN:  35   Port: C2        BTN_ENC                                Input  = 1  
PIN:  36   Port: C1        E1_STEP_PIN                            Input  = 0  
PIN:  37   Port: C0        BEEPER_PIN                             Output = 0  
PIN:  38   Port: D7        X_ENABLE_PIN                           protected 
PIN:  39   Port: G2        <unused/unknown>                       Input  = 0                         
PIN:  40   Port: G1        E_MUX0_PIN                             Output = 1  
.                          X_SERIAL_TX_PIN                        Output = 1  
PIN:  41   Port: G0        KILL_PIN                               Input  = 1  
PIN:  42   Port: L7        E_MUX1_PIN                             Input  = 0  
.                          Z_SERIAL_TX_PIN                        Input  = 0  
PIN:  43   Port: L6        <unused/unknown>                       Input  = 0                         
[... cut ...]
PIN:  64   Port: K2 (A10)  Y_SERIAL_RX_PIN                        Input  = 1  
PIN:  65   Port: K3 (A11)  Z_SERIAL_RX_PIN                        Input  = 0  
PIN:  66   Port: K4 (A12)  MAX6675_SS_PIN                         Input  = 0  
.                          E0_SERIAL_RX_PIN                       Input  = 0  
PIN:  67   Port: K5 (A13)  TEMP_0_PIN                             protected 
PIN:  68   Port: K6 (A14)  TEMP_BED_PIN                           protected 
PIN:  69   Port: K7 (A15)  TEMP_1_PIN                             Analog in =  1023
```

borland1, 2022-01-28:
This is an old problem report, however AUTO_FAN glitch still exists.

Has the above submitted M43 data been reviewed for clues as to the glitch source and possible fix?

ellensp, 2022-01-28:
The PWM system has been re-written since 2.0.8  This needs re tested with on current bugfix.

borland1, 2022-01-31:
Tested with latest bugfix-2.0.x.  Same AUTO_FAN glitch observed with 2.0.8.

[bugfix-2.0.x-configs.zip](https://github.com/MarlinFirmware/Marlin/files/7967572/bugfix-2.0.x-configs.zip)

I think this is an interrupt problem, but don't know for sure.  In my `Configuration.h` file, I have these lines added under `EndStops Settings` in order to remap Z-max to allow using it for `SERIAL_PORT_2` ( Serial1 at Z-MIN(D18(TX)) & Z-MAX(D19(RX)) on MKS_GEN_L_V2 board) ....  could be this has nothing to do with the glitch problem.
```
// ****** Remap Limit Z-MIN (Pin 18) & disable Z-MAX Limit Switch Connectors on MKS_GEN_L_V2 Board,  ***************************************************
// ****** this to override pins definition file ( pins_RAMPS.h ) to free Serial1 (Arduino Mega pins 19(RX), 18(TX)) for use as secondary serial as WiFi ************************
#define Z_MIN_PIN 15  // ******** remapping Z_MIN_PIN function to otherwise unused Y_MAX_PIN (Pin 15) Connector  *****************************************
#define Z_MAX_PIN -1  // ******** disable Z_MAX_PIN function (Pin 19) used with Serial1(RX) **************************************************************
```

borland1, 2022-01-31:
If I disable the SPEAKER in `Configuration.h`, the glitch is gone.

```
// SPEAKER/BUZZER
//#define SPEAKER
```

descipher, 2022-01-31:
When SPEAKER is defined buzzer.cpp code will use timer 2 generate the tone pulses, this will clobber any PWM signal if it's on the same channel. Without the SPEAKER define it will generate a click sound without the timer.

closed 2022-04-24 by github-actions[bot]