MarlinFirmware/Marlin #21117: [BUG] wrong fan pins on trigorilla_pro

leonhma, 2021-02-17:
### Bug Description

When flashing marlin on trigorilla_pro the fans don't work.

### Configuration Files

in src/pins/stm32f1/TRIGORILLA_PRO.h:
#define CONTROLLER_FAN_PIN                  PD6   // FAN
#define E1_AUTO_FAN_PIN                    PG13  // FAN
#define FAN_PIN                            PG14  // FAN

controller fan works but the other two don't

### Steps to Reproduce

flash marlin on anycubic i3 mega

**Expected behavior:**

the cooling fan should react to m106

**Actual behavior:**

the hotend fan stays on all the time (which is not that bad if i think about it) and the part cooling fan is off. this can't be changed through m106 or heating the hotend

#### Additional Information

i have tried to flash the stock (chitu) firmware, which i pulled off the board before ...aaaaand it worked fine

so i guess the problem is wrong pins... again. #21100 

[ai3m_stock.zip](https://github.com/MarlinFirmware/Marlin/files/5995121/ai3m_stock.zip)

ellensp, 2021-02-17:
https://3dtoday.ru/upload/posts/main/2LI2fzZsxfpcdeQ3dZ7Py77CStU6h9MBCj1ANTXM.jpeg

PG13 // FAN
PG14 // FAN

Are correct as far as we know.

leonhma, 2021-02-17:
I have Just measured the resistance between pg13, pg14 and the mosfets. They seem to be the right Pins... Could the mosfets be broken, even though they don't Show any burning, etc. ?

leonhma, 2021-02-17:
Okay im closing this and looking for Support on dc. The Pins are indeed correct and the e0 auto Fan wasn't working because i defined E1_AUTO_FAN. The Part cooling Fan still isn't controllable but i think it rather is a technical/wiring fault.

closed 2021-02-17 by leonhma

leonhma, 2021-02-18:
The problem turned out to be a weird fan, that doesn't like pwm. Had to use software pwm for a lower frequency.