MarlinFirmware/Marlin #21118: [FR] Z Safe Homing – option to end Z homing with XY homing as well

LemmeDasker, 2021-02-17:
### Description

Most of the time I use the Slicers GUI to control my printer. And quite often I use the ‘Home all’ button on the GUI. Basically, I use it to ‘park’ the nozzle even though it is not intended for that purpose. But it is very easy and convenient way to do it.

That has changed with Z Safe Homing. With Z Safe Homing enabled, the probe ends up in the middle of the plate, and just stands there after homing. And that is the expected behavior. However – the convenience by just pressing the ‘Home all’ button and get the nozzle off the plate are lost.

I propose a third option in Configuration.h to get the ‘old’ feeling back:

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT X_CENTER  // X point for Z homing
  #define Z_SAFE_HOMING_Y_POINT Y_CENTER  // Y point for Z homing
  #define Z_SAFE_HOMING_END_WITH_XY_HOMING  // Homing XY after Z homing
#endif

### Feature Workflow

1. Enable Z_SAFE_HOMING in Configuration.h
2. Enter G28 or press ‘Home all’ in the Slicer GUI
3. X and Y are homing (part of normal procedure)
4. Z are travelling to a safe place on the plate to do the homing
5. X and Y are homing (new feature - moving the nozzle away from the plate) 

#### Additional Information

[G28.cpp.txt](https://github.com/MarlinFirmware/Marlin/files/5995911/G28.cpp.txt)

I have made this code change two places in G28.cpp - see also attached file. The code works as intended on my printer.

#if ENABLED(Z_SAFE_HOMING_END_WITH_XY_HOMING)	// end with XY homing efter Z homing
  if (home_all) {
    homeaxis(X_AXIS);
    homeaxis(Y_AXIS);
  }
#endif // END_WITH_XY_HOMING

xeddog, 2021-02-17:
What about just adding a "Park Nozzle" to the motion menu.  Then you could define where exactly you want to park it via the Nozzle Park feature in configuration.h
You could also define a custom user menu item for it.  It's just a G27 command.

ManuelMcLure, 2021-02-17:
`G27` requires `NOZZLE_PARK_FEATURE` to be enabled. On boards with small amounts of flash it may be impossible to enable that option.

However, the custom menu item is an interesting option:
```
  #define USER_DESC_5 "Home XY"
  #define USER_GCODE_5 "G28 X Y"
```

LemmeDasker, 2021-02-17:
If the custom menu requires an external LCD display, then it will not work as I intend.
I want it to work with different slicers GUI like Cura, Simplify3D etc. And then the change must be part of the G28 rutine itself.
Unless I am wrong in my assumption.

qwewer0, 2021-02-17:
Pronterface allows to add custom buttons, such as `G28 XY`, if you only want X and Y to home.

LemmeDasker, 2021-02-17:
I am sure that there are host software out there where buttons can be custom configured.

However - it is my opinion that this feature should be host agnostic. Whatever host software or LCD display you use, or if you just enter G28 on the command line, the option should be avaliable for you.

That is why I think the change has to be part of the G28 rutine itself. I can not see how it otherwise would be avaliable to everyone.

bulcke, 2022-01-02:
Makes sense to me.
I want this to prevent oozing on the hotbed. Usually Z_SAFE_HOMING is in the center of the bed, so your nozzle would be near.
Suggested change would - in my case - park the nozzle off the bed.

bulcke, 2022-01-03:
Hi @LemmeDasker I tested your code (using 2.9.3). It obviously works.

But there is a small issue when using the Z Probe Wizard: printer homes first as expected: first X and Y, then goes to the center of bed for Z homing, then it homes X and Y again.
But then the wizard probably wants to home again. X carriage goes left, but it was already homed and therefore crashes into the X stop switch.

So if this feature ever makes it, some more checks are needed.

LemmeDasker, 2022-01-03:
Hi Bulcke
I implemented the feature myself, and it did work. But I also experienced some unexpected behavior at times. So I decided to abandon the Z_SAFE_HOMING function instead. The reason is that I am not a real programmer, just a hobbyist. And I have not the skills to know what is fully going on in the full code, and to make the required checks as you ask for.

I realised that if this feature is requested by others, some Pro really needs to implement it correctly.

jantman, 2022-04-17:
I have the same feature request, but a very different use case. I recently added a BLtouch probe to my printer, and enabled Z_SAFE_HOMING because of it. That works great, but all of my pre-sliced GCode (sliced using Cura) for this printer is now broken... it homes XY, then moves to the middle of the bed, then homes Z. and... then starts a print, starting from the middle of the bed instead of the origin corner. I've been able to fix this by adding to the GCode, but it seems like the intent of G28 is to actually home the nozzle, not leave it in the middle of the bed...

fjulian79, 2024-12-19:
Found this while searching for a solution to exactly this issue. The point is: Z Save Homing does find the home positions, but does not move the print head to the home position as usual. From my point of view this is a bug.

ManuelMcLure, 2024-12-19:
Not a bug. Homing does not mean "move to the home position". It means "determine where the home position is". If you want to move somewhere after homing you can add a "G0 X0 Y0" after the "G28".

If you are seeing a print starting from the middle of the bed after homing with Z_SAFE_HOMING, it probably means that you have the printer in relative mode which is _not_ what slicer g-code expects. Make sure to add a "G90" to your startup code so the printer is in absolute position mode.