MarlinFirmware/Marlin #21119: [BUG] Multiple bugs in the queue code (meta-bug)

X-Ryl669, 2021-02-17:
### Bug Description

While trying to solve #21010, I've focused on the queue code and I've found many bugs for borderline cases.
Here are the descriptions of those and the proposed solution.

#### 1. Command queue buffer overflow
Here's the troublesome code in queue.cpp (I've skipped part we don't care about to understand the bugs):
```cpp
void GCodeQueue::get_serial_commands() {
[...]
  while (length < BUFSIZE && serial_data_available()) {
    LOOP_L_N(p, NUM_SERIAL) {
[...]
      const int c = read_serial(p);
      if (c < 0) continue;

[...]
        const char serial_char = TERN(MEATPACK, c_res[char_index], c);

        if (ISEOL(serial_char)) {
[...]
_enqueue(serial_line_buffer[p], true); // This commit the message and does length++
      }
    }
  } 
}
```
In the bug report, the user is using MultiSerial, so NUM_SERIAL > 1. 
Let's consider BUFSIZE = 32 and length = 31 case here when entering the loop. 

If both serial port have an end of line character, then LOOP_L_N will loop twice and enqueue 2 serial line buffer, increasing length twice, meaning that `length > BUF_SIZE` => **overflow**

##### Proposed solution
The loop checking serial_data_available()  should happen per serial port. Hopefully, with new serial code, it's very easy to do. Checking the length variable can be done per serial port iteration instead of globally.

#### 2. Out-of-bound access to the serial buffers
In this case, it's a problem of the enqueuing code sending `ok` message. As far as I've understood the code, the queue code is like 2 (virtual) threads, one is reading the serial commands from the serial port (or sd card) and the other is advancing the processing of each command. The former is increasing the queue's write pointer, and only the latter is increasing the queue's read pointer. Only the latter should send `ok` when it's done processing command.

Yet, we observed erratic behavior with the OP of the initial bug that I'm listing here:
```
2021-02-16 11:50:15,839 - Send: N1063 G0 F7800 X110.116 Y114.641*116
2021-02-16 11:50:15,850 - Recv: ok N1038 P0 B31
2021-02-16 11:50:15,854 - Send: N1064 G1 F1200 X111.199 Y115.725 E74.59496*5
2021-02-16 11:50:15,886 - Recv: ok N1039 P0 B31
2021-02-16 11:50:15,890 - Send: N1065 G0 F7800 X111.379 Y115.339*115
2021-02-16 11:50:15,902 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 1039
2021-02-16 11:50:15,903 - Recv: Resend: 1040
HERE =>>>>> 2021-02-16 11:50:15,908 - Recv: ok N1008 P0 B32
2021-02-16 11:50:15,911 - Send: N1040 G3 X110.318 Y114.069 I4.986 J3.711 E73.26023 F1200*4
2021-02-16 11:50:16,424 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 1039
2021-02-16 11:50:16,427 - Recv: Resend: 1040
2021-02-16 11:50:16,430 - Recv: ok N1008 P0 B32
```
In his case, the printer is lagging behind the host processing command. As some point in time, it's receiving too much command for its buffer and skipping some. It sends an error "Line number is not last line number +1", but then send a `ok N1008 P0 B32`.
This is completely impossible message, since:
1. `N1008` was processed minutes earlier
2. `B32` means that length == 0 (since the OP has BUF_SIZE=32)

They are 2 possible path that send ok:
1. At end of `queue::advance` (which calls `gcode::process_parsed_command` which calls `queue::ok_to_send`)
2. In `queue::flush_and_request_resend` that's called while receiving data from the serial port.

In the first case, the length of the queue is checked so it's not possible to make such error
In the second case, the length of the queue isn't checked, and this code access a memory location that's invalid.
In the OP's example serial dump, that's what happened: length is 0 (as confirmed by B32 in the answer), and the line number is extracted out of thin-air (out of bound access to a deprecated buffer) in ok_to_send:
```
void GCodeQueue::ok_to_send() {
[...]
    char* p = command_buffer[index_r]; // <<<<< Access to an invalid read pointer since length = 0 here
    if (*p == 'N') {
      SERIAL_CHAR(' ', *p++);
      while (NUMERIC_SIGNED(*p))
        SERIAL_CHAR(*p++); // <<<<< could have emitted Romeo and Juliet as well
    }
    SERIAL_ECHOPAIR_P(SP_P_STR, planner.moves_free(),
                      SP_B_STR, BUFSIZE - length);
```
##### Proposed solution
I don't think it's required to check for length in ok_to_send, instead, I think the logic in `flush_and_request` should be modified not to send a ok message like this:
```
void GCodeQueue::flush_and_request_resend() {
  const serial_index_t serial_ind = command_port();
  #if HAS_MULTI_SERIAL
    if (serial_ind < 0) return;                   // Never mind. Command came from SD or Flash Drive
    PORT_REDIRECT(SERIAL_PORTMASK(serial_ind));   // Reply to the serial port that sent the command
  #endif
  SERIAL_FLUSH();
  SERIAL_ECHOPGM(STR_RESEND);
  SERIAL_ECHOLN(last_N[serial_ind] + 1);
- ok_to_send(); // WRONG: This is wrong since it should happen in gcode's advance thread while it's being called in the get_serial_command
}
```
@thinkyhead  You included this line in bae1cb859c1 do you remember why ?

#### 3. get_serial_command is not consuming the whole serial buffer 

Well, the title says it all. `get_serial_command` stops after each line from the host in case of multiserial. Since the code is doing more or less:
```
idle_loop():
   queue.get_serial_command()  >>> enqueue a command
   queue.advance()  >>>  execute the last command
```
this prevent all the circular buffer to work as intended.

##### Proposed solution
Rewrite the loop condition so it's transparent for multiserial as well. Right now, if one serial port triggers an exit condition, then it prevent all other serial port from making progress. 

  
### Configuration Files

Multiple case, see bug description above

GhostlyCrowd, 2021-03-14:
I just fell victim to this issue today, Is there anything we can do about t yet? of should I disable my other serial? 

`2021-03-14 10:22:41,664 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 1021672, current line = 1021691
| Last lines in terminal:
| Recv: ok N1021654 P0 B19
| Send: N1021682 G1 X171.294 Y100.438 E0.0316*94
| Send: N1021683 G1 X171.370 Y100.474 E0.0326*95
| Recv: ok N1021655 P1 B19
| Send: N1021684 G1 X171.379 Y100.479 E0.0327*93
| Recv: ok N1021656 P1 B20
| Send: N1021685 G1 X171.451 Y100.548 E0.0336*82
| Recv: ok N1021657 P2 B21
| Send: N1021686 G2 X171.536 Y100.630 I1.846 J-1.828 E0.03460*79
| Recv: X:168.49 Y:99.96 Z:25.70 E:0.00 Count X:12849 Y:7928 Z:10280
| Recv: ok N1021658 P2 B22
| Send: N1021687 G1 X175.079 Y101.452 F3600*79
| Recv: ok N1021659 P1 B22
| Send: N1021688 G92 E0.0000*109
| Recv: ok N1021660 P0 B23
| Send: N1021689 G3 X175.154 Y101.532 I-1.139 J1.142 E0.00090 F900*22
| Recv: ok N1021661 P0 B16
| Send: N1021690 G1 X175.225 Y101.607 E0.0020*90
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 1021671
| Recv: Resend: 1021672`

Goes on like this untill Octo emergency stops the printer due  `| Recv: Error:Line Number is not Last Line Number+1, Last Line: 1021671
| Recv: Resend: 1021672` repeating

thinkyhead, 2021-03-14:
The code referenced in the original post is no longer present. You'll have to take another look to see which points above still apply.

closed 2021-03-14 by thinkyhead

Snargaloo, 2021-03-14:
> 
> 
> I just fell victim to this issue today, Is there anything we can do about t yet? of should I disable my other serial?
> 
> ```
> 2021-03-14 10:22:41,664 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 1021672, current line = 1021691
> | Last lines in terminal:
> | Recv: ok N1021654 P0 B19
> | Send: N1021682 G1 X171.294 Y100.438 E0.0316*94
> | Send: N1021683 G1 X171.370 Y100.474 E0.0326*95
> | Recv: ok N1021655 P1 B19
> | Send: N1021684 G1 X171.379 Y100.479 E0.0327*93
> | Recv: ok N1021656 P1 B20
> | Send: N1021685 G1 X171.451 Y100.548 E0.0336*82
> | Recv: ok N1021657 P2 B21
> | Send: N1021686 G2 X171.536 Y100.630 I1.846 J-1.828 E0.03460*79
> | Recv: X:168.49 Y:99.96 Z:25.70 E:0.00 Count X:12849 Y:7928 Z:10280
> | Recv: ok N1021658 P2 B22
> | Send: N1021687 G1 X175.079 Y101.452 F3600*79
> | Recv: ok N1021659 P1 B22
> | Send: N1021688 G92 E0.0000*109
> | Recv: ok N1021660 P0 B23
> | Send: N1021689 G3 X175.154 Y101.532 I-1.139 J1.142 E0.00090 F900*22
> | Recv: ok N1021661 P0 B16
> | Send: N1021690 G1 X175.225 Y101.607 E0.0020*90
> | Recv: Error:Line Number is not Last Line Number+1, Last Line: 1021671
> | Recv: Resend: 1021672
> ```
> 
> Goes on like this untill Octo emergency stops the printer due `| Recv: Error:Line Number is not Last Line Number+1, Last Line: 1021671 | Recv: Resend: 1021672` repeating
> 
> This is over 12 hours into a print, then just out of the blue.

Your printer was lagging 28 commands behind the host.  Try to post your configuration for inspection.

GhostlyCrowd, 2021-03-14:
> Your printer was lagging 28 commands behind the host. Try to post your configuration for inspection.

Sure, but this has only started since the multi serial issues.  ive done a dozen of this same gcode over the past year fine.  `Recv: Error:Line Number is not Last Line Number+1` has never appears until Meatpack and multi serial issues came into the mix

Note in the attached configs i have my second serial port disabled now until i can rule out multi serial issues.  It was on for mt TFT at the time of the failure. 
Configs attached 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6137256/Marlin.zip)

X-Ryl669, 2021-03-14:
There are currently changes related to Meatpack and multiserial being merged in. Please let few days until all the work has settled down and retry. If it still fail, please open another issue.

GhostlyCrowd, 2021-03-14:
> There are currently changes related to Meatpack and multiserial being merged in. Please let few days until all the work has settled down and retry. If it still fail, please open another issue.

Will do!

rhapsodyv, 2021-03-14:
> Your printer was lagging 28 commands behind the host. Try to post your configuration for inspection.

Exactly!! @GhostlyCrowd do you have the full log? So we can see where the lag started? I should never happen.

GhostlyCrowd, 2021-03-14:
> > Your printer was lagging 28 commands behind the host. Try to post your configuration for inspection.
> 
> Exactly!! @GhostlyCrowd do you have the full log? So we can see where the lag started? I should never happen.

I do indeed, It actually failed while i was sitting beside it in a 15 minute window i had my headphones on so i didnt clue in it was a bit too quiet. 

[octoprint.log.2021-03-13.zip](https://github.com/MarlinFirmware/Marlin/files/6137336/octoprint.log.2021-03-13.zip)

She is kinda fat issues are at the end of the log

rhapsodyv, 2021-03-14:
This is the begin of the crazyness: 
```
2021-03-11 04:25:18,851 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 92490, current line = 92494
| Last lines in terminal:
| Send: N92485 G2 X144.652 Y171.509 I-16.002 J131.165 E0.21250*118
| Recv: //action:notification Good accuracy
| Recv: ok N92454 P0 B1
| Send: N92486 G3 X139.693 Y171.185 I8.355 J-165.997 E0.29610*65
| Recv: ok N92455 P0 B2
| Send: N92487 G2 X138.635 Y171.263 I-0.064 J6.345 E0.31400*69
| Recv: ok N92456 P0 B2
| Send: N92488 G1 X138.254 Y171.259 E0.3204*88
| Recv: ok N92457 P0 B2
| Send: N92489 G3 X129.842 Y169.677 I13.496 J-94.917 E0.46460*71
| Recv: ok N92458 P0 B3
| Send: N92490 G3 X123.402 Y166.381 I3.960 J-15.676 E0.58750*117
| Recv: ok N92459 P0 B4
| Send: N92491 G3 X120.621 Y163.189 I25.447 J-24.978 E0.65880*72
| Recv: ok N92460 P0 B5
| Send: N92492 G1 X119.839 Y162.338 E0.6783*90
| Recv: ok N92461 P0 B6
| Send: N92493 G2 X122.719 Y158.937 I-290.100 J-248.580 E0.75330*99
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 92489
| Recv: Resend: 92490
2021-03-11 04:25:18,858 - octoprint.util.comm - INFO - Firmware didn't send an 'ok' with their resend request. That's a known bug with some firmware variants out there. Simulating an ok to continue...
2021-03-11 04:25:18,865 - octoprint.util.comm - INFO - Ignoring resend request for line 92490, that still originates from lines we sent before we got the first resend request
```

Everything after it, is just luck and fail.

I don't know much about octo, but he should never ignore a missing 'ok'!! If a 'ok' is missing, it should timeout and resent the command, or abort. A missing 'ok' is an error!!

Do you have the serial log too?

X-Ryl669, 2021-03-14:
```
Recv: ok N1021683 P6 B31
| Send: N1021677 G1 X170.508 Y100.301 E0.0223*93
| Send: N1021678 G1 X170.773 Y100.338 E0.0250*82
| Recv: ok N1021684 P6 B31
```
This. 2 sends without a ok in between. You'll need to check your octoprint configuration, there's nothing the printer can do to purge the extra command.

GhostlyCrowd, 2021-03-14:
> This is the begin of the crazyness:
>
> ```
> 2021-03-11 04:25:18,851 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 92490, current line = 92494
> | Last lines in [...]

I had serial logging off (performance suggestion by octo, its off by default) I have it on now and am currently reprinting, but again like I said with serial 2 off.   I am just concerned this is a serial stream mix-up with multi where the ok ended up to the tft and not octo. 

As for Octo "simulating" a missed ok, I've never actively told it to do so it must be a default enabled option.  With that aside, i agree it should fail if the printer stops sending an ack but the issue is why is the ack going missing and why now?

rhapsodyv, 2021-03-14:
Take a look in this option, it should be always 1, always: 

[image]

GhostlyCrowd, 2021-03-14:
Appears to be fine 

[image]

rhapsodyv, 2021-03-14:
Well... the issue here is that Marlin should never skip an 'ok' response... So, we need the serial log to see where and possibly why, this is happening.

GhostlyCrowd, 2021-03-14:
> Well... the issue here is that Marlin should never skip an 'ok' response... So, we need the serial log to see where and possibly why, this is happening.

I wish i had one, serial logging was off due to recommendations on octo and it being the default. I have it on now and am attempting to recreate failure.

X-Ryl669, 2021-03-15:
If you look at the buffer usage (the `Bxx` part in Marlin's answer), it's swinging from 31 to 16. The only way it can decrease from 31 to 30 and less on is when the host sends 2 successive commands (that are stored in the serial's port buffer). Because Marlin is doing a simple "fetch a command from Serial to store into the queue / process a command from the queue" loop. 
There are few exceptions here, when the printer is busy processing command, it outputs a "busy processing" and fetch commands from the serial again. But a compliant host will not jam more command in that case and just wait for ok before continuing filling more into.

So I think it's the host that's generating too many commands here.

GhostlyCrowd, 2021-03-15:
> If you look at the buffer usage (the `Bxx` part in Marlin's answer), it's swinging from 31 downto 16.
> The only way it can decrease from 31 to 30 and less, is when the host sends 2 successive commands (that are stored in the serial's port buffer). Because Marlin is doing a simple "fetch a command from Serial to store into the queue / process a command from the queue" loop.
> 
> There are few exceptions here, when the printer is busy processing command, it outputs a "busy processing" and fetch commands from the serial again. But a compliant host will not jam more command in that case and just wait for ok before continuing filling more into.
> 
> So I think it's the host that's generating too many commands here.

We deduced its bufferbuddy plugin, however I have the serial logs I recreated the failure. Here are the logs if it is indeed bufferbuddy thats a easy fix, just ditch it. if not maybe the logs will be useful. 

Here is a link to my onedrive storage as its too large for git https://1drv.ms/u/s!Aht5eQ6fqQw3ofcJrkPNFM-SEHHskA?e=jGOHJr

rhapsodyv, 2021-03-15:
@GhostlyCrowd your octoprint is sending two commands at time, lots of times. This is what is cause the issues.

Right from the start:
```
2021-03-14 15:59:19,598 - Changing monitoring state from "Starting" to "Printing"
2021-03-14 15:59:19,609 - Send: N1 G90*17
2021-03-14 15:59:19,616 - Send: N2 M82*27
2021-03-14 15:59:19,624 - Recv: ok N1 P31 B31
2021-03-14 15:59:19,628 - Recv: ok N2 P31 B31
```

This don't work on Marlin.

ldursw, 2021-03-15:
This behavior seems to be what BufferBuddy does. It uses `ADVANCED_OK` to try to fill the queue buffer as much as possible to reduce underruns. It will send more than one command at a time without waiting for `ok`.

https://chen.do/diagnosing-reduced-print-quality-with-octoprint/
https://chen.do/adding-buffer-monitoring-to-marlin/
https://github.com/chendo/BufferBuddy

rhapsodyv, 2021-03-15:
Bufferbuddy should not be used. It breaks the reprap protocol and breaks Marlin.

rhapsodyv, 2021-03-15:
https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer

```
The basic protocol responses look like this:

xx [line number to resend] [T:93.2 B:22.9] [C: X:9.2 Y:125.4 Z:3.7 E:1902.5] [Some debugging or other information may be here]

xx can be one of:

ok - The RepRap machine is ready to receive the next line from the host.
```

GhostlyCrowd, 2021-03-15:
Thanks for having a look guys I really appreciate it, I've ditched BufferBuddy.

X-Ryl669, 2021-03-15:
Please open another bug if you think it's a bug with BufferBuddy. 
The reported Bxx part in the Advanced OK feature would be valid if Marlin was multitasked. 
It's not so while it's processing the first command, it's not receiving the next command. 

This command is temporary stored in the serial's buffer and the place for commands in the serial's buffer is limited, leading to corruption of command if too much data feeds in.

GhostlyCrowd, 2021-03-15:
> Please open another bug if you think it's a bug with BufferBuddy.
> The reported Bxx part in the Advanced OK feature would be valid if Marlin was multitasked.
> It's not so while it's processing the [...]

It is a BufferBuddy issue, but since it violates the reprap protocol, is this an issue for marlin to address?   Or is this something that you're considering adding into marlin, multitasking commands?  I'm willing to open a bug report if its something that you guys want to dig into, if it's a non issue on marlins side I won't, as it would be out of the scope of marlins intended usage?

X-Ryl669, 2021-03-15:
I've not planned to address this. But if @chendo wants to take a look, you should create a specific issue for him (either here or on his repository).

GhostlyCrowd, 2021-03-15:
> I've not planned to address this. But if @chendo wants to take a look, you should create a specific issue for him (either here or on his repository).

Will do.