MarlinFirmware/Marlin #21123: TMC Question

pokornass, 2021-02-18:
I want to ask you guys, Iam starting printer owner and iam gonna to ask you if I need any stepper driver for Bigtreetech skr E3-dip or if is any driver integrated or something like that ?

Thank you so much for any answer to help me.

MohanMaker, 2021-02-18:
If you purchase the "dip" version you need to buy external stepper drivers to plug in (I recommend TMC 2209's). If you buy the non dip version, the drivers are already integrated into the board.

pokornass, 2021-02-18:
> If you purchase the "dip" version you need to buy external stepper drivers to plug in (I recommend TMC 2209's). If you buy the non dip version, the drivers are already integrated into the board.

so how many stepper drivers type TMC2208 do I need?

thisiskeithb, 2021-02-18:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-18 by thisiskeithb