MarlinFirmware/Marlin #21129: [BUG] Uncorrect behavior of CASE_LIGHT_ENABLE on SKR mini E3 V1.2, pin PA1 (SERVOS)

trengtor, 2021-02-19:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6008728/Configuration.zip)
[image]

### Bug Description

When you change the brightness setting, the brightness of the LED backlight itself does not change. It turns off when set to 0% and turns on at full brightness at 1%. An oscilloscope check confirmed that the PA1 output is either 0 (0%) or 1 (> = 1%). Turning on/off works OK.
But if you turn off the brightness control (uncomment #define CASE_LIGHT_NO_BRIGHTNESS), the on/off also stops working. The backlight is always on in this case.
Tested on Marlin-bugfix-2.0.x from 17.01.2021

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

See description

**Expected behavior:**

See description

**Actual behavior:**

See description

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2021-02-19:
> Tested on Marlin-bugfix-2.0.x from 17.01.2021

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

trengtor, 2021-02-19:
> > Tested on Marlin-bugfix-2.0.x from 17.01.2021
> 
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Ok

trengtor, 2021-02-20:
[image]
The same behavior ­- bug confirmed

ellensp, 2021-02-20:
Please enable #define PINS_DEBUGGING in Configuration_adv.h then send a M43 and capture the output.
If anything else is on PA1 make sure to disable it

trengtor, 2021-02-20:
> Please enable #define PINS_DEBUGGING in Configuration_adv.h then send a M43 and capture the output.
> If anything else is on PA1 make sure to disable it

[image]
Only this. Need to comment or no?

ellensp, 2021-02-20:
M43 may reveal more, but yes comment that out. Marlin may be trying to use it twice

trengtor, 2021-02-20:
> M43 may reveal more, but yes comment that out. Marlin may be trying to use it twice

Tested: Comment (//) not affects to PA1 bechavior

trengtor, 2021-02-20:
> Please enable #define PINS_DEBUGGING in Configuration_adv.h then send a M43 and capture the output.
> If anything else is on PA1 make sure to disable it

I can't install maple usb driver for capture M43 output via Pronterface etc. (W7 x64 PRO) 
Via TFT35 V3.0 it's impossible (listing is too long & too fast).

trengtor, 2021-02-20:
So, I think the ploblem live in code interaction with new HAL (because PWM silutions was changed)

trengtor, 2021-02-22:
> > Tested on Marlin-bugfix-2.0.x from 17.01.2021
> 
> Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Need another info?

trengtor, 2021-02-24:
@thisiskeithb do you have any thoughts or ideas on this issue?

trengtor, 2021-04-20:
The problem is still present. Tested on April 19 bugfix version.

trengtor, 2021-05-22:
Hi guys. Hawe you any news? Problem still hawe.

thinkyhead, 2021-06-05:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2021-11-25:
Lots of PWM stuff was reworked recently, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, is there any difference between the `_maple` & non-`_maple` environments?

closed 2022-04-24 by github-actions[bot]