MarlinFirmware/Marlin #21130: [FR] (Seperate bed dimension for autoleveling for offset probes) or is there already?

Hobby4life, 2021-02-19:
### Description

I have an Ender 3 v2, with a BLTouch probe, works fine but can always be better.
The probe is offset about 40mm to the left.

`#define X_BED_SIZE 220` is limiting the travel when doing a level procedure.
The probe can not fully use the whole bed to probe properly.
It is limited to the max travel of the nozzle not probe.
I can trick this by allowing the travel after homing by adding 20mm more travel in:

```
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MAX_POS X_BED_SIZE + 20 // Extended X travel for proper bed leveling
```

This works, but this means I can now manually travel the nozzle outside of the printing area.
And thats tricky.

### Feature Workflow

Add seperate probe travel limits to marlin.
Only be used when probing.

For example:
// Travel limits (mm) when **probing**, corresponding to endstop positions.
`#define X_MAX_PROBE_POS X_BED_SIZE + 20`

If probing is done then maintain:
`#define Y_BED_SIZE 220`
and
`#define X_MAX_POS X_BED_SIZE`

thisiskeithb, 2021-02-19:
This can already be achieved with `X_MIN_POS`/`X_MAX_POS` & `Y_MIN_POS`/`Y_MAX_POS`.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](https://marlinfw.org)
- RepRap.org [Marlin Forum](https://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Discord server](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-19 by thisiskeithb

Hobby4life, 2021-02-19:
I dont know why you have closed this so quickly?

But it doesnt answer my question.

Homing center totally behaves different the the center of the autolevel probe pattern.

qwewer0, 2021-02-19:
@Hobby4life 
The X and Y_BED_SIZE isn't limiting the travel area, it is only the XYZ min and max position.
If your printer can move beyond the bed area, then set it so, that is what XY min and max position is for.

If you are afraid of printing outside of the bed area, then set the bed size correctly in the slicer and it won't print outside.

Hobby4life, 2021-02-19:
Thats exactly what I am doing now.. but a proper way would be that the user never manually or if invalid bed dimensions are set in a slicer that can exceed the physical bed dimensions.. only probing may do that.

Thats where I was pointing at.

ManuelMcLure, 2021-02-19:
There are reasons other than probing for the nozzle to be able to exceed the bed limits. For example, my start gcode puts the nozzle at X=-14 while the nozzle heats so it can be swiped by a brush attached to the bed once the nozzle is at temperature. Or the parking position for the nozzle while changing filament can often be off the bed to avoid spitting purged filament onto the print.