MarlinFirmware/Marlin #21137: [FR] I2C Filament diameter sensor

Fabiandelete, 2021-02-20:
### Description

Support for I2C filament diameter sensors such as Thomas Sanladerer's infidel. [Youtube vid.](https://youtu.be/RYgdLPe_T0c)

### More Info

Marlin currently only supports analog filament diameter sensors. The Infidel currently doesn't support analog out. Furthermore; I2C saves analog in pins. Is this something that can easily be implemented or is it harder to do?

Grogyan, 2021-03-01:
I saw the same, and Prusa himself has I2C filament sensors.
Good place to start with to support I2C filament sensors.

Problem with both, is that I2C is not really designed for being in cable longer than 15cm.  Depending on the printer configuration, and the potential that these cables are not EMI shielded, can result in incorrect data being received by the control board.

Grogyan, 2021-03-01:
We need these sensors on CAN bus.  
Which will save a lot on headaches down the road trying to fix intermittent faults caused by EMI.

ksanislo, 2021-03-01:
@Grogyan Adding CAN hardware would also add to the cost of the device, and it seems the goal of this is to be as cheap as possible. Since this is on an ATTINY85 it's got a little bit of processing power, maybe just adding a checksum byte would be sufficient to identify when there has been a failed reading and disregard it?

ellensp, 2021-03-01:
@ksanislo please stop derailing all project that are i2c based by pushing for CAN bus. Start your own FR for can bus.

ellensp, 2021-03-01:
@Grogyan please stop derailing all project that are i2c based by pushing for CAN bus. Start your own FR for can bus.

Grogyan, 2021-03-01:
@ellensp Not derailing, but making sure that people are aware of the inherent flaws of I2C.

As I stated much earlier, it is good to get I2C filament sensing in.

jrmoserbaltimore, 2021-03-17:
Can this be done with a bidirectional protocol, e.g. Marlin sends requests for diameter, diameter sensor sends a response; `M405 C<mm>` tells the sensor to take a `C`alibration reading based on something `<mm>` in diameter already inserted; `M405 C` prints the calibration table (because `C` is pointless without a measurement); and so forth.

Sensors do need to be calibrated.  My micropython firmware for a Pi Pico sensor (adapting Sanladerer's design, although aside from the basic approach it turned out to be a complete redesign built on his engineering research) can read text from a table and can be induced to calibrate by jumping GPIOs and pressing the button, but it'd all be less-convoluted if I could take calipers to various round-profile rods, stick them inside, and tell Marlin how big each is so it can instruct the sensor to calibrate.

I can't decide if I'd rather incorporate the sensor into the extruder, as I've considered putting it beyond the bowden extruder so Marlin can back the filament out until the sensor reports a sudden drop, then nudge it forward, auto-sensing whatever the value of `M405 D` should be.

bolsoncerrado, 2023-07-28:
> ### Description
> Support for I2C filament diameter sensors such as Thomas Sanladerer's infidel. [Youtube vid.](https://youtu.be/RYgdLPe_T0c)

+100