MarlinFirmware/Marlin #21138: [BUG]  I2C position encoders for closed loop control

DrumClock, 2021-02-20:
Hy!
I enabled the `#define I2C_POSITION_ENCODERS` function and used this sensor with AS5600 chip.

[image]

Set in configuration_adv (so far only for X axis):

```
#define I2C_POSITION_ENCODERS
#if ENABLED(I2C_POSITION_ENCODERS)

  #define I2CPE_ENCODER_CNT         1                       // The number of encoders installed; max of 5
                                                            // encoders supported currently.

  #define I2CPE_ENC_1_ADDR          I2CPE_PRESET_ADDR_X     // I2C address of the encoder. 30-200.
  #define I2CPE_ENC_1_AXIS          X_AXIS                  // Axis the encoder module is installed on.  <X|Y|Z|E>_AXIS.
  #define I2CPE_ENC_1_TYPE          I2CPE_ENC_TYPE_ROTARY   // Type of encoder:  I2CPE_ENC_TYPE_LINEAR -or-
                                                            // I2CPE_ENC_TYPE_ROTARY.
  #define I2CPE_ENC_1_TICKS_UNIT    4096                    // 1024 for magnetic strips with 2mm poles; 2048 for
                                                            // 1mm poles. For linear encoders this is ticks / mm,
                                                            // for rotary encoders this is ticks / revolution.
  #define I2CPE_ENC_1_TICKS_REV     (16 * 200)            // Only needed for rotary encoders; number of stepper
                                                            // steps per full revolution (motor steps/rev * microstepping)
  //#define I2CPE_ENC_1_INVERT                              // Invert the direction of axis travel.
  #define I2CPE_ENC_1_EC_METHOD     I2CPE_ECM_MICROSTEP     // Type of error error correction.
  #define I2CPE_ENC_1_EC_THRESH     0.10                    // Threshold size for error (in mm) above which the
                                                            // printer will attempt to correct the error; errors
                                                            // smaller than this are ignored to minimize effects of
                                                            // measurement noise / latency (filter).
```
Unfortunately, when trying to move the X-axis motor, the motor did not return to its original position as I expected.

What's wrong, please?

My configuration:  [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6014896/Configuration.zip)

DrumClock, 2021-02-20:
the problem will probably be in the addressing of the AS5600 module
```

>>> M861 x
SENDING: M861 X
Encoder 0:
Warning. X axis encoder not detected!
```

ellensp, 2021-02-20:
Datasheet for the chip says 
The 7-bit slave address of the AS5600 is 0x36

Marlin/src/feature/encoder_i2c.h says
#define I2CPE_PRESET_ADDR_X           30
#define I2CPE_PRESET_ADDR_Y           31
#define I2CPE_PRESET_ADDR_Z           32
#define I2CPE_PRESET_ADDR_E           33

So try changing I2CPE_PRESET_ADDR_X to 36

But I doubt the i2c protocol in this chip uses is anything like the original it was written for.

It like attaching a dogs leg to a snake...  the snake just doesn't know what to do with it.

ellensp, 2021-02-20:
also it once again isnt a bug, but a feature request for a different type of encoder

DrumClock, 2021-02-20:
Hi @ellensp 
the Aus3D module cannot be purchased  - it would be good to convert it to AS5600 

Changing the address to 36 didn't help.

```
>>> M861 x
SENDING: M861 X
Warning. X axis encoder not detected!
```

I tried 54 (in binary 0110110)
 .... it reads something but I don't know if the snake will be able to run with it .... :)

```

>>> M861 x
SENDING: M861 X
Encoder 0:
X axis encoder passes test; field strength good.
>>> G28
SENDING: G28
echo: Active Extruder: 0
X: LARGE ERR 242; diffSum = 8000.0000
echo: busy: processing
X: LARGE ERR 3030; diffSum = 8000.0000
echo: busy: processing
X: LARGE ERR 2101; diffSum = 4936.0000
echo: busy: processing
X: LARGE ERR 9293; diffSum = 21520.0000
X axis encoder homed, offset of 0 ticks.
echo: busy: processing
X: LARGE ERR 3515; diffSum = 8000.0000
echo: busy: processing
X: LARGE ERR 3757; diffSum = 4000.0000
echo: busy: processing
X: LARGE ERR 3636; diffSum = 8000.0000
echo: busy: processing
X: LARGE ERR -8942; diffSum = 1732.0000
echo: busy: processing
X: LARGE ERR -7272; diffSum = 19200.0000
echo: busy: processing
X: LARGE ERR -7272; diffSum = 19200.0000
echo: busy: processing
X: LARGE ERR -7272; diffSum = 19200.0000
echo: busy: processing
X: LARGE ERR -8727; diffSum = 19200.0000
echo: busy: processing
X: LARGE ERR -9309; diffSum = 0.0000
echo: busy: processing
X: LARGE ERR -8436; diffSum = 19200.0000
echo: busy: processing
X: LARGE ERR -8727; diffSum = 19200.0000
echo: Active Extruder: 0

```

DrumClock, 2021-02-20:
I have several modules with an AS5600 chip, so I could test the modified FW before releasing PR

ellensp, 2021-02-21:
One issue I see is that is there doesn't seem to be any way to change the address.
You need a unique address per axis.Or unique i2c busses...
might need these too TCA9548A I2C Multiplexer, this adds extra levels of complexity.

Grogyan, 2021-02-21:
I would not recommend I2C be used on closed loop drivers.  I2C is meant for very short distances .

CAN bus is far more resilient EMI

Just my 2c

DrumClock, 2021-02-21:
Hi @Grogyan @ellensp

Is it worth solving the close Loop function in Marlin at all? 

I haven't seen the CAN bus on any board.

I

Grogyan, 2021-02-21:
[Trinamic](https://www.trinamic.com/products/pandrive-smart-motors/) and others, do have complete closed loop stepper controllers.

Closed loop should really be on the motor itself.    The control board should just need to state how far motor should travel.

The discussion of whether the control board really needs to know how far it has travelled, is always ongoing.
My personal take on it, is that the control board doesn't really need to know, it may need to know when it has finished it's move, but such instances are rare.

Better to fix the issue at the source than try to fix the symptom.  Ie skipped steps is the symtom, either by EMI, under current, wrong motor, over current etc

DrumClock, 2021-02-21:
@Grogyan 
OK, so it is more advantageous to use eg MKS Servo42B module close loop.

That's probably why Aus3D has finished with the module of its module, 
and the function in Marlin is just a remnant that is no longer used.

Thanks.

closed 2021-03-29 by DrumClock