MarlinFirmware/Marlin #21143: Change filament (M600) disconnects octoprint from printer

jpokorny12, 2021-02-20:
Hello,
i have problem with filament change.
I add to gcode M600. When the printer reads the M600, the filament change begins. After a while, communication with the printer is disconnected.
I try send M600 via terminal and it works.
When replacing the filament, I can't control the filament change via the dialog box in octoprint (continue, purge more). Could there be an error associated with this?
I try send M600 via terminal and it works.
in octoprint i have installed filament change plugin.
In marlin i have uncomment host_action_commands and in octoprint i have M600 in long running comands.

I am adding a link where I asked about this problem on Octoprint github
https://github.com/OctoPrint/OctoPrint/issues/4027

cp2004, 2021-02-20:
Provide the serial log (or the paste from the OctoPrint ticket) as well as all the stuff from the template such as configuration for people to look into it.

---

For others, if you look in my [last comment](https://github.com/OctoPrint/OctoPrint/issues/4027#issuecomment-782723325) you'll notice this has been reported a number of times, but I haven't been able to find a ticket here that says 'It's fixed!'. Filament change works just fine for me, and many others, probably configuration specific (**hint to @jpokorny12**: provide your config files)

jpokorny12, 2021-02-20:
[serial (2).log](https://github.com/MarlinFirmware/Marlin/files/6016142/serial.2.log)
[serial (1).log](https://github.com/MarlinFirmware/Marlin/files/6016143/serial.1.log)
 and my marlin conf
[Configuration - kopie.txt](https://github.com/MarlinFirmware/Marlin/files/6016147/Configuration.-.kopie.txt)
[Configuration_adv - kopie.txt](https://github.com/MarlinFirmware/Marlin/files/6016148/Configuration_adv.-.kopie.txt)

ellensp, 2021-02-20:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

jpokorny12, 2021-02-21:
Thanks, it works.
When replacing the filament, the octoprint does not disconnect.

But the filament exchange control does not work via pop-ups in octoprint. When i click in pop-up window on something action. After filament change, it disconect

[Configuration - kopie.txt](https://github.com/MarlinFirmware/Marlin/files/6017661/Configuration.-.kopie.txt)
[Configuration_adv - kopie.txt](https://github.com/MarlinFirmware/Marlin/files/6017662/Configuration_adv.-.kopie.txt)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/6017667/serial.log)

ellensp, 2021-02-22:
Please provide a screen shot of this new pop windows your now taking about.

I presume you talking about "HOST_PROMPT_SUPPORT"

Also what version of octoprint are you running?

ellensp, 2021-02-22:
Found the version on your other post
Version of OctoPrint: OctoPrint 1.5.3
Operating system running OctoPrint: OctoPi 0.18.0

jpokorny12, 2021-02-22:
I try make video with this problem
[https://youtu.be/l_TpWDLn6-M](url)
[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/6021097/octoprint.log)
[serial.log](https://github.com/MarlinFirmware/Marlin/files/6021098/serial.log)

closed 2021-06-24 by github-actions[bot]

JamesBewley, 2021-07-01:
I have this issue with octoprint 1.6.1 and this marlin build https://marlin.crc.id.au/firmware/Ender%203%20V2%20-%20Stock/