MarlinFirmware/Marlin #21144: [BUG] Impossible to compile when LEVELING_NOZZLE_TEMP is disabled

Guilouz, 2021-02-20:
Impossible to compile when #define LEVELING_NOZZLE_TEMP 120 is disabled.

`#define PREHEAT_BEFORE_LEVELING
#if ENABLED(PREHEAT_BEFORE_LEVELING)
  //#define LEVELING_NOZZLE_TEMP 120   // (°C) Only applies to E0 at this time
  #define LEVELING_BED_TEMP     60
#endif`

VsCode return : 
'LEVELING_NOZZLE_TEMP' was not declared in this scope; did you mean 'LEVELING_BED_TEMP'?

```
Marlin/src/gcode/bedlevel/abl/G29.cpp: In static member function 'static void GcodeSuite::G29()':
Marlin/src/gcode/bedlevel/abl/G29.cpp:413:48: error: 'LEVELING_NOZZLE_TEMP' was not declared in this scope; did you mean 'LEVELING_BED_TEMP'?
  413 |         if (!dryrun) probe.preheat_for_probing(LEVELING_NOZZLE_TEMP, LEVELING_BED_TEMP);
      |                                                ^~~~~~~~~~~~~~~~~~~~
      |                                                LEVELING_BED_TEMP
```

ellensp, 2021-02-21:
This PR above fixes it.

ellensp, 2021-02-21:
now merged, closing

closed 2021-02-21 by ellensp