MarlinFirmware/Marlin #21146: G26 mesh validation will not continue without R flag, Documentation says R defaults to 999

tastyratz, 2021-02-21:
### Bug Description

Documentation https://marlinfw.org/docs/gcode/G026.html for G26 states:

[R<int>] | Number of G26 Repetitions (otherwise 999)

If a default value is to be populated then the flag should not be required to perform the function.

### Configuration Files

### Steps to Reproduce
Run G26 from terminal

**Expected behavior:**
Running G26 without flags should perform validation with default values as documented

**Actual behavior:**

G26 run without R999 reads Home xyz on the lcd and does not continue. 
G26 run WITH R999 performs mesh validation print.

tastyratz, 2021-02-21:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6017127/config.zip)

ellensp, 2021-02-21:
```
*   R #  Repeat      Prints the number of patterns given as a parameter, starting at the current location.
 *                    If a parameter isn't given, every point will be printed unless G26 is interrupted.
 *                    This works the same way that the UBL G29 P4 R parameter works.
 *
 *                    NOTE:  If you do not have an LCD, you -must- specify R. This is to ensure that you are
 *                    aware that there's some risk associated with printing without the ability to abort in
 *                    cases where mesh point Z value may be inaccurate. As above, if you do not include a
 *                    parameter, every point will be printed.
```

tastyratz, 2021-02-21:
I do, in fact, have an LCD display (this is on a Monoprice maker ultimate aka Wanhao d6).

I use manual mesh leveling. I did not see a menu item for it (I wish it did and ABL wasn't assumed but that was covered in another now closed post). I used G26 from octoprint terminal.
 
If the function did in some way, in fact, require the flag to continue, should it not have returned an error both in the terminal and on the LCD?

Also, where did you copy that documentation from? It's not clear from the page on G26 linked in the original post. If that is the case, the documentation page on the web should likely reflect it.

ellensp, 2021-02-21:
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/bedlevel/G26.cpp#L77

The comment in source are always more up to date that the documentation, which is often several versions behind.

ellensp, 2021-02-21:
R is a flag, you need it to activate this feature, 999 refers to the parameter with the flag, if you omit the parameter this is the default value for the parameter.
It is not the same as omitting the R flag.

closed 2021-02-21 by ellensp