MarlinFirmware/Marlin #21147: [BUG] UBL - M421 C does not modify the correct spot

lightmaster, 2021-02-21:
### Bug Description

If I use `G42 Ix Jx` to move to a spot, then use `M421 C Z/Qxxx.x` to try to set that spot's z-offset, it will modify the spot to the right of where the nozzle currently is. Since I used `G42` to position the nozzle at a particular spot, then `M421 C` should always adjust the spot that I am currently at.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

Delta using env `mks_robin_nano35_stm32`, using UBL leveling. Config compiles on `bugfix-2.0.x` as of 2021-02-20 23:16
[Marlin Bugfix-2.0.x Configuration 2021-02-20.zip](https://github.com/MarlinFirmware/Marlin/files/6017188/Marlin.Bugfix-2.0.x.Configuration.2021-02-20.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Bed level with `G29 P1`
2. Move to first point to fine tune leveling with `G42 Ix Jx`
3. Check spot's current value with `G29 T`
4. Set new Z-offset for spot using `M421 C Z/Qxxx.x`
5. Check spot's new value with `G29 T`
6. See that the spot to the right (`I+1`) has been modified instead.

**Expected behavior:**

The spot that you are currently at should be modified by `M421 C`.

**Actual behavior:**

The spot to the right of your current spot is modified by `M421 C`

#### Additional Information

```
G42 I0 J4
ok
G29 T

Bed Topography Report:

    (  -99, 99)                                                              ( 99, 99)
        0       1       2       3       4       5       6       7       8       9
 9 |   .       .       .       .       .       .       .       .       .       .
   |
 8 |   .       .       .       .     +0.091  +0.136    .       .       .       .
   |
 7 |   .       .     +0.013  +0.027  +0.039  +0.065  +0.075  +0.011    .       .
   |
 6 |   .       .     +0.024  +0.012  +0.021  +0.013  +0.024  -0.001    .       .
   |
 5 |   .     +0.079  +0.037  +0.008  -0.002  -0.003  +0.015  -0.029  -0.122    .
   |
 4 |[  .   ] +0.085  +0.035  +0.015  -0.008  -0.004  +0.009  -0.026  -0.109    .
   |
 3 |   .       .     +0.039  +0.002  -0.011  -0.011  -0.021  -0.023    .       .
   |
 2 |   .       .     -0.026  -0.037  -0.049  -0.051  -0.022  -0.050    .       .
   |
 1 |   .       .       .       .     -0.118  -0.108    .       .       .       .
   |
 0 |   .       .       .       .       .       .       .       .       .       .
        0       1       2       3       4       5       6       7       8       9
    (  -99,  -99)                                                              ( 99,  -99)

ok
M421 C Z0.33
ok
G29 T

Bed Topography Report:

    (  -99, 99)                                                              ( 99, 99)
        0       1       2       3       4       5       6       7       8       9
 9 |   .       .       .       .       .       .       .       .       .       .
   |
 8 |   .       .       .       .     +0.091  +0.136    .       .       .       .
   |
 7 |   .       .     +0.013  +0.027  +0.039  +0.065  +0.075  +0.011    .       .
   |
 6 |   .       .     +0.024  +0.012  +0.021  +0.013  +0.024  -0.001    .       .
   |
 5 |   .     +0.079  +0.037  +0.008  -0.002  -0.003  +0.015  -0.029  -0.122    .
   |
 4 |[  .   ] +0.330  +0.035  +0.015  -0.008  -0.004  +0.009  -0.026  -0.109    .
   |
 3 |   .       .     +0.039  +0.002  -0.011  -0.011  -0.021  -0.023    .       .
   |
 2 |   .       .     -0.026  -0.037  -0.049  -0.051  -0.022  -0.050    .       .
   |
 1 |   .       .       .       .     -0.118  -0.108    .       .       .       .
   |
 0 |   .       .       .       .       .       .       .       .       .       .
        0       1       2       3       4       5       6       7       8       9
    (  -99,  -99)                                                              ( 99,  -99)

ok
```

digant73, 2021-02-21:
With MBL it seems that the value is saved on the previous row (e.g. i2 J2 is saved on I2 J1).

Furthermore, it also seems that we have to disable BL before fine tuning a z height otherwise it seems that a wrong z heigh is saved (such as BL is applied on top of the z height we want to save).

lightmaster, 2021-02-21:
> such as BL is applied on top of the z height we want to save

Are you using `M421 C Zxxx.x` or `M421 C Qxxx.x`? 
`Z` should apply your offset from 0, ie: previous value was `+0.085`, command used is `Z0.330`, new value is `0.330`.
`Q` should add or subtract your offset from the previous value, ie: in the previous example, your new offset would be `0.415`.

digant73, 2021-02-22:
@lightmaster I know that option. I'm using M421 Ix Jx Zx to save the new z value. AFAIK meshes should be applied only when printing if BL is enabled not during fine tuning

closed 2021-05-07 by github-actions[bot]

thinkyhead, 2021-05-07:
The `M421 C` command uses `ubl.find_closest_mesh_point_of_type(REAL, current_position)` so it can only be used to select points that have a value. There doesn't seem to be an option in `ubl.find_closest_mesh_point_of_type` to get just any point — it must be `INVALID`, `REAL`, or `SET_IN_BITMAP`. Maybe this is by design, but for `M421` we may want to just add a new type.

thinkyhead, 2021-05-07:
Now patched in `bugfix-2.0.x`.

lightmaster, 2021-05-07:
Will add this to my list of things to test and verify tomorrow.

lightmaster, 2021-05-07:
Seems to work now. Thx