MarlinFirmware/Marlin #21150: [FR] set dual baud rate for creality v4 boards with DWIN

spacecabbie, 2021-02-21:
### Description

I use this guide:(https://www.reddit.com/r/ender3/comments/ihg2k7/dwin_display_on_ender3_v2_does_not_work_with_a/gbc1f0u?utm_source=share&utm_medium=web2x&context=3) to enable 25000 baud rate between Octoprint and printer

Assuming this is ok and supported (it working fine for me) could this be made part of standard config for Ender 3 v2 (4.2.2 and 4.2.7 boards)

### Feature Workflow

basically adding: (or simulair)
`#if ENABLED(DWIN_LCD)
  #define LCD_SERIAL_PORT 1
  #define BAUDRATE 115200
#endif`

#### Additional Information
Of course the above workaround works i am to wonder why its not in the config perhaps a limitation (only 1 baud rate can be set)?
Since I prefer to keep everything in the config files and recommended I do not like this solution and prefer an "official" one.

IF this is a know workaround but is not included due to adverse effects or possible problems is it possible to include a comment line in either the config file or MarlinCore.cpp since it seems harmless and i am sure quite some people are applying it.

CRCinAU, 2021-02-21:
Why not just install Meatpack and use that?

spacecabbie, 2021-02-22:
> Meatpack

This is the first time I heard about it and sure looks like a cool plugin still (from a quick glance) having the baud rate at 25000 would make that even quicker (unless there is some sort of max).

But, there are so many roads to rome,  reason to request it is to check if this is a good way to go also meatball might be a good one. In the end if no changes need to happen that at least (i hope) a small info text can be addedd to the recommended way to go. (l love the excelend documentation of marlin.)

ellensp, 2021-02-22:
In bugfix (and thus future releases) the LCD baud rate is already separated.

ie
```
#if HAS_DGUS_LCD
  #define LCD_SERIAL_PORT 3
  #define LCD_BAUDRATE 115200
```

closed 2021-02-22 by ellensp