MarlinFirmware/Marlin #21151: Sensorless Homing Fails on TMC2209 after PSU Cycle

drewskix86, 2021-02-21:
Hey Guys,

I have been having an issue with Marlin 2.0.7 where sensorless homing fails to trigger on TMC2209 driver after turning the PSU off and on again via M80/M81 or from connected graphic display.

Sensorless homing works fine when the printer first turns on (PSU defaults to on) however consistently fails after PSU has power cycled. The only way to get it working again is to reset the MCU.

I have tried increasing PSU_POWER_UP_DELAY with no effect. 

I had the same issue since getting the TMC2209's with Marln 2.0.6 and was hoping updates would fix it but no luck. 

I decided to look into the code in more detail and found `restore_stepper_drivers()` in power.cpp which looks like it should restore the stepper functionality but didn't. I compared it to `reset_trinamic_drivers()` which runs when the printer first turns on and saw the additional `#if USE_SENSORLESS` section. After adding the additional section to restore_stepper_drivers() it seems to fix the problem.

Currently running Re-Arm & Ramps 1.6 CoreXY and RepRap Full Graphic Smart Controler. Configs attached.
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6017548/configs.zip)

ellensp, 2021-02-22:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I'm fairly sure this one has been seen and fixed in bugfix already.

drewskix86, 2021-02-22:
I've tried the bugfix release, updated configs/pins files to match my printer and it compiles fine but once on the printer won't load. Can you point me in the direction of the previous issue as all I was able to find was issues related to SPI-based TMC drivers sensorless homing issues?

GMagician, 2021-02-22:
Does your board give power (3.3V) on vmot when PSU is off?

drewskix86, 2021-02-23:
No Vmot is powered directly by PSU

closed 2021-05-07 by github-actions[bot]