MarlinFirmware/Marlin #21165: [FR] marlin to use dual-core ARM4 to have Peripheral control on ARM0+ core

Hellmut1956, 2021-02-22:
Marlin to support dual-core ARM4 controllers to separate G Code processing from peripheral control like display.
Benefits:
1. Dual-Core ARM 4 Controller available up to 220 MHz
2. ARM3-Core to exclusively execute G-Code processing
3. ARM0+-Core to handle peripheral functionality to off-load ARM3 Core.

X-Ryl669, 2021-02-25:
Do you have any reference in mind ?

Hellmut1956, 2021-02-25:
@X-RYl669: What do you mean by your question? Sorry, but I am not a native English speaker.

X-Ryl669, 2021-02-25:
Do you know of a board with such CPU ? If yes, what is the model / reference ?

Hellmut1956, 2021-02-25:
Not a controller board made to drive 3D printers. The 3D printing ecosystem is totally focused on low cost and the ecosystem is far from understanding the components of their printers, far from understanding the process of 3D printing as for ways how to improve it. 3D printing is not a trivial process, a dance of many parameters. The 3D market for 3D printers for the consumer and the DIY users. When I did mention the topics related to improving the G Code processing and the NEMA17 stepper motors options to improve their performance at basically all 3D printing communities I was attacked not on a factual basis with one exception but on a personal level. This all did stop when I documented my position using a chapter of the FAQ section of the Trinamics website. After that suddenly all went quiet.
I have contacted an important source specialized in offering inexpensive boards, LPCXpressocxxxx, and asked what board they would recommend to me for my experiments. I started using their LPCXpresso1769 board nearly 8 years ago with boards from Trinamic that they gave me for free. It was their stepRocker board that as of today is basically phased out and available at astronomic prices. The other board they gave me sometime later was a board called "motionCookie" that never was made a product by them. You should be aware that the 3D printing ecosystem is evading boards that would put their products with good profit margins at risk. Even BTT support does not know enough officially anything relating to their systems. I i.e. asked them if on their board BTT-SKR-PRO V1.2 the connectors for the motor power input were passed one-to-one to the power pins of the sockets for the SilentStepSticks. They praise that that port accepts voltages of up to 30 VDC. If this is true, that there cannot be a connection of this circuit to other circuits on the board that is fed from their other Power Input with either 12 VDC or 24 VDC. This means that they translate the 12 VDC or 24 VDC to 5 VDC or 3.3 VDC required in other parts of their board. I have purchased a Meanwell 600W 48 VDC power supply to check if that voltage is passed to the proper pins on the socket for the SilentStepperSticks driver boards which I will then test using the SilentStepSticksTMC5160hv that I purchased from Watterott, the partner of Trinamic for those boards. It is also interesting that BTT does hide their internal circuitry as a yellow box on their schematic.
So be aware of, 3D printing for the consumer market is not yet showing any thinking towards dual-core controller that also can be sourced with up to 220 MHz. Supplier of systems and boards for this section of the 3D printing market is not displaying the information about nominal values for voltage and current and the specified torque that normally always on a plate on the device. When you investigate the market you will find NEMA17 stepper motors that have 3x the torque of the weakest and therefore cheapest by pennies only stepper motors used in 3D printers. There seems to be no problem replacing the stepper motors used to move along the X, Y, and Z-axis. For moving the filament no change should be required.
But also using SilentStepperSticks5161 for heating the plate would mean getting it to the required temperature faster and achieving higher temperatures.
But be aware of! What I am looking into does not yet exist! So it is a process to start with a board that exists and runs Marlin2.0.7.2 using SilentStepSticks And do experimenting and also do analyze Marlin code to understand enough to get a feeling of what effort it would take. Remember, it makes a lot of sense to start from something that does work and start into new grounds from there. For example, is the code for Marlin for ARM Cortex controllers using CIMSIS? CIMSIS is kind of a HAL common to all ARM Cortex MCU. If the software was implemented using CIMSIS, then porting it to another ARM Cortex Controller becomes comparatively simple as CMSIS hides the peculiarities of an MCU by making available the same API. So I could imagine that if Marlin 2.0.7.2 uses CMSIS I could purchase a development board or an evaluation board and interface it with one of mine SilentStepSticks5160hv.
So I am using this thread to find members in the community experienced with Marlin so as to share our thoughts. Until then I have enough work to do on my workshop and later on my Ender 5 Plus.