MarlinFirmware/Marlin #21166: firmware German

Julasch73, 2021-02-22:
it is possible to make a firmware from https://marlin.crc.id.au/ in German ? mfg

ellensp, 2021-02-22:
We can not assist on 3rd party web sites.  This site is not run or controlled by Marlin. 

Please contact @CRC on the discord.

closed 2021-02-22 by ellensp

thisiskeithb, 2021-02-23:
@CRCinAU Can you please add a disclaimer or make it more clear that these are two separate entities and support requests should go to you?

CRCinAU, 2021-02-23:
I'm not sure how much more direct I can be.....

[image]

None of the links end up here...

Julasch73, 2021-02-23:
@CRCinAU könnten sie bitte kontakt mit mir aufnehmen ? 

 @CRCinAU please contact me ?