MarlinFirmware/Marlin #21170: [BUG] Occasional command truncation

ldursw, 2021-02-23:
### Bug Description

At any moment while printing the first few characters of a command is cut off. The firmware requests a resend and the host sends the command again. In a few cases the resend is also received truncated just enough to remove the `N12345` part bypassing the buffer and running immediately causing a wrong movement.

This issue is also non-deterministic. The exact same GCode file can cause 0 resends or a few ones. None of them are in the same location in each print. A 3DBenchy can have more than 50 resends during printing.

Example from a serial log
```
2021-02-22 19:18:28,906 - Send: N5343 G0 X112 Y115.869*7
2021-02-22 19:18:28,932 - Recv: ok N5343 P0 B3
2021-02-22 19:18:28,936 - Send: N5344 G1 F1500 E5.5*55
2021-02-22 19:18:28,986 - Recv: ok N5344 P0 B3
2021-02-22 19:18:28,990 - Send: N5345 G1 F1457.2 X111.963 Y115.983 E0.00239*42
2021-02-22 19:18:29,052 - Recv: ok N5345 P0 B3
2021-02-22 19:18:29,057 - Send: N5346 G1 X112 Y116.096 E0.00237*69
2021-02-22 19:18:29,168 - Recv: ok N5346 P0 B3
2021-02-22 19:18:29,173 - Send: N5347 G1 X112 Y115.869 E0.00453*75
2021-02-22 19:18:29,248 - Recv: ok N5347 P0 B3
2021-02-22 19:18:29,253 - Send: N5348 G1 X112.449 Y114.496 E0.02883*93
2021-02-22 19:18:29,513 - Recv: ok N5348 P0 B3
2021-02-22 19:18:29,518 - Send: N5349 G1 X114.272 Y112.899 E0.04837*88
2021-02-22 19:18:29,777 - Recv: ok N5349 P0 B3
2021-02-22 19:18:29,781 - Send: N5350 G2 X111.841 Y110.889 I-541.901 J652.929 E0.06295 F1457*27
2021-02-22 19:18:29,816 - Recv: echo:Unknown command: "5350 G2 X111.841 Y110.889 I-541.901 J652.929 E0.06295 F1457"
2021-02-22 19:18:29,819 - Recv: ok P0 B3
2021-02-22 19:18:29,824 - Send: N5351 G1 X110.317 Y109.804 E0.03733*85
2021-02-22 19:18:29,829 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 5349
2021-02-22 19:18:29,850 - Recv: Resend: 5350
2021-02-22 19:18:29,856 - Recv: ok N5347 P0 B4
2021-02-22 19:18:29,860 - Send: N5350 G2 X111.841 Y110.889 I-541.901 J652.929 E0.06295 F1457*27
2021-02-22 19:18:30,059 - Recv: ok N5350 P0 B3
2021-02-22 19:18:30,062 - Send: N5351 G1 X110.317 Y109.804 E0.03733*85
2021-02-22 19:18:30,104 - Recv: ok N5351 P0 B3
2021-02-22 19:18:30,110 - Send: N5352 G1 X109.831 Y109.219 E0.01518*94
2021-02-22 19:18:30,135 - Recv: ok N5352 P0 B3
2021-02-22 19:18:30,142 - Send: N5353 G1 X108 Y108.471 E0.03947*71
2021-02-22 19:18:30,169 - Recv: ok N5353 P0 B3
2021-02-22 19:18:30,175 - Send: N5354 G1 X108 Y104.525 E0.07875*72
2021-02-22 19:18:30,200 - Recv: ok N5354 P0 B3
2021-02-22 19:18:30,205 - Send: N5355 G1 X106.429 Y103.97 E0.03325*103
2021-02-22 19:18:30,259 - Recv: ok N5355 P0 B3
2021-02-22 19:18:30,263 - Send: N5356 G1 X106.183 Y103.214 E0.01587*84
2021-02-22 19:18:30,316 - Recv: ok N5356 P0 B3
2021-02-22 19:18:30,320 - Send: N5357 G1 X105.186 Y102.487 E0.02462*87
2021-02-22 19:18:30,576 - Recv: ok N5357 P0 B3
2021-02-22 19:18:30,581 - Send: N5358 G1 F1500 E-5.5*23
2021-02-22 19:18:30,612 - Recv: ok N5358 P1 B3
```

If the GCode contains only linear movements the issue doesn't happen. When printing with arcs the issue shows up. After some help from @X-Ryl669 in #21122 it looks like the serial buffer receives more data that can process and ends up overwriting the oldest data causing the command truncation. G2/G3 commands are larger than G0/G1 so it's more likely to saturate the buffer.

I'm printing via serial using OctoPrint. I've already tested different USB cables but doesn't make any difference.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6028636/Configuration.zip)

I've tried changing `MAX_CMD_SIZE` (from 64 to 128), `BUFSIZE` (from 4 to 64), `TX_BUFFER_SIZE` (from 0 to 64), `RX_BUFFER_SIZE` (from 0 to 2048). These changes either doesn't make difference or makes it worse.

Enabling `MEATPACK` doesn't make difference.

Serial log with 211 resends in a single print
[resends-log.zip](https://github.com/MarlinFirmware/Marlin/files/6028767/resends-log.zip)

#### Additional Information

Printer: Ender-3 Pro
Board: MKS Robin E3
CPU: STM32F103RCT6

More details here: https://github.com/FormerLurker/ArcWelderPlugin/issues/30#issuecomment-722758433

X-Ryl669, 2021-02-23:
Please try to change the buffer size in `~/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple/include/libmaple/usart.h`, change line 381 from:
```
#define USART_RX_BUF_SIZE               64
```
to 
```
#define USART_RX_BUF_SIZE               128
```
And rebuild. It should have fixed your issue.

ldursw, 2021-02-23:
I did some test prints and all of them were flawless! No resends at all. Thank you so much for helping diagnose this bug.

For a PR should I change only this board like this
```patch
#
# MKS Robin E3D (STM32F103RCT6) and
# MKS Robin E3 with TMC2209
#
[env:mks_robin_e3]
platform      = ${common_stm32f1.platform}
extends       = common_stm32f1
board         = genericSTM32F103RC
platform_packages = tool-stm32duino
extra_scripts = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/mks_robin_e3.py
build_flags   = ${common_stm32f1.build_flags}
- -DDEBUG_LEVEL=0 -DSS_TIMER=4
+ -DDEBUG_LEVEL=0 -DSS_TIMER=4 -DUSART_RX_BUF_SIZE=128
```

Or change the STM32F1 definition like this?
```patch
#
# HAL/STM32F1 Common Environment values
#
[common_stm32f1]
platform          = ststm32@~11.0
board_build.core  = maple
build_flags       = !python Marlin/src/HAL/STM32F1/build_flags.py
  ${common.build_flags}
  -DARDUINO_ARCH_STM32
+ -DUSART_RX_BUF_SIZE=128
build_unflags     = -std=gnu11 -std=gnu++11
src_filter        = ${common.default_src_filter} +<src/HAL/STM32F1>
lib_ignore        = SPI, FreeRTOS701, FreeRTOS821
lib_deps          = ${common.lib_deps}
  SoftwareSerialM
platform_packages = tool-stm32duino
extra_scripts     = ${common.extra_scripts}
  buildroot/share/PlatformIO/scripts/fix_framework_weakness.py
```

Would this behavior occur with all STM32F1 boards?

X-Ryl669, 2021-02-23:
Nope, it won't work if you set the flags here. libmaple is in platformio's framework and it's not built with this flags anyway. There is no way to configure such buffer from outside as far as I've read the code. And to be honest, libmaple is deprecated, this is another reason why we should focus in using the STSTM32 version instead.

But back to the initial issue, the issue happens because Marlin is too slow to read the serial buffer. It's doing "read the buffer, process the command, read the buffer, process the command..." so since processing command is slow, we need large buffers to accumulate data coming from the host.

Yet, Marlin is plagued by the "hidden" buffers in each framework. If one is not suited for the task then it'll loose data. I've spent a lot of time with a LPC176x on this one too.

I think a true solution would be to write the queue code to react on interrupt (that is: bypassing all "framework" buffers), but not all platform support this.

rhapsodyv, 2021-02-23:
It will work if added to the `builg_flags` in the `platformio.ini`..... Maple code uses ifndef...:
```
#ifndef USART_RX_BUF_SIZE
#define USART_RX_BUF_SIZE               64
#endif
```

X-Ryl669, 2021-02-23:
They could be a completely hacky solution by overflowing the RX's ring buffer over the adjacent TX's ring buffer (and allocating another TX buffer for the ring buffer), or less hacky solution by wasting 128 bytes of memory by re-assigning ring buffer's pointers to other, Marlin's provided, buffers...

## Solution 1:
we have this in libmaple:
```
/** USART device type */
typedef struct usart_dev {
    usart_reg_map *regs;             /**< Register map */
    ring_buffer *rb;                 /**< RX ring buffer */
    ring_buffer *wb;                 /**< TX ring buffer */
    uint32 max_baud;                 /**< @brief Deprecated.
                                      * Maximum baud rate. */
    uint8 rx_buf[USART_RX_BUF_SIZE]; /**< @brief Deprecated.
                                      * Actual RX buffer used by rb.
                                      * This field will be removed in
                                      * a future release. */
    uint8 tx_buf[USART_TX_BUF_SIZE]; /**< Actual TX buffer used by wb */
    rcc_clk_id clk_id;               /**< RCC clock information */
    nvic_irq_num irq_num;            /**< USART NVIC interrupt */
} usart_dev;

```
So we need to do something like this in Marlin/src/HAL/STM32F1/MarlinSerial.h:
Replace:
```
struct MarlinSerial : public HardwareSerial {
  MarlinSerial(struct usart_dev *usart_device, uint8 tx_pin, uint8 rx_pin) : HardwareSerial(usart_device, tx_pin, rx_pin) { }
```
by
```
struct MarlinSerial : public HardwareSerial {
  #if defined(RX_BUFFER_SIZE) && RX_BUFFER_SIZE == 2 * USART_RX_BUF_SIZE
    uint8_t txBuf[TX_BUFFER_SIZE];
  #endif
  MarlinSerial(struct usart_dev *usart_device, uint8 tx_pin, uint8 rx_pin) : HardwareSerial(usart_device, tx_pin, rx_pin) {
  #if defined(RX_BUFFER_SIZE) && RX_BUFFER_SIZE == 2 * USART_RX_BUF_SIZE
        rb_init(dev->rb, RX_BUFFER_SIZE, dev->rx_buf);
        rb_init(dev->wb, TX_BUFFER_SIZE, txBuf);
  #endif
   }
```
## Solution 2:
Replace the previous code by:
```
struct MarlinSerial : public HardwareSerial {
  #if defined(RX_BUFFER_SIZE)
    uint8_t rxBuf[TX_BUFFER_SIZE];
  #endif 
  #if defined(TX_BUFFER_SIZE)    
    uint8_t txBuf[TX_BUFFER_SIZE];
  #endif
  MarlinSerial(struct usart_dev *usart_device, uint8 tx_pin, uint8 rx_pin) : HardwareSerial(usart_device, tx_pin, rx_pin) {
  #if defined(RX_BUFFER_SIZE) && RX_BUFFER_SIZE == 2 * USART_RX_BUF_SIZE
  #if defined(RX_BUFFER_SIZE)
        rb_init(dev->rb, RX_BUFFER_SIZE, rxBuf);
  #endif
  #if defined(TX_BUFFER_SIZE)
        rb_init(dev->wb, TX_BUFFER_SIZE, txBuf);
  #endif
   }
```

X-Ryl669, 2021-02-23:
@rhapsodyv Ok, but how do you know what the user has set in his RX_BUFFER_SIZE macro in Configuration.h ?

rhapsodyv, 2021-02-23:
why not all STM32F1 boards that use maple?! 64 bytes is almost nothing... so there's no drawback in setting it to all

rhapsodyv, 2021-02-23:
Or just make marlin do:
```
#define USART_RX_BUF_SIZE  RX_BUFFER_SIZE
```
On some script.

X-Ryl669, 2021-02-23:
@rhapsodyv But it'll be too late. libmaple's usart.c is built independantly of Marlin's definition.

X-Ryl669, 2021-02-23:
While I'm at it, there are many features in platformio.ini and they are somehow linked with Configuration's defines. How do they do that ? We could re-use the logic here ?

rhapsodyv, 2021-02-23:
> @rhapsodyv But it'll be too late. libmaple's usart.c is built independantly of Marlin's definitions.

No, it's not. All PIO libraries are compiled using the same env flags/options.

Run `platformio run -v`.
Here you will see that it will compile maple using our flags:
```
arm-none-eabi-gcc -o .pio/build/mks_robin_nano35/FrameworkArduino/libmaple/rcc.c.o -c -Os -mcpu=cortex-m3 -mthumb -fsigned-char -fno-move-loop-invariants -fno-strict-aliasing --specs=nano.specs --specs=nosys.specs -MMD -MP -fmax-errors=5 -g3 -fmerge-constants -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -nostdlib -march=armv7-m --param max-inline-insns-single=500 -Wall -Os -DPLATFORMIO=50100 -DSTM32F103xE -DSTM32F1 -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103VE -DSS_TIMER=4 -DARDUINO=10808 -DARDUINO_ARCH_STM32F1 -DARDUINO_GENERIC_STM32F103V -DMCU_STM32F103VE -D__STM32F1__ -DBOARD_generic_stm32f103v -DF_CPU=72000000L -DERROR_LED_PORT=GPIOE -DERROR_LED_PIN=6 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DVECT_TAB_ADDR=0x08007000 -IMarlin/src/HAL/STM32F1 -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple/include -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple/stm32f1/include -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple/usb/stm32f1 -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/system/libmaple/usb/usb_lib -I/Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103v /Users/victor/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/libmaple/rcc.c
```

rhapsodyv, 2021-02-23:
Any change to `env` on `pre:script` will be applied on libs and marlin (pio project), any change on `projenv` on `post:script` will be applied only on marlin (pio project).

X-Ryl669, 2021-02-23:
Victor, yes, you're talking of buildflags, but I'm talking of defines in Configuration_adv.h. We need a way to fetch the configuration's macro called "RX_BUFFER_SIZE" and put that in buildflags

X-Ryl669, 2021-02-23:
And buildflags are set up in python, if I understand correctly.

rhapsodyv, 2021-02-23:
> And buildflags are set up in python, if I understand correctly.

Yes, I said that the two options: or set it globally on build_flags (ignoring RX_BUFFER_SIZE), or change our dependency script to add a global build flag with the value of `RX_BUFFER_SIZE`.
All marlin configs are available on env['MARLIN_FEATURES'].

As this is maple specific and maple is deprecated, I would just add `-DUSART_RX_BUF_SIZE=128` to all stm32f1... no issue with that.

X-Ryl669, 2021-02-23:
Ok, I've parsed the configuration file and set the build flags accordingly. 

@ldursw Please pull #21122 again, set RX_BUFFER_SIZE to 128 in Configuration_adv.h (and revert your change to libmaple's platform file), and report if it fixed the issue for you.

ldursw, 2021-02-23:
Now it doesn't even start compiling.

```
Traceback (most recent call last):
  File "Marlin/src/HAL/STM32F1/build_flags.py", line 20, in <module>
    rxBuf = findDefinition("RX_BUFFER_SIZE")
  File "Marlin/src/HAL/STM32F1/build_flags.py", line 7, in findDefinition
    f = open("Marlin/Configuration_adv.h").read().split("\n")
  File "~\.platformio\python3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 68942: character maps to <undefined>
OSError: 'python Marlin/src/HAL/STM32F1/build_flags.py' exited 1:
  File "~\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 334:
    apply_features_config()
  File "Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 148:
    if not env.MarlinFeatureIsEnabled(feature):
  File ~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 299:
    load_marlin_features()
  File "Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 270:
    build_flags = env.ParseFlagsExtended(build_flags)
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "~\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 185:
    for key, value in env.ParseFlags(str(raw)).items():
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 821:
    do_parse(arg)
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 673:
    arg = self.backtick(arg[1:])
  File "~\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Environment.py", line 585:
    raise OSError("'%s' exited %d" % (command, status))
```

p3p, 2021-02-23:
> Yet, Marlin is plagued by the "hidden" buffers in each framework. If one is not suited for the task then it'll loose data. I've spent a lot of time with a LPC176x on this one too.
> I think a true [...]

@X-Ryl669 If you are having issues with my framework put in an issue, I can't fix things if I don't know about them, There are no 'hidden' buffers in the HardwareSerial implementation, the HardwareSerial template class is instantiated with the size required from the client code (Marlin), it will drop data of course if the Marlin reads from it slower than the host transmits as there is no flow control, there are no other buffers between the hardware and Marlin.

The CDC Serial receive is lossless, it can't drop data, it blocks the host using usb flow control rather than overwrite its buffer.

Moving data from the peripheral to the Serial buffers is not always trivial and can't be left to Marlin.

X-Ryl669, 2021-02-23:
@ldursw Please try again with the updated PR. I don't like setting a hard value in a configuration expecting it'll fit all.

X-Ryl669, 2021-02-23:
@p3p The main issue that I can't explain is described in the long bug #21010. To sum up the issue, the OP has 2 serial port in use, a USB that's plugged on a screen (IIUC), and the UART0 that's plugged to his computer. It seems that there are some characters dropping from the buffer or that the serial class does not work anymore after a while. The printer is not crashed, since it's still sending autoreport, but it does not receive anything else anymore.

When checking the "serial.available()" value, it's almost always 0 so Marlin does not make progress until more characters are streamed. I've read your code numerous time and I could not find an issue with it (unless, obviously serial buffer overflow), but I don't think it's an overflow going on here (the OP even increased the HardwareSerial's buffer size without any change to behavior). I've double checked that the OP's firmware binary does actually call HardwareSerial's method and it does.

If you can reproduce his situation and you have a JTAG probe, I'd really love to check the status when it happens. It bugs me.

ldursw, 2021-02-23:
@X-Ryl669 It's working perfectly now. I just had to apply your changes from `mks_robin_nano35` to `mks_robin_e3` which is the board I'm using.

X-Ryl669, 2021-02-23:
@ldursw What do you mean ? Changing `common.extra_scripts` by `common_stm32f1.extra_scripts` ?

ldursw, 2021-02-23:
Yes, exactly.

X-Ryl669, 2021-02-25:
Should be fixed by #21177 (now merged). Please confirm that the last bugfix solved your bug and in that case, please close it. Thanks!

ldursw, 2021-02-25:
Can confirm it is working as expected. 👍 Thanks a lot for fixing this issue.

PR #21177 (2021-02-25): Let libmaple accept RX_BUFFER_SIZE and TX_BUFFER_SIZE

### Description

Fix #21170 
As requested by @rhapsodyv I'm splitting the refactoring of the serial queue from small fixes I've identified while reworking the code. Here's a one of the fix.

### Requirements

libmaple

### Benefits

Allow to increase the low level serial buffer size so a complete line can be stored in the buffer before the queue's is able to consume it, instead of dropping it. Long lines are caused by arc command and are longer than the default's libmaple's 64 chars buffer.

### Configurations

N/A

### Related Issues

#21 [... cut]
closed 2021-02-25 by ldursw