MarlinFirmware/Marlin #21173: [FR] detect updated Firmware

wuast94, 2021-02-23:
We Talked about this a bit on Discord and i think its worth opening an issue for this.

Actual things get stored in eeprom wich makes problems from time to time, i first get over it on Z Alignment where probing fails becouse the probing positions get stored in eeprom.
So after flashing instead of using the new values when i align the z axis the first one stored in eeprom was used
also the xyz offsets are stored the same way and many other values too.

We could implement a build time variable that is compared at startup to inform the user or ovveride changed/all values?
i mean we dont have much that is user specific that can be safely stay as it is, ubl mesh and such thinks could be stored but all thinks i declare and set in firmware should be overriten when i flash new firmware. becouse this is what i suspect

This issue is more about tracking on this behavior :) sry when this is the wrong place :)

thisiskeithb, 2021-02-24:
Have you seen https://github.com/MarlinFirmware/Marlin/projects?

wuast94, 2021-02-24:
I didn't so thanks :) 

Bit I didn't think that this is addressing the same issue I described or I miss understand his new implementation. Is there a way to link this issue to this Project? I thinks it's clever to Adress this problem there so it could be implemented that changed "features" in New firmware automatically overwrite the eeprom part :)

thisiskeithb, 2021-02-24:
> I miss understand his new implementation. 

No, I thought they were related and wanted to make sure they were compatible/efforts weren’t duplicated anywhere.

> Is there a way to link this issue to this Project?

Possibly? That’d have to be done by Thinkyhead if it could be done, but I’m not familiar with the Projects feature.

organizedkaos, 2021-03-03:
Generally when I flash a new version of Marlin I first do M502 then M500 on the first boot.

Maybe marlin can add a text file to sdcard (or other readable location) that has the name "reset" or something.  The file can be added when a new firmware is flashed and then deleted/renamed after first boot from new flash.

On first boot from new flash if that file exists run M502 and M500 to reset/restore eeprom to new firmware configured defaults.  Then delete/rename the "reset" file.  Maybe?

stef-ladefense, 2021-03-05:
Hi all,
yes it's not the same as "#define EEPROM_AUTO_INIT"
on the other hand we must find a way to detect a new flash even if it is the same version of the firmware!
in fact, a firmware flash resets the values in eeprom before reading them

thisiskeithb, 2022-05-10:
Use `EEPROM_INIT_NOW` to automatically initialize EEPROM after a new firmware install.