MarlinFirmware/Marlin #21181: [BUG] Temperature errors cause heaters only to be disabled with USE_BEEPER defined

Yveaux, 2021-02-24:
### Bug Description

The loud_kill() function (https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L869) that gets called in case of temperature errors (e.g. thermal runaway) only calls thermalManager.disable_all_heaters() when USE_BEEPER is defined.

### Configuration Files

Bare code

### Steps to Reproduce

Code review

**Expected behavior:**

thermalManager.disable_all_heaters() should be called in all cases and should not depend on defining USE_BEEPER

ManuelMcLure, 2021-02-24:
`loud_kill()` unconditionally calls `kill()` which unconditionally calls `thermalManager.disable_all_heaters()`.

Yveaux, 2021-02-24:
So luckily it's not critical then, but it doesn't make sense and is confusing to call it explicitly when USE_BEEPER is defined.

ManuelMcLure, 2021-02-24:
You always want to disable the heaters as early as possible, so it makes sense to redundantly call `thermalManager.disable_all_heaters()` if `USE_BEEPER` is defined so it doesn't wait until after the beeps to shut down the heaters.
Without `USE_BEEPER` enabled the redundant call is not needed since it will immediately drop into `kill()` which calls `thermalManager.disable_all_heaters()` as the very first call.
I see nothing wrong with the way it's currently being done.

Yveaux, 2021-02-25:
Clear, thanks @ManuelMcLure!

I have one remaining question. My setup uses a Fysetc Mini12864 display on a BTT SKR v1. 4 board, so I defined SPEAKER. When a thermal runaway occurs, I only hear some faint clicks, no tones. 
Is there also a simple explanation for this, or should I file a bug report?

ManuelMcLure, 2021-02-25:
The display you’re using has a buzzer, not a speaker, so you shouldn’t define `SPEAKER` - that’s for devices that can generate different tones.

Yveaux, 2021-02-25:
I tried with buzzer first, but that gave me clicks in all other cases (e.g M300).
Enabling SPEAKER works correctly (including M300), except in case of thermal runaways.

closed 2021-02-26 by rhapsodyv