MarlinFirmware/Marlin #21185: [BUG] Very different ABL results when using different grid size

HubSteven, 2021-02-25:
### Bug Description

I'm using an Ender 3 Pro with an SKR V1.4 turbo board (LPC1769) with TMC2209 drivers and a bltouch clone. When doing bilinear mesh bed leveling in a 3x3 grid I get these results:
[image]
```Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.117 +0.077 +0.142
Recv:  1 +0.007 +0.012 +0.058
Recv:  2 +0.112 +0.112 +0.135
Recv: 
Recv: X:235.0000 Y:227.0000 Z:11.0964 E:0.0000 Count X:18800 Y:18160 Z:4500
Recv: ok
```
When printing using this leveling grid the first layer won't be printed level on the bed.
That's why I tried using a 10x10 leveling grid for better accuracy, I expected the same leveling grid as for a 3x3 grid but with a higher resolution. Instead I got a very different result:
[image]
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8      9
Recv:  0 +0.157 +0.100 +0.150 +0.127 +0.085 +0.130 +0.122 +0.120 +0.105 +0.160
Recv:  1 +0.090 +0.080 +0.107 +0.097 +0.070 +0.115 +0.120 +0.160 +0.102 +0.168
Recv:  2 +0.120 +0.092 +0.120 +0.117 +0.107 +0.110 +0.145 +0.140 +0.120 +0.130
Recv:  3 +0.092 +0.080 +0.105 +0.117 +0.092 +0.117 +0.150 +0.142 +0.125 +0.127
Recv:  4 +0.125 +0.102 +0.112 +0.127 +0.102 +0.102 +0.125 +0.152 +0.150 +0.125
Recv:  5 +0.125 +0.122 +0.132 +0.157 +0.097 +0.170 +0.180 +0.165 +0.165 +0.185
Recv:  6 +0.187 +0.163 +0.175 +0.210 +0.163 +0.185 +0.197 +0.175 +0.178 +0.212
Recv:  7 +0.217 +0.212 +0.235 +0.227 +0.247 +0.237 +0.270 +0.270 +0.275 +0.257
Recv:  8 +0.298 +0.320 +0.315 +0.315 +0.303 +0.303 +0.317 +0.325 +0.298 +0.303
Recv:  9 +0.340 +0.350 +0.365 +0.347 +0.327 +0.352 +0.350 +0.402 +0.375 +0.367
Recv: 
Recv: X:235.0000 Y:227.0000 Z:10.8762 E:0.0000 Count X:18800 Y:18160 Z:4500
Recv: ok
```
When printing using this grid the first layer looks far more worse than with a 3x3 grid.
It seems to me as if every new probing point makes the inaccuracy more worse. That's why the last few probing points in the grid show such a big z offset relative to the middle of the bed (which isn't actually there in real life).

Therefore I did a G30 command on the same point on the bed, from what I understand the Z offset should stay constant:
```
Send: G30
[...]
Recv: Bed X: 195.0000 Y: 215.0000 Z: 0.3600
Recv: X:235.0000 Y:227.0000 Z:11.2500 E:0.0000 Count X:18800 Y:18160 Z:4500
Recv: ok
[...]
Send: G30
[...]
Recv: Bed X: 195.0000 Y: 215.0000 Z: 0.3650
Recv: X:235.0000 Y:227.0000 Z:11.2500 E:0.0000 Count X:18800 Y:18160 Z:4500
Recv: ok
[...]
Send: G30
[...]
Recv: Bed X: 195.0000 Y: 215.0000 Z: 0.3750
Recv: X:235.0000 Y:227.0000 Z:11.2500 E:0.0000 Count X:18800 Y:18160 Z:4500
Recv: ok
[...]
```
Oddly enough the Z offset increases? (although sometimes it slightly decreases but then it increases again after a few probes)

My probe repeatability looks like this:
```
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.392500 Min: 0.377 Max: 0.407 Range: 0.030
Recv: Standard Deviation: 0.008803
```
As you can see the Min and Max values are rather large, just like what I've seen after a few g30 commands.
I'm not sure how the standard deviation is calculated but it seems to correspond with the results from other bltouch users.

If I move the bltouch to another random point on the bed and again send a few G30 commands it seems to remember the z offset from the previous point and it keeps incrementing with every g30 command.

It's as if the z-offset is never reset between z-probing actions unless the z-axis is rehomed.

I've already tried the newest Marlin bugfix release of marlin and the latest stable release, I've also tried the dedicated probe input and de z-min input of the SKR board. This didn't make a difference. The stallguard pin is removed from the XYZ stepper controller to prevent interference with the endstop switches and bltouch.
I've also ruled out electrical interference from the two hot end fans and heating element in the nozzles since I turn these off during ABL.

I've been trying to fix this issue for the past two weeks now so I would really appreciate any help! 

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/6039976/config.zip)

thisiskeithb, 2021-02-25:
Can you enable `EXTRAPOLATE_BEYOND_GRID` and `ABL_BILINEAR_SUBDIVISION` and see if it improves?

Also, below is the canned response in regards to collecting more data for leveling issues.

---
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

hamster65, 2021-02-25:
@thisiskeithb: OT: Would it make sense to add M48 Z-probe-test to that template? Cold help to rule out probe issues and machanical problems.

HubSteven, 2021-02-26:
I enabled `EXTRAPOLATE_BEYOND_GRID` and `ABL_BILINEAR_SUBDIVISION` but it doesn't solve the issue with a 10x10 grid:
[image]
I've included the log file
[Uploading serial.log…]()

rhapsodyv, 2021-02-26:
Can you try intermediate grid sizes? Like: 5x5 and 7x7 ?

In a fast view, the issue seems to be with the 3x3 grid getting "similar" points... and when you goes to 10x10 you see that you have some undulations in the bed.
Do you have a glass bed? A PEI sheet?

HubSteven, 2021-02-28:
I'm using a magnetic bed that come with my Ender 3.
These are the results of a 5x5 grid:
[image]
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6
Recv:  0 +0.242 +0.175 +0.178 +0.152 +0.192 +0.195 +0.225
Recv:  1 +0.165 +0.125 +0.112 +0.092 +0.132 +0.110 +0.127
Recv:  2 +0.152 +0.095 +0.110 +0.075 +0.130 +0.130 +0.145
Recv:  3 +0.120 +0.107 +0.120 +0.085 +0.110 +0.092 +0.100
Recv:  4 +0.188 +0.157 +0.173 +0.117 +0.152 +0.120 +0.180
Recv:  5 +0.230 +0.190 +0.225 +0.195 +0.210 +0.205 +0.202
Recv:  6 +0.335 +0.305 +0.335 +0.242 +0.305 +0.262 +0.288
Recv: 
Recv: X:235.00 Y:235.00 Z:11.25 E:0.00 Count X:18800 Y:18800 Z:4500
Recv: ok
```
7x7:
[image]
```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6
Recv:  0 +0.242 +0.175 +0.178 +0.152 +0.192 +0.195 +0.225
Recv:  1 +0.165 +0.125 +0.112 +0.092 +0.132 +0.110 +0.127
Recv:  2 +0.152 +0.095 +0.110 +0.075 +0.130 +0.130 +0.145
Recv:  3 +0.120 +0.107 +0.120 +0.085 +0.110 +0.092 +0.100
Recv:  4 +0.188 +0.157 +0.173 +0.117 +0.152 +0.120 +0.180
Recv:  5 +0.230 +0.190 +0.225 +0.195 +0.210 +0.205 +0.202
Recv:  6 +0.335 +0.305 +0.335 +0.242 +0.305 +0.262 +0.288
Recv: 
Recv: X:235.00 Y:235.00 Z:11.25 E:0.00 Count X:18800 Y:18800 Z:4500
Recv: ok
```
This is with `ABL_BILINEAR_SUBDIVISION` disabled and `EXTRAPOLATE_BEYOND_GRID` enabled.

It seems as the problem worsens with more points.

rhapsodyv, 2021-03-01:
I think it’s the opposite. It’s getting more precise with more points.

comparing the evolution of the images, it’s clear that is becoming more precise to identify points where your bed have level issues.

Do you have problems printing with 10x10 mesh?

HubSteven, 2021-03-01:
When printing with a 10x10 mesh it looks even worse  than with a 3x3 mesh.
I think my issue is caused by z drifting, this not I mechanical issue since I can succesfully print the first layer without abl.

HubSteven, 2021-03-04:
I finally solved my issue, turns out it was a mechanical problem after all.
I used a plastic spacer between my z-axis motor and the frame to make the x gantry run more smoothly on the z axis.
This caused the x gantry to drop under it's on own weight when the z stepper was turned off. I wouldn't drop down when the z axis stepper motor was turned on but apparently it still drop a little after a few z motions.
I removed the plastic spacer and my results look like this now using a 10x10 grid:
[image]
My z-axis stepper also doesn't make as much noise anymore when idle and it doesn't run so hot anymore.
I've already ordered a anti backlash nut for the z axis to have even better results

closed 2021-03-04 by HubSteven