MarlinFirmware/Marlin #21187: [BUG] Marlin bugfix-2.0.x will no longer compile after "Allow Zero Endstops (e.g., for CNC) (#21120)"

lightmaster, 2021-02-25:
### Bug Description

Merged Pull Request #21120 will not compile. If I remove the commit with `git rebase -p --onto 468e437390afdb1de7059e7f9049d0b30c312024^ 468e437390afdb1de7059e7f9049d0b30c312024`, then the bugfix-2.0.x branch compiles without issue.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

Compiles using `mks_robin_nano35_stm32`
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6043430/Marlin.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Pull all commits from bugfix-2.0.x including #21120 
2. Try to compile

**Expected behavior:**

Marlin should compile fine with a seemingly unrelated PR, since it says it's meant for "Cheap Chinese CNC machines".

**Actual behavior:**

Fails to compile with the Errors below

#### Additional Information

```
Compiling .pio\build\mks_robin_nano35_stm32\src\src\module\settings.cpp.o
In file included from Marlin\src\module\../inc/../core/boards.h:24,
Compiling .pio\build\mks_robin_nano35_stm32\src\src\module\stepper.cpp.o
                 from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\../inc/../core/macros.h:183:48: error: pasting "ENA_" and "{" does not give a valid preprocessing token
  183 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                                ^~~~
Marlin\src\module\../inc/../core/macros.h:174:22: note: in definition of macro '_CAT'
  174 | #define _CAT(a,V...) a##V
      |                      ^
Marlin\src\module\../inc/../core/macros.h:183:44: note: in expansion of macro 'CAT'
  183 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                            ^~~
Marlin\src\module\../inc/../core/macros.h:156:29: note: in expansion of macro '_ENA_1'
  156 | #define _DO_3(W,C,A,V...)  (_##W##_1(A) C _DO_2(W,C,V))
      |                             ^
Marlin\src\module\../inc/../core/macros.h:169:28: note: in expansion of macro '_DO_3'
  169 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
Marlin\src\module\../inc/../core/macros.h:170:28: note: in expansion of macro '__DO_N'
  170 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
Marlin\src\module\../inc/../core/macros.h:171:29: note: in expansion of macro '_DO_N'
  171 | #define DO(W,C,V...)       (_DO_N(W,C,NUM_ARGS(V),V))
      |                             ^~~~~
Marlin\src\module\../inc/../core/macros.h:185:29: note: in expansion of macro 'DO'
  185 | #define ENABLED(V...)       DO(ENA,&&,V)
      |                             ^~
Marlin\src\module\motion.cpp:1391:7: note: in expansion of macro 'ENABLED'
 1391 |   #if ENABLED(TMC_HOME_PHASE)
      |       ^~~~~~~
Marlin\src\module\../inc/../../Configuration_adv.h:2692:26: error: token "{" is not valid in preprocessor expressions
 2692 |   #define TMC_HOME_PHASE { 896, 640, 640 }
      |                          ^
Marlin\src\module\../inc/../core/macros.h:450:26: note: in definition of macro 'SECOND'
  450 | #define SECOND(a,b,...)  b
      |                          ^
Marlin\src\module\../inc/../core/macros.h:181:29: note: in expansion of macro 'IS_PROBE'
  181 | #define _ISENA(V...)        IS_PROBE(V)
      |                             ^~~~~~~~
Marlin\src\module\../inc/../core/macros.h:183:29: note: in expansion of macro '_ISENA'
  183 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                             ^~~~~~
Marlin\src\module\../inc/../core/macros.h:175:21: note: in expansion of macro '_CAT'
  175 | #define CAT(a,V...) _CAT(a,V)
      |                     ^~~~
Marlin\src\module\../inc/../core/macros.h:183:36: note: in expansion of macro 'CAT'
  183 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                    ^~~
Marlin\src\module\../inc/../core/macros.h:175:21: note: in expansion of macro '_CAT'
  175 | #define CAT(a,V...) _CAT(a,V)
      |                     ^~~~
Marlin\src\module\../inc/../core/macros.h:183:44: note: in expansion of macro 'CAT'
  183 | #define _ENA_1(O)           _ISENA(CAT(_IS,CAT(ENA_, O)))
      |                                            ^~~
Marlin\src\module\../inc/../core/macros.h:156:29: note: in expansion of macro '_ENA_1'
  156 | #define _DO_3(W,C,A,V...)  (_##W##_1(A) C _DO_2(W,C,V))
      |                             ^
Marlin\src\module\../inc/../core/macros.h:169:28: note: in expansion of macro '_DO_3'
  169 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
Marlin\src\module\../inc/../core/macros.h:170:28: note: in expansion of macro '__DO_N'
  170 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
Marlin\src\module\../inc/../core/macros.h:171:29: note: in expansion of macro '_DO_N'
  171 | #define DO(W,C,V...)       (_DO_N(W,C,NUM_ARGS(V),V))
      |                             ^~~~~
Marlin\src\module\../inc/../core/macros.h:185:29: note: in expansion of macro 'DO'
  185 | #define ENABLED(V...)       DO(ENA,&&,V)
      |                             ^~
Marlin\src\module\motion.cpp:1391:7: note: in expansion of macro 'ENABLED'
 1391 |   #if ENABLED(TMC_HOME_PHASE)
      |       ^~~~~~~
Marlin\src\module\motion.cpp:1391:15: note: in expansion of macro 'TMC_HOME_PHASE'
 1391 |   #if ENABLED(TMC_HOME_PHASE)
      |               ^~~~~~~~~~~~~~
Marlin\src\module\motion.cpp: In function 'void homeaxis(AxisEnum)':
Marlin\src\module\motion.cpp:1729:7: error: 'backout_to_tmc_homing_phase' was not declared in this scope
 1729 |       backout_to_tmc_homing_phase(axis);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mks_robin_nano35_stm32\src\src\module\motion.cpp.o] Error 1
=================================================================================== [FAILED] Took 170.98 seconds ===================================================================================

Environment             Status    Duration
----------------------  --------  ------------
mks_robin_nano35_stm32  FAILED    00:02:50.978
=============================================================================== 1 failed, 0 succeeded in 00:02:50.978 ===============================================================================The terminal process "C:\Users\light\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

rhapsodyv, 2021-02-25:
Just sent a quick fix

thisiskeithb, 2021-02-25:
Fix has been merged. Closing.

closed 2021-02-25 by thisiskeithb

lightmaster, 2021-02-25:
Won't have a chance to flash it until tomorrow to make sure it works completely, but it does now compile without error after #21188

PR #21188 (2021-02-25): Follow up for Allow Zero Endstops #21120
### Description

Fix compilation error when TMC_HOME_PHASE is defined. 

### Benefits

Fix #21187

### Configurations

TMC_2208 / TMC_2209

### Related Issues

#21187