MarlinFirmware/Marlin #21189: EXTRUDER_AUTO_FAN set  default hotend will not go past 160  trying to achieve a Max 260c

REVIEWGAMINGMAN, 2021-02-25:
my Marlin version Marlin-1.1.x

hotend fan is 24v

motherboard version Motherboard Creality Version 2.2

REVIEWGAMINGMAN, 2021-02-25:
and Micro Swiss All Metal Hotend Kit

thisiskeithb, 2021-02-25:
This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

**This includes attaching your configuration files, regardless of whether you think they are involved.**

For reference, these are the instructions included in the template:

---
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

---

### Bug Description

_Describe the bug in this section._

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

_Describe the steps needed to reproduce the issue._

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

_Describe what you expected to happen here._

**Actual behavior:**

_Describe what actually happens here._

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

closed 2021-02-25 by thisiskeithb