MarlinFirmware/Marlin #21198: [BUG] M355 reporting brightness (on SKR GTR 1.0)

viappidu, 2021-02-26:
### Bug Description
If used with #define CASE_LIGHT_USE_NEOPIXEL M355 always report ON or OFF but not the actual brightness.
(So the latest release of BTT TFT35 V3 had some issue working out the value for caselight)

### Configuration Files
Latest bugfix-2.0.x
- Configuration_adv.h
#define CASE_LIGHT_ENABLE
#define CASE_LIGHT_PIN     NEOPIXEL_PIN
#define CASE_LIGHT_USE_NEOPIXEL

- .pins_BTT_GTR_V1_0.h
#define NEOPIXEL_PIN    PF13

### Solution

diff --git a/Marlin/src/gcode/feature/caselight/M355.cpp b/Marlin/src/gcode/feature/caselight/M355.cpp
index 12ae5cff1d..c943ab20b0 100644
--- a/Marlin/src/gcode/feature/caselight/M355.cpp
+++ b/Marlin/src/gcode/feature/caselight/M355.cpp
@@ -61,7 +61,7 @@ void GcodeSuite::M355() {
     SERIAL_ECHOLNPGM(STR_OFF);
   else {
     #if CASELIGHT_USES_BRIGHTNESS
-      if (PWM_PIN(CASE_LIGHT_PIN)) {
+      if (TERN0(CASELIGHT_USES_BRIGHTNESS, TERN(CASE_LIGHT_USE_NEOPIXEL, true, PWM_PIN(CASE_LIGHT_PIN)))) {
         SERIAL_ECHOLN(int(caselight.brightness));
         return;
       }

viappidu, 2021-02-26:
(Had problems properly pasting the code...)

thisiskeithb, 2021-02-26:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

closed 2021-02-26 by thinkyhead