MarlinFirmware/Marlin #21202: [BUG] (short description)

remfpage, 2021-02-26:
When pressing preheat PLA part cooling pan  start at high speed. HotEnd fan don't start.
Changing code E0 from -1 to FAN1_PIN. The HotEnd Fan  start on desire temp but part cooling fan still start at high speed on given temp. when staring printing. the Part-Cooling Fan stop and runs on 2nd layer.

#define E0_AUTO_FAN_PIN FAN1_PIN
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define E5_AUTO_FAN_PIN -1
#define E6_AUTO_FAN_PIN -1
#define E7_AUTO_FAN_PIN -1
#define CHAMBER_AUTO_FAN_PIN -1

#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed
#define CHAMBER_AUTO_FAN_TEMPERATURE 30
#define CHAMBER_AUTO_FAN_SPEED 255

ellensp, 2021-02-26:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2021-04-10:
Can you attach your configs? Your `PREHEAT_*_FAN_SPEED` setting is like set to a value other than 0.

thisiskeithb, 2021-09-24:
Your `PREHEAT_*_FAN_SPEED` setting is like set to a value other than 0.

closed 2021-09-24 by thisiskeithb