MarlinFirmware/Marlin #21203: [BUG] New warning generated with 02/25/2021 Bugfix.

bill-orange, 2021-02-26:
Bug Description: compilation is successful with BugFix but with this warning at the beginning of the compile. 

`Can not remove temporary directory `C:\Users\William\Documents\Marlin\Marlin\.pio\build`. Please remove it manually to avoid build issues`

Manually deleting the build folder makes no difference.  Compiling for a different microcontroller makes no difference.  Compilation with 2.0.x does not generate this error.

Configuration Files ( I also included my platformio.ini):

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6050969/Marlin.zip)

Steps to Reproduce:

1. Compile
2. Observe warning
3. Manually delete build and recompile
4. Observe warning

Expected behavior:  No warning indicating a potential defective build.

Actual behavior: Warning as noted

#### Additional Information

One user reported in the Facebook group that this did not occur when using AutoBuild.  I am not an AutoBuild user.

Foxies-CSTL, 2021-02-26:
I have the same issue with last bump (2021-02-26)
[image]

BirkBinnard, 2021-02-26:
I got that error too - fixed it by setting VSCode to operate in Administrator mode. But then I got the Platformio rebuilding cache message that never goes away. Fixed that by deleting everything from the platformio\.cache directory, deleting and reinstalling PlatformIO. That let Marlin my Marlin config compile. 

But when I tried to boot it on my printer I got the dreaded "All Low" error message (I was never able to find out what it means.) So I reverted to an earlier BugFIx version that does compile and boot OK. 

Unfortunately it doesn't work properly, but at least it boots without errors.

rhapsodyv, 2021-02-26:
I find it... it's a marlin config issue that is generating a exception on some internal PIO method...

Here:
```
Invalid '/Users/victor/Development/Marlin/platformio.ini' (project configuration file): 'No section: 'LERDGEX''
```

I'm sending a PR to fix that

rhapsodyv, 2021-02-26:
Fixed: #21205

ellensp, 2021-02-26:
closing as is merged

closed 2021-02-26 by ellensp