MarlinFirmware/Marlin #21204: [BUG] Random pauses on print that need user input to resume

wolfstrassen, 2021-02-26:
### Bug Description

Ender 3 pro with 4.2.2 board
Printer randomly pauses, usually with large prints (10+ hours), but it has happened on 1 hour or less. The printer pauses and when I press the knob, the print resumes. If I don't notice it soon enough, a large blob of plastic ozzes from the nozzle

During the time the print was paused the bed and extruder did not cool down, so this is a fire hazard.

This was doing by printing using an SD card, I have not tried with USB because my cat most certainly will yank the cable from the closet where the printer is if I let it several hours with no supervition.

If I retry the same g-code again, the printier do not stop at the same layer or does not pause at all.

This behaviour happens about 30-40% of my large prints (6 hours or more), and I've seen it both with `bugfix-2.0.x` and `release-2.0.7.2` branches but it doesn't happens with the creality precompiled .bin

The G-Code is generated with Cura, and the the bug has happened with 0.08, 0.12, 0.2 and 0.28 layer settings
### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6051659/Configuration.zip)

I pretty much use stock configuration besides enabling MESH_BED_LEVELING and DOGM_SD_PERCENT

wolfstrassen, 2021-02-27:
I've uploaded a video to youtube with the behaviour [here](https://youtu.be/sdaXIiCTFwQ), the screen shows weird characters until I press the knob (h°z`)

XDA-Bam, 2021-03-07:
This may be a Marlin bug, but it could also be related to other things. Electromagnetic interference in the display cable comes to mind. You could check how it is routed and make sure that all fan and heater cables are far enough away. Also, how big is your compiled firmware (in kB and/or % of ROM)?

wolfstrassen, 2021-03-07:
> This may be a Marlin bug, but it could also be related to other things. Electromagnetic interference in the display cable comes to mind. You could check how it is routed and make sure that all fan [...]

I've checked the cables. I'm an electronic engineer with a master in electrical engineering and I think it's not EMI, but I'm working from home and I don't have all the tools I'd like to narrow this down, I'm kind of suspicious of the power supply but I haven't monitored the voltage long enough to be sure. I've tried several times now with USB instead of the SD card, the problem still happens but with USB the print continues on its own after about 10 second leaving a small blob of plastic, most of the time I can sand it out and at least its not a fire hazard anymore.

RAM:   [==        ]  17.5% (used 11488 bytes from 65536 bytes)
Flash: [====      ]  35.5% (used 185876 bytes from 524288 bytes)

I also can try to track this bug on my own but Marlin is a big code... I would need some pointers, like for example where to enable relevant debugging

XDA-Bam, 2021-03-07:
OK, RAM/flash isn't critical at all - I was somehow assuming that the 4.2.2 boards are still ATMega, but they are in fact 32 bit ARM and have plenty of flash.

I just read that it does not happen on the original Creality FW, so EMI and PSU are unlikely - but I would agree that it makes sense to check the PSU/voltage stability nonetheless. Overall, is sounds more like a problem with the serial connection or - possibly - the STM watchdog triggering.

I think it could be helpful to keep a serial console running while printing to check the output in case the printer stalls. Did you try that?

wolfstrassen, 2021-03-07:
> OK, RAM/flash isn't critical at all - I was somehow assuming that the 4.2.2 boards are still ATMega, but they are in fact 32 bit ARM and have plenty of flash.
>
> I just read that it does not happen on [...]

Yes, I also think that PSU/EMI is unlikely, but just because I didn't see it happen on the creality FW it doesnt mean it couldn't happen. I doubt is a problem with the serial connection, because the problem happens both with SD card and the USB. I haven't keep a serial console while running, mainly because I have space issues and I need my PC to work, but I think I can cramp my PC and printer on my work desk and try. 

I'm sorry if this is a dumb question, but how do I enable a serial console while Cura has a hold of my serial port? or it is only possible while printing from the SD card?

XDA-Bam, 2021-03-07:
That should only be possible while printing from SD.

"Serial connection" in this case did not refer to USB. The LCD is, AFAIK, internally adressed using the serial protocol - just via a different port than e.g. the USB connector.

wolfstrassen, 2021-03-07:
> That should only be possible while printing from SD.
> 
> "Serial connection" in this case did not refer to USB. The LCD is, AFAIK, internally adressed using the serial protocol - just via a different port than e.g. the USB connector.

wolfstrassen, 2021-03-07:
I'll do my next prints with serial terminal and let you know if I found something

wolfstrassen, 2021-03-20:
Update: Since I enabled MARLIN_DEV_MODE and POSTMORTEM_DEBUGGING in Configuration_adv.h I haven't had any random pauses

CarlSugoi, 2021-04-05:
I've been experiencing the same issue for weeks now, tried the latest bugfix too. 

Tried;
- tried multiple SD cards
- enabling MARLIN_DEV_MODE and POSTMORTEM_DEBUGGING. 

Same symptoms, the print would randomly stop with extruder and bed hot, leaving a blob of filament. Random letters or symbols would show up, and print will resume when pressing the button. Can happen multiple times during a print or none at all.

wolfstrassen, 2021-04-06:
I've have the same issues again, the MARLIN_DEV_MODE  and POSTMORTEM_DEBUGGING  seems to have lessen the issues, but they still happen. Sometimes the text is random gibberish and sometimes it shows a message that says click to resume or something along those lines

thisiskeithb, 2021-04-06:
Several related PRs have been merged since this was opened that may have addressed the issue.

For those still experiencing issues, please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, you can try shielding the LCD cable or picking up an [EMI filter board](https://www.th3dstudio.com/product/lcd-emi-filter-board-for-skr-and-creality-boards/) to reduce noise in the LCD cable.

thinkyhead, 2021-04-15:
If someone can isolate the day when the issue first appears, that should give a very strong clue to the problem. How do we quickly find that date? By repeatedly testing Marlin from the date half-way between the time we know the bug is absent and the time we know the bug is present. Those dates are now months apart. But after only 4 or 5 tests, those dates will be only days apart. GitHub helpfully provides the ability to download Marlin at any point in time from the "Commits" tab above.

CarlSugoi, 2021-04-19:
Yeah at this point, after trying all the suggestions from multiple forums regarding the same issue, I've tried installing the latest TH3D unified firmware to see if it was a firmware issue or a hardware issue. After switching I haven't had the issue again, and I've done multiple long prints (12hrs+) each. 

The problem with this issue is that it's so random, that it's very hard to find the cause.

EtStLouis, 2021-04-22:
Hello, I did the same, TH3D firmware has not given any issue yet after 3
days. I got 2 errors before, first pause on print like you but another one
was my BLTouch, 1 read every 20 or 30 was not working. I changed the wiring
to the mobo of it by using the Z end limit slot. Everything is solved but I
don't know if they were one and unique issue as I did both repair at the
same time. I'm wondering if you also have a BLTouch and if it could be a
voltage issue for Marlin?

CarlSugoi, 2021-04-25:
Yes I do actually have a BLTouch installed as well. I know that setting up TH3D lets you choose which way the BLTouch is wired, either all 5 pins, or by separating them into 3 + 2 pins. I don't recall ever choosing this option in the bugfix.

I wired mine in the 3 + 2 configuration, I'm wondering if yours is the same?

EtStLouis, 2021-04-26:
Hello,
Now it is. First it was set with 5 pins. No bug since those 2 changes.

Shortybw, 2021-05-10:
Same issue. I'm also using the BLtouch with it and connected to an octoprint Pi to monitor it. the serial shows:

"2021-05-10 17:59:40,967 - Recv: echo:busy: paused for user
2021-05-10 17:59:42,646 - Recv:  T:215.00 /215.00 B:64.00 /65.00 @:78 B@:127
2021-05-10 17:59:42,967 - Recv: echo:busy: paused for user
2021-05-10 17:59:44,646 - Recv:  T:215.00 /215.00 B:65.00 /65.00 @:77 B@:29"

consistently on my log. I'm updating the firmware with the above. Will update in a few days.

Shortybw, 2021-05-11:
> @XDA-Bam yes I’m currently having the problem where is randomly pauses and the only way to get it going again is press the knob. Since updating the firmware last night to what was suggested above, it happened again... just woke up to stuck on layer 100ish so it been idle at 215 degrees for roughly 6 hours. I’ll will pull all the debugging logs I have and reply to this. I’m also attaching a picture of the status at the bottom of the screen. This time it just said “D” before it’s been random symbols and letters.

https://github.com/Shortybw/Log-files-for-ender-3-pro/blob/main/serial%20short%20log.txt

attached is the link of a shortened version of when the pause occurred from the serial log last night. I left the very beginning in, what it says just before the pause (22:23:08) and the pause itself (22:23:11). I also showed what happened when I got up around 6am and pressed the knob again to resume (06:09:00).  I don't think this is a serial problem. Is there anymore logs I can pull to help that you are aware of?

wolfstrassen, 2021-05-13:
I still have this issue sometimes, but way less than before. (don't know what changed) I experienced the issue before installing BLTouch, so I don't think it is a factor. Also, if I remember correctly, my logs did not show the "paused for user"  that @Shortybw logs do

take2pm, 2021-06-16:
I have exact same issue on my Creality Ender 3 Pro V4.2.2 (aka 1.5)
Sometimes I'm able to successfully print 20h prints, and sometime it pauses on a very short prints - I haven't observed any specific rule around that.
My only impression is that with recent releases > 2.8.1 it is not happening that often.

And this is example communication log when that happened and that goes on until I press the knob on the printer.
```
2021-06-16 18:03:33,827 - Recv: ok
2021-06-16 18:03:33,829 - Send: N137113 G1 X127.657 Y98.13 E2849.99595*105
2021-06-16 18:03:33,846 - Recv: ok
2021-06-16 18:03:33,848 - Send: N137114 G1 X125.405 Y97.866 E2850.0412*106
2021-06-16 18:03:33,879 - Recv: ok
2021-06-16 18:03:33,880 - Send: N137115 G1 X123.151 Y97.583 E2850.08653*80
2021-06-16 18:03:35,530 - Recv:  T:200.00 /200.00 B:60.00 /60.00 @:69 B@:35
2021-06-16 18:03:35,889 - Recv: echo:busy: paused for user
2021-06-16 18:03:37,529 - Recv:  T:200.00 /200.00 B:60.00 /60.00 @:69 B@:35
2021-06-16 18:03:37,889 - Recv: echo:busy: paused for user
```

Kfillo, 2021-06-29:
I have the same issue.  MKS GEN-L Rev 2.0m and MKS_LCD12864B.  I found that if I disable the lcd by commenting out the line "#define MKS_LCD12864B" will resolve the issue.  Just disconnecting the display does not solve the problem.

AddChickpeas, 2021-07-31:
I am also having the issue of _momentary_ pauses when using Octoprint with Ender 3 pro with 4.2.2 board. I am not sure if this is the same issue as described here since the initial report was permanent pauses when printing from SD card. 

It might be the same since octoprint could just be continuing to send lines and overriding the pause and forcing it to continue. I did not see any "paused for user" lines in the serial output like some of the previous commenters. 

I am having it happen at least once per print, regardless of how small using 2.0.9.1. It freezes just long enough for for a nice blob to form before continuing. A quick 20 minutes print might end up with 2-3 pauses/blobs on the outside. 

I switched to the latest TH3D build and have not had any issues so I believe it is firmware related. It seems to be a Marlin fork so I'm not sure what the difference is. Could just be they are based on an older version that did not have the issues.

kaustubhtapke, 2021-08-14:
I am also having the same issue. I have Ender 3 pro with 4.2.2 board as well. Print stops randomly and gets resumed after pressing the knob. I have the "bugfix 2.0.x version" installed. Any recent updated in the Marlin regarding this? or any other solution?

MatthewMazaika, 2021-08-25:
Today was my first pause since I got my printer running - I was about 2 hours into a 3 hour print (5th print in 24 hours) - unfortunately, even though I have Octoprint set up, it wasn't set up to record the serial log, so I missed the initial message and then didn't jot down the exact message in the terminal - though it did at least say it was paused waiting for user input and I was able to recover the print by simply pressing the knob

I downloaded and compiled Marlin back in June and according to octoprint I'm running: `Marlin 2.0.9.1 (Jun 30 2021 02:10:23)` on the Ender 3 Pro with the 4.2.2 board

AddChickpeas, 2021-08-25:
I recently upgraded my board to a SKR Mini V2 and haven't had this issue since so it seems specific to the 4.2.2 Creality board (or potentially 4.2.X, but I haven't seen any reports from an upgrade to the silent board).

@MatthewMazaika @shriGanesha I was able to workaround the issue by using TH3D Studio's firmware before upgrading. It's based on Marlin and currently uses the 2.0.8 version so you could also try downgrading to 2.0.8 and seeing if it resolves the issue before giving TH3D's firmware a shot.  

The 4.2.2 board is not consistent between printers and seems to ship randomly with different stepper drivers. I'm curious if that inconsistency could somehow be playing into things? I'm honestly pretty clueless on firmware/motherboards, but I wouldn't be surprised if something like that could introduce issues. 

Would you guys be able to check what drivers your board has? According to [this article on TH3D Studio,](https://www.th3dstudio.com/hc/product-information/3rd-party-control-boards/creality-boards/creality-v4-2-x-board-driver-codes/) there is a mark on the SD card slot on the motherboard that indicates which stepper drivers were used: 

> > The letters written on the card reader slot (with marker) are different depending on which stepper driver it has. These are for the ones 4.2.X board with STM32F103RET6 controller:
> 
> C = HR4988
> E = A4988
> A = TMC2208
> B = TMC2209
> H = TMC2225

Mine personally had the HR4988, which is like a budget version of the A4988, which is already a cheap driver.

MatthewMazaika, 2021-09-19:
I'll check my stepper revisions - I definitely have the 4.2.2 mainboard

Another note that seems to correlate with _machine runtime_ or _total print time_ and not _individual print time_ was that after a reboot I was able to get through a 23 hour print without a pause, but, after a few days of successful prints, I hit the pause about 45 mins into a 2 hour print

I often power my printer down either because my BLTouch decides not to engage or I'm done printing until my next great idea, so it also possibly explains why I've only seen the pause issue twice since I flashed my board back in June

Buxtonion, 2022-02-20:
Hi all, I was wondering if anyone determined a solution to this?  I have been dealing with this issue also, in someway it is good to know I am not alone in this issue @wolfstrassen are you still having this issue?

AddChickpeas, 2022-02-20:
I was having issues with temporary pauses as opposed to permanent, but
switching to TH3D's firmware was the workaround I used until it went away
when i upgraded to an skr mini v1.3.

I haven't seen any other discussions of it with further updates on a fix.

wolfstrassen, 2022-02-21:
> Hi all, I was wondering if anyone determined a solution to this? I have been dealing with this issue also, in someway it is good to know I am not alone in this issue @wolfstrassen are you still having this issue?

Yes. I'm still having this issue. It happens somewhat less now tough. Don't know why. I'd say that 1 in 5 long prints I have this problem

Kfillo, 2022-02-21:
Still have the same issue.  Happens several times a print to me.  I have
gotten around the issue with the Terminal Response plugin for octoprint. (
https://github.com/xia0/OctoPrint-TerminalResponse ).  Each time M1 Stop is
issued (whereever that comes from) I send M108 to resume the print.

descipher, 2022-02-21:
If it works on TH3D then its not likely a Marlin bug, TH3D uses Marlin upstream releases. Currently TH3D is at 2.0.9.3. 

I would suspect noise and LCD timing first, then config options that can interplay with it. The main difference with TH3D is config options.

Eddiiie, 2022-02-24:
Wow so this just started happening to me...   I have used Marlin for about 5 years, many different printers, never had this problem.
But this is a new printer, new main board.   

And DAMN it stops the printing  what!!!!     Have to use bug fix because that is the only version that supports my board...

Here are the specifics -

Ender 3  using BTT SKR Mini v3  (TMC 2209 drivers)
BTT 3.5 TFT
Creality 3D Touch
Octoprint

I don't even have to be printing.  The printer can be sitting there idle and it appears.   
Before I changed to this mainboard I was using BTT SKR2  Rev B  with no issues.   

Attached is my configuration.h and configuration_adv.h
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/8137308/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/8137312/Configuration_adv.h.txt)

Bugfix version  02000903   <-- taken from config

I am going to open an issue with Marlin to and ask how to troubleshoot

thisiskeithb, 2022-02-25:
For anyone running BigTreeTech's TFT firmware, update to the latest version. There was a bug where commands were being truncated and would turn into `M0`/`M1`, so users were disabling those commands in Marlin: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295

You should also add an extra ground wire to the LCD & your frame, add shielding to your LCD/data cables, re-route them to avoid electrical interference, or pick up an EMI filter board.

I still have my doubts that this is a software bug within Marlin because it would affect everyone running this hardware & it doesn't.

Eddiiie, 2022-02-25:
Thank you.  I will update soon as this 10 hour print job (now 15 hours due to pause) is done.
Now that I think about it, I did not connect the TFT until I changed main boards.  
So, yeah, not Marlin .  

Forgot to check for new TFT firmware.
Will report back.

wolfstrassen, 2022-02-25:
> For anyone running BigTreeTech's TFT firmware, update to the latest version. There was a bug where commands were being truncated and would turn into `M0`/`M1`, so users were disabling those commands in Marlin: [bigtreetech/BIGTREETECH-TouchScreenFirmware#2295](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2295)
> 
> You should also add an extra ground wire to the LCD & your frame, add shielding to your LCD/data cables, re-route them to avoid electrical interference, or pick up an EMI filter board.
> 
> I still have my doubts that this is a software bug within Marlin because it would affect everyone running this hardware & it doesn't.

I don't think this is a 100% firmware issue, but I think is something that the firmware could handle (i'm not saying is easy sinse it happens at random intervals). A little interference from the LCD should be filtered and not trigger a halt on the printer that can be resumed by a button, also this is not a normal pause since the pause command parks the nozzle and this problem don't. At this point I learned to live with it, but it would be cool if at least it show a proper message instead of giberish when the error happens

Kareareawbg, 2022-02-25:
The same issue also happens on my SKR Mini E3 V3 with a tft35 V3.0.1

Will print perfectly from MB SD Card Slot as long as emlulated 12864 LCD is
used with no devices plugged into tft panel.
With TFT active, random M1, M0, M109 pauses and stops. Print halts with
heater and bed still powered up. Infrequently the TFT completely freezes.

Running freshly compiled drivers for mainboard, and 1 week old ones for
tft.
Using cat6E cable between mb and tft.

Will try tft firmware fix.
Failing that use th3d's drivers

thisiskeithb, 2022-02-25:
> Failing that use th3d's drivers

TH3D’s firmware *is* Marlin, but if he’s patched a confirmed bug, we welcome a PR to fix it.

houseofbugs, 2022-02-25:
@thisiskeithb I have not patched anything like that specifically. This sounds like hardware issues, not firmware. Hardware very likely being either EMI issues and/or improperly built printers.

descipher, 2022-02-25:
It could be an issue with prompting code and a display serial command problem, if a prompt occurred and was not kept in a modal foreground it would be missed.

Eddiiie, 2022-02-25:
I have an update.  Installed new firmware for BTT E3 35 TFT.   Before doing so I let the running print job complete.  During this time the event log on the TFT showed many different errors, SD CARD IN / OUT (or whatever it says saying it was there then it wasnt), during the print, messages would come up saying invalid commands..    Things that sure could be EFI  but if you look closely it is not line noise 'garbage' in the message from the old days with modems  lol.   It is what @thinkyhead thisiskeithb said it was.  -- An issue with the BTT firmware corrupting messages and that corruption was being translated into 'whatever' ...  Like a buffer overflow attack.  heh    Anyway, I am 8 hours into a print now, 10 left.   NO MESSAGES YET.  

As for EFI, I can tap into the serial line and log the whole stream to help prove / disprove this theory.   LMK

Thank you for Marlin  I don't know how you people keep on top of all this.

thisiskeithb, 2022-03-19:
https://github.com/MarlinFirmware/Marlin/pull/23925 promises to help with EMI noise if you're following this issue & would like to try it out.

I'd be interested to know if features that require a double-click (`DOUBLECLICK_FOR_Z_BABYSTEPPING`, `MOVE_Z_WHEN_IDLE`, etc.) still work when holding the encoder down instead of double-clicking.

thisiskeithb, 2022-03-21:
https://github.com/MarlinFirmware/Marlin/pull/23925 has been merged.

closed 2022-03-21 by thisiskeithb