MarlinFirmware/Marlin #21206: [BUG] Duplication modes are not handled in M106/M107

Kachidoki2807, 2021-02-26:
Hello,

Please, be indulgent with my English, and let me put some context.
With the awesome MakerFr community, we built a new i3-RDX printer, which is an IDEX one. We are currently in a beta phase and setting it up, and are facing to a bunch of issues inherent to the IDEX's specificities that are not spotted yet.
We are in the burden of solving them one by one. Most of them are on the slicer side, so we have a custom compiled PrusaSlicer to better handle the IDEX, especially for the fan management in Dual mode.

### Bug Description
One of the issues we are facing is while using the duplication modes (duplication/mirror). Everything seems to be smoothly handled regarding the axis and the temperatures, but there is one missing point: the proper fan management.
Indeed, when the M106 (M107 also) G-Code is parsed, the speed is properly applied to FAN0, but not duplicated to FAN1 when duplicating.

**Expected behavior:**

The both tools should have the same behavior regarding the fan speed.

**Actual behavior:**

Only the T0 fan is started up. The T1 fan stay off.

#### Additional Information

I looked everywhere over the internet, and found no mention to something similar. However there are already some IDEX printers on the market running under Marlin. IMHO, that's very weird to discover this point only now. This is not a feature request since the duplication modes are partially functional.

### Fix proposal
What I have done to solve this issue, is to add these code snippets to the M106:
```cpp
    #if ENABLED(DUAL_X_CARRIAGE)
      if (dxc_is_duplicating())
        thermalManager.set_fan_speed(1, speed);
    #endif
```
And M107:
```cpp
  #if ENABLED(DUAL_X_CARRIAGE)
    if (dxc_is_duplicating())
      thermalManager.set_fan_speed(1, 0);
  #endif
```
Both right after the original thermalManager.set_fan_speed(...);

Is this the proper way to fix this issue? Should I create a pull request? I'm a developer but new to this workflow.

Thanks.

thisiskeithb, 2021-02-27:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

rhapsodyv, 2021-02-27:
The fix was merged. Thanks!

closed 2021-02-27 by rhapsodyv