MarlinFirmware/Marlin #21220: Neopixel PRINTER_EVENT_LEDS - Robin Nano 1.2

galluccib, 2021-02-28:
Not sure if I'm missing something here.. Enabled everything in the config and when I power up the printer its goes through each of the colors fine. If I try and set a color via M150 it works fine. When I go to print I noticed the lights are coming on at all so not sure if I missed something or there is a open bug. 

To get it working platformio was updated to use - NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel

Running bug fix from 2/26

#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     LED_PIN       // LED driving pin
  #define NEOPIXEL_PIXELS 10      // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  //#define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. 
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup
  // Use a single NeoPixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

Configs attached.

Thanks in advanced...

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6056288/Configurations.zip)

ellensp, 2021-02-28:
only works is you using wait variant of heating gcode ie M109 and M190

galluccib, 2021-02-28:
Is there a doc that outlines what I should be using in the gcode? Found a bunch of folks talking about enabling in config but haven't seen anything stating it had to be added to gcode... 

Thanks,

hakimio, 2021-03-01:
You can just use [M150](https://marlinfw.org/docs/gcode/M150.html) in your start/end gcode:
```
M150 P R255                ; Set NeoPixel Red
M150 P R255 B255 U255      ; Set NeoPixel White
```

thisiskeithb, 2021-03-06:
Closing since NeoPixels are not supported on STM32F1 (there's a sanity check you have to disable) and as you've noted, you need to use a patched NeoPixel fork.

closed 2021-03-06 by thisiskeithb