MarlinFirmware/Marlin #21222: [BUG] (short description)

kmelihk, 2021-02-28:
### Bug Description

Unable to compile for STM32F103RET6_creality

### Configuration Files

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6056416/configs.zip)

I've also modified platformio.ini line below:

default_envs = STM32F103RET6_creality

### Steps to Reproduce

When I hit compile, with the current release version, I get the message output below:

`
Processing STM32F103RET6_creality (platform: ststm32@~6.1; board: genericSTM32F103RE; framework: arduino)
...lines without errors omitted
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:85:4: error: #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
   #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:78:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:55:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:85:4: error: #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
   #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:78:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:55:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../HAL/./STM32F1/HAL.h:85:4: error: #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
   #error "SERIAL_PORT must be -1 or from 1 to 5. Please update your configuration."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/../pins/pins.h:78:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:34,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:51:4: error: #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
   #error "Oops! Set MOTHERBOARD to an STM32F1-based board when building for STM32F1."
    ^~~~~
Marlin\src\HAL\STM32F1\../../inc/../pins/ramps/pins_RAMPS.h:55:4: error: #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
   #error "Oops! Select 'Arduino/Genuino Mega or Mega 2560' in 'Tools > Board.'"
    ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\MarlinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/../HAL/HAL.h:26:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
[... 235 lines cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\timers.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\watchdog.cpp.o] Error 1
================================================================================================================================================ [FAILED] Took 7.74 seconds ================================================================================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:07.744
`

If I try the bugfix version with the same configuration and platformio.ini lines changed, it's a much shorter run:

`
Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
Error: your selected build environment 'STM32F103RET6_creality' is not compatible with MOTHERBOARD=BOARD_RAMPS_14_EFB in Configuration.h. Please use one of compatible build environments for this board: mega1280,mega2560
================================================================================================================================================ [FAILED] Took 0.66 seconds ================================================================================================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:00.660
`

Thank you for reading.

thisiskeithb, 2021-02-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](https://marlinfw.org)
- RepRap.org [Marlin Forum](https://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Discord server](https://discord.gg/n5NJ59y)

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-02-28 by thisiskeithb

kmelihk, 2021-02-28:
Thank you, I did end up figuring it out, I had configuration files on the root folder instead of Marlin. I searched the web and not much came up, other than one person with same issue on reprap forum.