MarlinFirmware/Marlin #21223: [FR] Create audible notification in case of thermal runaway and using SPEAKER

Yveaux, 2021-02-28:
### Description

Currently, when USE_BEEPER is defined, thermal runaway causes the function loud_kill() in temperature.cpp to produce 20 beeps.
When SPEAKER is defined, no sound is produced.
It would be nice some sound would be generated in case of temperature issues and SPEAKER defined.

#### Additional Information

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/temperature.cpp#L848

slowbro, 2021-02-28:
It may be worthwhile to simplify the speaker/buzzer terms in config & code too.. looking around, I see `HAS_BUZZER`, `USE_BEEPER`, `HAS_CHIRP`, `SPEAKER`, `LCD_USE_I2C_BUZZER`, and the pin seems to always (sans i2c) be `BEEPER`.

thinkyhead, 2021-07-21:
> When SPEAKER is defined, no sound is produced.

So it's a bug. All code that makes sound should be using the same method. If it's not, that is a bug. The `SPEAKER` may rely on a Tone interrupt to be enabled. If the kill screen is turning off interrupts before playing the alarm sound, that is a bug.