MarlinFirmware/Marlin #21226: [BUG] BLTouch not compensating the tiled bed

kuroro20, 2021-02-28:


hakimio, 2021-03-01:
Did you save the mesh (`M500`) and add `M420 S1` after `G28` to your start gcode?

ellensp, 2021-03-01:
Please attach your start gcode

Wangke98, 2021-03-01:
1------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;ellensp&quot;<notifications@github.com&gt;
发送时间:&nbsp;2021年3月1日(星期一) 晚上7:11
收件人:&nbsp;&quot;MarlinFirmware/Marlin&quot;<Marlin@noreply.github.com&gt;;
抄送:&nbsp;&quot;Subscribed&quot;<subscribed@noreply.github.com&gt;;
主题:&nbsp;Re: [MarlinFirmware/Marlin] [BUG] BLTouch not compensating the tiled bed (#21226)

kuroro20, 2021-03-01:
Hello,

Here is my start gcode:
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S160; start warming extruder to 160
G28 ; Home all axes
G29 ; Auto bed-level (BL-Touch)
M420 S1;
G92 E0 ; Reset Extruder
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
; G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode

In fact i tought that G29 was enough, isn't it the case ?

hakimio, 2021-03-01:
Yes, if you want to relevel the bed before every print `G29` is enough, but if you just want to level it once and then use it, you can save the leveling data with `M500` and use `M420 S1` after homing (`G28`) to enable it.

kuroro20, 2021-03-01:
With my Gcode i don't see where i did wrong :(

kuroro20, 2021-03-05:
Is there any news about this bug ?

kuroro20, 2021-03-07:
Hello,
I've done an other test,
while printing i sent M420V
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2
Recv:  0 +0.348 +0.133 -0.042
Recv:  1 +0.278 +0.033 -0.197
Recv:  2 +0.288 -0.015 -0.325
Recv: 
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height 10.00

And still no Bed leveling ?

holocronology, 2021-03-13:
> add `M420 S1` after `G28` to your start gcode

DO you really need this if you have `RESTORE_LEVELING_AFTER_G28` enabled in the firmware?

hakimio, 2021-03-13:
> > add `M420 S1` after `G28` to your start gcode
> 
> DO you really need this if you have `RESTORE_LEVELING_AFTER_G28` enabled in the firmware?

No, you don't.

enderextender, 2021-03-17:
@kuroro20 

Dont use m420 and g29 together.

remove m420 line and test again . Let us know

GdLakeIoT, 2021-04-27:
Hi. I had this problem, G29 does Bed leveling but then when the printer is printing there is no compensation for the unleveled bed.
Las year I compiled marling following  Teaching Tech's [videos](https://www.youtube.com/watch?v=neS7lB7fCww&t=60s) on how to upgrade to Creality 32 bit boards.
Bed Leveling worked perfect. I could see the Z Stepper motor compensating for the bed.
In March 2021 I downloaded the current version of the BugFix 2.x code and recompiled for the Z_Probe_Wizard and the compensation stopped working.
Today I was able to find the ZIP I downloaded on Oct 2 2020, made the changes for the Ender 3 Pro 4.2.2 with BLTouch and once again, the Z Stepper is compensating for the unleveled bed.
I Hope this helps find the solution of this problem.
BUG #19839 is closed and I asume it is the same problem.

GdLakeIoT, 2021-04-27:
BTW, here is the 2 ZIP files.
[Configurations-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/6381045/Configurations-bugfix-2.0.x.zip)
[Marlin-bugfix-2.0.x_021020.zip](https://github.com/MarlinFirmware/Marlin/files/6381047/Marlin-bugfix-2.0.x_021020.zip)

closed 2021-07-06 by github-actions[bot]