MarlinFirmware/Marlin #21227: [BUG] Input knob triggered when bed is heated

Sebastian-Vogt, 2021-02-28:
### Bug Description

I have a Creality Ender 5 and upraded it with an v4.2.7 silent board, BLTouch and a filament runout sensor. After a while i found a configuration that seemed working for all these things, except, that everytime, when i heat the print bed, it triggers the input knob. So it basically presses itself in unregular intervals.
Then i tried if the BLTouch and filament sensor stuff is maybe triggering this behaviour. But also the vanilla configuration produces this bug.
The precompiled version of creality does not do that. But it seems to have a reversed extruder stepper setting and no filament runout sensor.

### Configuration Files

I am on the bugfix-2.0.x branch.
I set the `default_envs` to `STM32F103RET6_creality`.
The configuration files are these: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6056936/Configuration.zip)

### Steps to Reproduce

1. Install marlin on the printer.
2. Head the bed to something like 70°C.

**Expected behavior:**

It should heat and do nothing else.

**Actual behavior:**

It heats up, but it triggers the input knob pressed action in irregular intervals.

#### Additional Information

Here is a video of that:
https://user-images.githubusercontent.com/12019404/109423034-4eff8000-79de-11eb-8f6a-fc45ab25eb43.mp4

hakimio, 2021-03-01:
Noticed the same behavior as well with bugfix branch and MKS Robin Nano v2 board.

ellensp, 2021-03-01:
To me that looks like you have the heated bed wires to close to the lcd wires. Causing electrical interference.

Sebastian-Vogt, 2021-03-01:
I dont think this is the case:
[image]

Sebastian-Vogt, 2021-03-07:
I tried the following things:
I connected the printer via USB and send some commands via Pronterface. When i heat it via Pronterface it does not trigger the button until i do something with the printer button. Then it gets triggered in unregulary again. When i then set the beds temteratur to 0 again (via Pronterface) it stops immediately.
Is there a way to debug a usb connected printer on source code level?

hamster65, 2021-03-07:
Looks like EMI to me, too. But this issue appeared on my printer (LPC-Controller) without any changes to the hardware. But i flashed Marlin before.... Maybe the current versions handle the button differently, in terms of low pass filtering, timing or debouncing etc. so that it is more prone to EMI? Still I think the culprit is in the wiring and ground paths.

militiaspack, 2021-03-09:
did you find a work around or anything for this or would I be better off switching firmware? just got a silent board for ender 5 and having this issue to the point where I can't print :/

Sebastian-Vogt, 2021-03-10:
I just shielded the LCD cable with some EMI shielding tape and it seems to work, but i could not really test it yet. I will let you know.

Sebastian-Vogt, 2021-03-10:
Ok, so for short prints (10min) it works without any issues. Thanks for the help @ellensp and @hamster65. It seems you are right.
Maybe it would be good to add some EMI resistance in the software...

closed 2021-03-10 by Sebastian-Vogt

militiaspack, 2021-03-10:
o btw it's a issue with the board, bought this https://www.th3dstudio.com/product/lcd-emi-filter-board-for-skr-and-creality-boards/ was also told that the foil works but might not as well.