MarlinFirmware/Marlin #21228: G26 starts opposite of origin, Inverted bed mesh, Bilinear

thetailwind, 2021-02-28:
Hello, I've been tracking down a bed leveling issue and I'm noticing some strange behavior. Initially I thought my mesh was inverted but now I think it's rotated 180*

The Setup: CoreXY, SKR 1.4 Turbo TMC2209, bl touch.

Build version : "bugfix-2.0.x" latest pulled yesterday.

I ran a test and I can confirm that the mesh is upside down or rotated:

I got my current mesh and found the worst corner (4,4)

M420 V

Recv: 0 1 2 3 4
Recv: 0 -0.175 +0.057 +0.142 +0.173 +0.230
Recv: 1 -0.023 +0.048 +0.030 +0.072 +0.023
Recv: 2 +0.010 +0.075 +0.012 +0.043 +0.063
Recv: 3 -0.035 +0.035 +0.050 +0.152 +0.148
Recv: 4 -0.070 +0.010 +0.082 +0.235 +0.313

I changed my mesh to invert the worst corner.

Command I ran:
M421 I4 J4 Z-0.313

Recv: 0 1 2 3 4
Recv: 0 -0.175 +0.057 +0.142 +0.173 +0.230
Recv: 1 -0.023 +0.048 +0.030 +0.072 +0.023
Recv: 2 +0.010 +0.075 +0.012 +0.043 +0.063
Recv: 3 -0.035 +0.035 +0.050 +0.152 +0.148
Recv: 4 -0.070 +0.010 +0.082 +0.235 -0.313

Did a test print and 4,4 had really good adhesion on the worst corner, by comparison to the previous attempt, it was .626 mm away from the bed.

Everything up until this issue has been pretty normal. It homes well, it probes well, the only thing that was abnormal was at one time when I was doing a 4x4 probing, it started at the right hand side corner rather than the left. When I did 5x5 probing, it went back to starting at the left corner, (0,0). Not sure if it's indicative of anything, but I thought it was worth mentioning.

Additionally, today I did a G26 for mesh validation and it started at the back right, (300,300). I'm starting to think this is indicative of something. 

Any Ideas regarding the mesh inversion/rotation?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6057280/Configuration.zip)

thetailwind, 2021-03-07:
In conclusion everything was fine. The Bed visualizer was flipped and rotated incorrectly. I found this out by placing quarters where some corner probe points were and looked at the visualization. The corners were not matching up so once I adjusted the rotation and actually adjusted the correct knob, everything was fine.

hakimio, 2021-03-08:
Close the issue then?

ellensp, 2021-03-08:
Im going to close it.

closed 2021-03-08 by ellensp