MarlinFirmware/Marlin #21234: M32 loads incorrectly, causes Subcall overflow[BUG]

Astrocytosis, 2021-03-01:
### Bug Description

I'm using the latest Marlin version on my SKR v1.4 Turbo for my Sandbot.
I set up an auto0.g file that contained 4 different M32 P"filename"**#** commands, as I wanted the printer to draw a number of patterns on power on automatically.
Each file was relatively large (about 500kb).

The Printer would either boot and then display: "subcall Overflow"
or it would (after increasing BUFSIZE a lot) start printing the 2nd file somewhere in the middle of the Gcode and skip the first file entirely.

From my very crude understanding of things it seems to be an issue with the M32 command pre-reading all the files eventhough I put the # behind the file (as specified on the marlin command wiki) to stop it from doing it.
This seems to cause a buffer overflow.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6062403/Configuration.zip)
**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.
I tried the following changes in the Marlin config with only limited results:
- Set BUFSIZE to 4, as this was recommended by a lot of people to fix common buffer issues....no change
- Maximized BUFSIZE all the way to 205, with Max CMD size down to 50.... now auto0.g was able to print 3 of the 4 files correctly, but skipping the first one entirely
- disabled every feature in marlin that wasn't used to maximize free RAM.... no change
- changed RX and TX Buffersizes to see if this had any impact on it....no change
- changed SD Procedure depth to 2 (eventhough all files that auto0.g calls on are in the same directory as auto0.g)....Result was: Printer printed a couple gcode lines more than previously, but still not the whole files before proceeding to the next file in the middle of the print job

### Steps to Reproduce

1. Created a file named auto0.g containing 4 M32 Commands in it, transfered to SD
2. Printer boots up
3. Printer skips the first file
4. Printer starts second file many Gcode lines below the start point
5. proceeds to print the rest of the files

**Expected behavior:**

Printing all specified files in series

**Actual behavior:**

Printer aborts with error message: "Subcall Overflow"

or (after BUFSIZE increase)

Prints only parts of the 4 files

#### Additional Information

If any more Information or pictures are needed I'll provide them
[image]

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-03-01:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Astrocytosis, 2021-03-11:
Using the bugfix branch sadly didn't resolve the issue

closed 2021-05-25 by github-actions[bot]