MarlinFirmware/Marlin #21235: Cyrillic (Ukrainian or Russian) lcd language not working

alexeyzel, 2021-03-01:
### Bug Description

When i try to set Ukrainian (uk) or Russian (ru) language in configuration.h file after firmware update screen menus are not showing Ukrainian/Russian text. 
I am using bugfix version, by printer mainboard is MKS Robin Nano 3, screen is MKS TS35 v2.0

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6062834/Marlin.zip)

### Steps to Reproduce

1. set #define LCD_LANGUAGE uk
2. set #define DISPLAY_CHARSET_HD44780 CYRILLIC
3. compile marlin and flash the printer
4. open the menu

**Expected behavior:**

Ukrainian language menu items showing.

**Actual behavior:**

Some strange symbols instead of Ukrainian words showing as menu items.

#### Additional Information

I tried to enable UTF8 support in configuration_adv.h file with #define TOUCH_UI_USE_UTF8 and #define TOUCH_UI_UTF8_CYRILLIC_CHARSET enabled - it did not helped.

Screenshot of the problem:
[image]

ellensp, 2021-03-02:
I see in Marlin/src/lcd/tft/fontdata/fontdata_ISO10646_1.cpp
`#define STM32_NOT_EXTENDED_ISO10646_1_5X7
`
  // reduced font (only symbols 1 - 127) - saves about 1278 bytes of FLASH

perhaps this is the issue?   
Sadly I don't have access to this lcd type to try

ellensp, 2021-03-02:
Didn't mean to close.....  reopened

alexeyzel, 2021-03-02:
I will try to test it.

alexeyzel, 2021-03-02:
I tried following:
In configuration.h:
```
#define LCD_LANGUAGE uk
#define DISPLAY_CHARSET_HD44780 CYRILLIC
```

In Marlin/src/lcd/tft/fontdata/fontdata_ISO10646_1.cpp disable this option:

` //#define STM32_NOT_EXTENDED_ISO10646_1_5X7 did not solve the problem.`

The problem remains.
I did not try enabling UTF8 support in configuration_adv.h, i think it is not related.

ellensp, 2021-03-02:
@rhapsodyv can you help with this?

rhapsodyv, 2021-03-02:
the current COLOR_UI font only have latin characters.... There's another issue to add Chinese characters too.

moonglow, 2021-05-14:
@alexeyzel as a temporary solution that not suitable as complete solution you can remap Russian symbols from Unicode position ( 0x410 - 0x44F ) to 0xC0 - 0xFF with two exception for 'Ё' symbol ( capital 'Ё' to 0xA8, small 'ё' to 0xB8 ), next you need generate new  `helvetica_18.cpp` ( used by default for COLOR_UI ), and add UNICODE->ASCII remap for  `TFT_String::add` function ( file:  `tft_string.cpp` ):

original code:
`if (wchar > 255) wchar |= 0x0080;`

add remap:
```
/* russian ? remap to ASCII => 0xC0 - 0xFF */
if( wchar >= 1040 && wchar <= 1103 ) wchar -= 848;
/* russian capital IO */
else if( wchar == 1025 ) wchar = 168;
/* russian small IO */
else if( wchar == 1105 ) wchar = 184;
else if (wchar > 255) wchar |= 0x0080;
```

If you want to try it, i attached two files `tft_string.cpp` and `helvetica_18.cpp`, replace original files with that and you can to use Russian localization, for Ukrainian language you need to do same trick and remap all needed symbols.

**Example for my board:**
[image]

**Modified sources:**
[patched_data.zip](https://github.com/MarlinFirmware/Marlin/files/6480455/patched_data.zip)

`helvetica_18.cpp` location `Marlin/Marlin/src/lcd/tft/fontdata`
`tft_string.cpp` location `Marlin/Marlin/src/lcd/tft`

rhapsodyv, 2021-05-14:
@moonglow what did you use to generate the helvetica_18.cpp?

moonglow, 2021-05-14:
@rhapsodyv

`otf2bdf` to convert ttf to bdf ( i use Ubuntu Mono as main font ), did not found original helvetica font
`gbdfed` to edit font data and copy symbol table from ( 0x410 - 0x44F ) to ( 0xC0 - 0xFF )
`bdf2u8g` to generate new font header data

moonglow, 2021-05-17:
@rhapsodyv Maybe tries to generate language specific font tables and if unicode character code match to specific language table try to get glyph data from it, i think it will reduce font data size if we slice it with only needed language regions.

bmk74, 2022-08-21:
Program for generate russian fonts
[](https://github.com/bmk74/Adafruit-fontconverter-for-russian-font)

thisiskeithb, 2022-12-08:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25073

closed 2022-12-08 by thisiskeithb