MarlinFirmware/Marlin #21236: [BUG] Unable to compile LPC1769 since today.

Magiquall, 2021-03-01:
Hello,

Sorry for my Limited English.

Yesterday I compiled the Marlin project for 3D printers several times without worry, but today not possible.

Yesterday every time I modified marlin to suit my printer, I saved the complete folder.
And today none of these backups cant to compile.
The program runs on an MKS SGen_L V2.0 motherboard, with an LPC1769 processor.

I haven’t changed anything since yesterday and it no longer works. **last night I had an update**, and I do not know if I made any tests after this update, which probably explains my problem.

When I run the compilation, I quickly get yellow and red error lines

I tried to put my folder with marlin, at the root of my hard drive, to have a short path. Without results.
I tried to uninstall, PlatformIO, then Visual Studio Code, to reinstall them, but without result.

Do you have any ideas please?

PlatformIO is in: Core 5.1.0 Home 3.3.3
Visual Studio Code in: 1.53.2

**### Terminal:**

> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\Chris\.platformio\penv\Scripts\pio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.6
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph        
|-- <Servo> 1.0.0       
Building in release mode
error:Command 'cmd /C dir H:\' returned non-zero exit status 1.
error:Command 'cmd /C dir I:\' returned non-zero exit status 1.
error:Command 'cmd /C dir K:\' returned non-zero exit status 1.
error:Command 'cmd /C dir L:\' returned non-zero exit status 1.
error:Command 'cmd /C dir M:\' returned non-zero exit status 1.
error:Command 'cmd /C dir N:\' returned non-zero exit status 1.

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o
In file included from Marlin\src\HAL\LPC1768\HAL_MinSerial.cpp:24:
Marlin\src\HAL\LPC1768\HAL.h:35:10: fatal error: algorithm: No such file or directory
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o
Compiling .pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_MinSerial.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\HAL_SPI.cpp:51:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\HAL_SPI.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\eeprom_flash.cpp:39:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_flash.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\eeprom_sdcard.cpp:25:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
arm-none-eabi-gcc: fatal error: cannot execute 'as': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_sdcard.cpp.o] Error 1
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\include\digipot_mcp4451_I2C_routines.c.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\eeprom_wired.cpp:24:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\eeprom_wired.cpp.o] Error 1
arm-none-eabi-g++: fatal error: cannot execute 'as': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\fast_pwm.cpp.o] Error 1
arm-none-eabi-gcc: fatal error: cannot execute 'as': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\include\i2c_util.c.o] Error 1
In file included from Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\tft\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:23:
Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
In file included from Marlin\src\HAL\LPC1768\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\HAL\LPC1768\../../inc/MarlinConfig.h:31,
                 from Marlin\src\HAL\LPC1768\main.cpp:34:
Marlin\src\HAL\LPC1768\../../inc/../HAL/./LPC1768/HAL.h:35:10: fatal error: algorithm: No such file or directory
   35 | #include <algorithm>
      |          ^~~~~~~~~~~
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\main.cpp.o] Error 1
arm-none-eabi-g++: fatal error: cannot execute 'as': CreateProcess: No such file or directory
compilation terminated.
*** [.pio\build\LPC1769\src\src\HAL\LPC1768\MarlinSerial.cpp.o] Error 1
====================================================================== [FAILED] Took 3.05 seconds ======================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:03.051
================================================================ 1 failed, 0 succeeded in 00:00:03.051 ================================================================

Important:
I just tried with a marlin download from the official site, and compiling keeping the mega2560 and BOARD_RAMPS_14_EFB, it works, but as soon as I put my LPC1769 and BOARD_MKS_SGEN_L_V2 card back, it doesn’t work anymore.

Thanks for your help

ellensp, 2021-03-01:
Cannot replicate.

delete your .platformio in your home directory, this forces platformio to reinstall all build environments.

Evg33, 2021-03-01:
```
Environment    Status    Duration
-------------  --------  ------------
LPC1769        SUCCESS   00:00:40.300

```

thisiskeithb, 2021-03-03:
> Cannot replicate.

Ditto. This is a Windows/machine issue.

closed 2021-03-03 by thisiskeithb

Magiquall, 2021-03-04:
> Cannot replicate.
> 
> delete your .platformio in your home directory, this forces platformio to reinstall all build environments.
> 
> Also since its windows download and run the free version of malwarebytes and give your system a good scan and clean. Then uninstall it if you wish. I have seen malware interfering with platformio a number of times.

Perfect, it works fine now.
Thank you