MarlinFirmware/Marlin #21239: [BUG] Z initial height is high for ALL points when performing manual mesh bed levelling

pinchies, 2021-03-02:
### Bug Description

Regression due to https://github.com/MarlinFirmware/Marlin/pull/20160
This addition of MANUAL_PROBE_START_Z being defined in Conditionals_post.h stuffs up manual mesh bed levelling.

For mesh bed levelling:
if MESH_BED_LEVELING --> if true, sets PROBE_SELECTED = 1 (line 795 in conditionals_LCD.h)
if PROBE_SELECTED, then if not defined, Z_CLEARANCE_BETWEEN_PROBES gets defined from Z_HOMING_HEIGHT (line 2613 in Conditionals_post.h)
If Z_CLEARANCE_BETWEEN_PROBES  is defined, then MANUAL_PROBE_START_Z is defined (line 2629 on Conditionals_post.h)
Then if MANUAL_PROBE_START_Z  is defined, then lines 224-230 in bedlevel.cpp will not run
and thus MANUAL_PROBE_START_Z will be used for all bed levelling points as the start Z height.

The result is, that after moving to each new XY point during manual mesh bed levelling, the nozzle starts from Z = MANUAL_PROBE_START_Z  rather than Z = Zprevious. (lines 224-230 in bedlevel.cpp)

borland1, 2021-03-02:
The alias `MANUAL_PROBE_START_Z` is define so that nozzle won't crash into the bed during **manual bed leveling** when first probe point is lower than any of subsequent probe points.  Default is undefined in `Configuration.h`..
```
//#define PROBE_MANUALLY
//#define MANUAL_PROBE_START_Z 0.2
```
Not able to see your config files, but why don't you enable:  `MANUAL_PROBE_START_Z 0.2` ?

pinchies, 2021-03-08:
@borland1 -- sure, that makes sense. Are you saying that you agree that this is a bug though? 

The bug I am referring to is that when MANUAL_PROBE_START_Z  is set, the previous Z height is not used. Even if MANUAL_PROBE_START_Z  = 0, then the previous bed height will NEVER be called. From what I can see this is a bug in the logic. 

The code is extremely convoluted and there are zero code comments... does that make sense?

borland1, 2021-03-08:
I think what you described is expected behavior when calibrating the bed mesh with the bed geometry.  When `#define MANUAL_PROBE_START_Z 0.2` is uncommented (enabled), then during the Manual Bed Leveling calibration/setup, Marlin will position the nozzle height at 0.2mm above the Z homing height position for each grid point.  As you calibrate each mesh point, Marlin keeps track of each point's adjustment until you save the mesh grid point offsets to EEPROM.

Before running the Manual Bed Leveling, you should have already leveled the bed's four corners with copy paper as a feeler gauge between nozzle and bed.  Depending on your bed's flatness, you can change the 0.2mm value of `MANUAL_PROBE_START_Z`, but that default value seems like a reasonable one for most large beds which are rarely perfectly flat.  After you save the mesh to EEPROM, for each reboot of the printer, you must first send G-code G420 S to enable the previously saved grid mesh.  Bed leveling won't be active until after you Home the printer or send G-code G28 (Home).

I agree that the code is not well laid out with _Probe Options_ and _Bed Leveling_ as two separate topics in the config.  You can find more information about bed leveling setup in the Configuration Documentation on Marlin's Web site.  There is also bed leveling information in the G-code help section.

pinchies, 2021-03-08:
I disagree that this is as expected -- because this behaviour has changed, and the users I support have all noticed and complained!  It did not operate as you describe previously. The commit I refer to does not give any indication that they are aware of how this change affected the manual mesh bed levelling process -- they only refer to levelling with probes.

Can you acknowledge those two points? That (a) this behaviour has changed, and (b) that it does not appear the behaviour change was anticipated?

I understand the process how to manually level a bed, have used it for many years! 0.2mm is not a useful value for all cases, some beds are out by up to +0.4mm, especially for warped aluminium beds on large printers -- my point being that one value does not work in all cases, and having a large MANUAL_PROBE_START_Z  makes the levelling process very slow (under the present mode of operation). I would suggest it should be perhaps previous levelling point Z + some new set offset like 0.2mm, but the way this used to work was quick and efficient and did not have any issues that I experienced.

I hope I have been able to be clear, thanks.

pinchies, 2021-03-20:
Requesting review for this one. 

Looking for someone else to check my conclusions that:
(a) this bed levelling behaviour has changed due to #20160 , and 
(b) that it does not appear the behaviour change was anticipated?

The lack of commenting in this section of obtuse convoluted code is unforgivable. For bonus points, if someone can confirm the correct behaviour during bed levelling as designed (documented anywhere?) that would also be great. Or I can propose a new system.

pinchies, 2021-04-13:
Trying again: prior to 18 Nov 2020, the previous behaviour was that for manual mesh bed leveling, the previous leveled point height, was used as the start height for the next point (but moving up while traveling between points, so as not to scrape the bed with the nozzle, of course). This was a very reasonable process of operation in my opinion. In the current build, the MANUAL_PROBE_START_Z is used as the starting Z height for ALL points. I think this code change was a **bug**, as this code change was done with relation to ABL levelling and not a deliberate change to the manual mesh bed levelling process.

thinkyhead, 2021-04-25:
Please check on the latest code, since the PR was merged. It attempts to achieve consistency and provide a proper fallback if `MANUAL_PROBE_START_Z` is not defined while one of the manual-only leveling systems is enabled. Hopefully the behavior is more sensible. The old behavior of MBL should now be achievable by simply leaving `MANUAL_PROBE_START_Z` undefined while using either MBL or ABL+`PROBE_MANUALLY`.

pinchies, 2021-04-25:
I’ll give it a look over. Cheers.

thisiskeithb, 2021-08-02:
Closing since https://github.com/MarlinFirmware/Marlin/pull/21692 was merged.

closed 2021-08-02 by thisiskeithb