MarlinFirmware/Marlin #21246: Inconsistent movement reading from SD card

grishien, 2021-03-04:
### Bug Description

I’m using X axis to turn a wheel. So I’m trying to move the wheel little by little with distance between the stops of around 8mm (measured on the outer radius of the wheel).
When job is being done using USB connection & repetier - no problems, it moves equally, same distance between stops.
But when I run the same gcode from SD card - it makes one bigger (1.5 mm more) movement each 2 or 3 times. The amount of normal movements between abnormal ones is stable:
It makes 2 normal movements. Then 1 abnormal one. Then 3 normal ones. Then 1 abnormal one. And so on. 2-1-3-1-2-1-3-2 etc.
When I repeat the job - it moves exactly where it was. Meaning that abnormal movements take the same coordinate where they were.

I tried this with NEMA17+DRV8825, 4498 and I even tried MKS CD + DM556 driver + nema23. The only thing changes between these setups is the pattern (how many normal movements there are between abnormal ones).

I also tried same on MKS Robin V2.4 with its corresponding old Marlin firmware. Abnormal movements remain.

Any ideas?

### Configuration Files
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6081523/configuration.zip)

### Steps to Reproduce

1. G1 X4.4
2. M0 (mark where the pointer is)
3. G1 X8.8
4. M0 (mark where the pointer is)
5.  G1 X13.2
6. M0 (mark where the pointer is)
7. G1 X17,6
8. M0 (mark where the pointer is)
9. G1 X22
10. M0 (mark where the pointer is)
11. G1 X26,4
12. M0 (mark where the pointer is)
13. G1 X30,8
14. M0 (mark where the pointer is)
15. G1 X35,2
16. M0 (mark where the pointer is)
17. G1 X39,6
18. M0 (mark where the pointer is)
19. G1 X44
20. M0 (mark where the pointer is)
21. G1 X48.4
22. M0 (mark where the pointer is)
23. G1 X52.8
24. M0 (mark where the pointer is)
25. G1 X57,2

**Expected behavior:**

All marks should have even distance between each other.

**Actual behavior:**

Distance between the following marks is bigger:
1. X8.8-X13.2
2. X26,4-X30.8
3. X39,6-X44
and so on

ellensp, 2021-03-04:
Try adding M400's after each move, to ensure they are complete before next gcode.

grishien, 2021-03-05:
I was lucky enough to notice yesterday that I was using comma instead of dot as a delimiter of floating point. Seems that machine understood only ceil number and ditched the fractional part of a number.

You live you learn.

ellensp, 2021-03-05:
So this was user error? can we close it then?

grishien, 2021-03-09:
Yes, this can be considered closed.

closed 2021-03-09 by grishien