MarlinFirmware/Marlin #21250: [BUG] (tevo little monster failure)

jagoff412, 2021-03-04:
### Bug Description

flashing marlin 2.0  the printer will home successfully only prior to moving any axis. after moving any axis g28 fails and hard crashes the head into the x tower. ON BUGFIX THIS IS WORSE AND NO G28 SUCCEEDS, G28 IMMEDIATELY CRASHES THE PRINT HEAD HORIZONTALLY
g33 fails every attempt
fans do not work
board is mks sbase v1.3

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

flash marlin stable to the board using provided example config edited for the proper motherboard. boot. run g28. move any axis. run g28, head crash. run g33, failure error message received.
on bugfix, flash marlin to the board. boot. run g28. g28 immediately crashes the print head horizontally

**Expected behavior:**

printer should home without crashing. g33 should succeed. printer should work and not do what it is doing.
**Actual behavior:**

SEE DESCRIPTION ABOVE

#### Additional Information
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6085109/Marlin.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2021-03-04:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

jagoff412, 2021-03-04:
> * `M114_DETAIL`

Connecting...
Printer is now online.
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
echo:M105
>>> G28
SENDING:G28
echo:G28
>>> G28  X0.00 Y0.00 Z522.27
Machine Type: Delta
Probe: BLTOUCH
Probe Offset X0.00 Y20.00 Z-1.10 (Middle-Back & Below Nozzle)
Auto Bed Leveling: LINEAR (disabled)
>>> home_delta  X0.00 Y0.00 Z522.27
  current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
>>> homeaxis(88)
Home Fast: 783.41mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(X, 783.41, [200.00])
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z533.37
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(X, -5.00, [200.00])
<<< do_homing_move  X0.00 Y0.00 Z533.37
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(X, 10.00, 20.00)
<<< do_homing_move  X0.00 Y0.00 Z533.37
<<< homeaxis(88)
>>> homeaxis(89)
Home Fast: 783.41mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(Y, 783.41, [200.00])
<<< do_homing_move  X0.00 Y0.00 Z533.37
Move Away: -5.00mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(Y, -5.00, [200.00])
<<< do_homing_move  X0.00 Y0.00 Z533.37
Re-bump: 10.00mm
>>> do_homing_move  X0.00 Y0.00 Z533.37
...(Y, 10.00, 20.00)
<<< do_homing_move  X0.00 Y0.00 Z533.37
<<< homeaxis(89)
>>> homeaxis(90)
  current_position= X0.00 Y0.00 Z533.37 : Probe::set_deployed
deploy: 1
Probe::do_z_raise(5.00)
>>> do_blocking_move_to  X0.00 Y0.00 Z533.37
>  X0.00 Y0.00 Z522.27
  destination= X0.00 Y0.00 Z533.37 : destination = current_position
  destination= X0.00 Y0.00 Z522.27 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z522.27 : danger zone move
<<< do_blocking_move_to  X0.00 Y0.00 Z522.27
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
>>> do_blocking_move_to  X0.00 Y0.00 Z522.27
>  X0.00 Y0.00 Z522.27
  destination= X0.00 Y0.00 Z522.27 : destination = current_position
  destination= X0.00 Y0.00 Z522.27 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z522.27 : danger zone move
<<< do_blocking_move_to  X0.00 Y0.00 Z522.27
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
Home Fast: -783.41mm
>>> do_homing_move  X0.00 Y0.00 Z522.27
...(Z, -783.41, [200.00])
echo:busy: processing
echo:busy: processing
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z522.27
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
Move Away: 5.00mm
>>> do_homing_move  X0.00 Y0.00 Z522.27
...(Z, 5.00, 200.00)
<<< do_homing_move  X0.00 Y0.00 Z522.27
BLTouch DEPLOY requested
BLTouch Command :10
bltouch.deploy_proc() end
Re-bump: -10.00mm
>>> do_homing_move  X0.00 Y0.00 Z522.27
...(Z, -10.00, 100.00)
echo:busy: processing
<<< do_homing_move  X0.00 Y0.00 Z522.27
BLTouch STOW requested
BLTouch Command :90
bltouch.stow_proc() end
  current_position= X0.00 Y0.00 Z522.27 : Probe::set_deployed
deploy: 0
Probe::do_z_raise(5.00)
>>> do_blocking_move_to  X0.00 Y0.00 Z522.27
>  X0.00 Y0.00 Z522.27
  destination= X0.00 Y0.00 Z522.27 : destination = current_position
  destination= X0.00 Y0.00 Z522.27 : prepare_fast_move_to_destination
  current_position= X0.00 Y0.00 Z522.27 : danger zone move
<<< do_blocking_move_to  X0.00 Y0.00 Z522.27
BLTouch STOW requested
BLTouch Command :90
Disconnected.

NOTE; as soon as the probe deployed the printer immediately crashes the print head off to the side. i am unwilling to crash my machine very many more times for testing

jagoff412, 2021-03-04:
2nd note; i killed the power to the printer after the first bltouch deploy and subsequent crash. some of the final messages in the log were received after hitting the power switch but before disconnecting pronterface

closed 2021-05-14 by github-actions[bot]