MarlinFirmware/Marlin #21251: [BUG] compiled error - gcode.cpp

DrumClock, 2021-03-04:
in Bugfix 2.0.x there is a compilation error.

```
Compiling .pio \ build \ rumba32 \ src \ src \ gcode \ gcode.cpp.o
Marlin \ src \ gcode \ gcode.cpp: In static member function 'static void GcodeSuite :: process_parsed_command (bool)':
Marlin \ src \ gcode \ gcode.cpp: 400: 9: error: duplicate case value
   400 | case 42: G42 (); break; // G42: Coordinated move to a mesh point
       | ^ ~~~
Marlin \ src \ gcode \ gcode.cpp: 368: 9: note: previously used here
   368 | case 42: G42 (); break; // G42: Coordinated move to a mesh point
       | ^ ~~~
*** [.pio \ build \ rumba32 \ src \ src \ gcode \ gcode.cpp.o] Error 1
===================== ================ [FAILED] Took 18.46 seconds ================ ====
```

GMagician, 2021-03-04:
Check #21248

DrumClock, 2021-03-04:
OK, thanks

closed 2021-03-04 by DrumClock