MarlinFirmware/Marlin #21258: [FR] (Azteeg X3/ X3 Pro Configuration File)

peteromo500, 2021-03-05:
Hey, I'm new to 3-D printing, and I was hoping that the developers could make a configuration for the Azteeg X3 and X3 PRO. I have been trying to do the configuration manually, and I come up with endless errors. I know that there are hundreds of configurations available, but only one for Azteeg. I don't know if this is the best place to post this request, or if I could ask a developer directly, but I hope this gets out! Thanks for reading this and have a great day!

thisiskeithb, 2021-03-05:
The best place to ask for firmware would be the manufacturer since they have to provide a copy if they are using Marlin.

Another good place would be a printer-specific group on Facebook.

If you find a config that works, please contribute it to the [Marlin Configurations](https://github.com/MarlinFirmware/Configurations) repo or drop off a .zip with the config in a reply so it can be added.

For general config building & troubleshooting, I’d recommend the following resources:

- MarlinFW.org [Marlin Documentation](https://marlinfw.org)
- RepRap.org [Marlin Forum](https://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Discord server](https://discord.gg/n5NJ59y)

ellensp, 2021-03-05:
As far as I'm aware the Azteeg X3/ X3 Pro are motherboards. 
You don't have full Configuration files just for motherboards. Configuration files are for full machines.

Marlin already support both of these
Set #define MOTHERBOARD to either BOARD_AZTEEG_X3 or BOARD_AZTEEG_X3_PRO

closed 2021-03-05 by ellensp