MarlinFirmware/Marlin #21259: [BUG] New warning generated with PR#21254

bill-orange, 2021-03-05:
Bug Description:

Bug Description: compilation is successful with BugFix but with this warning at the beginning of the compile.

Can not remove temporary directory C:\Users\William\Documents\Marlin\Marlin.pio\build. Please remove it manually to avoid build issues

Manually deleting the build folder makes no difference. Compiling for a different microcontroller makes no difference. Compilation with 2.0.x does not generate this error.   This appears related to PR #21254 

Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6091548/Marlin.zip)

Steps to Reproduce

1. Compile Observe warning
2. Manually delete build and recompile
3. Observe warning

Expected behavior:  No Warning

Actual behavior:  Warning

Additional Information:

This problem cropped up earlier in issue #21203, now closed.  I would expect a similar cause.

thisiskeithb, 2021-03-05:
Find the `LPC1768_btt_skr_v1_4` section in `platformio.ini` and update as follows:

```diff
# BTT SKR 1.4 needs a UART3 tweak
[env:LPC1768_btt_skr_v1_4]
-platform    = ${LPC1768.platform}
+platform    = ${env:LPC1768.platform}
extends     = env:LPC1768
build_flags = ${env:LPC1768.build_flags} -DLPC_PINCFG_UART3_P4_28
```

I put in a PR with the fix as well: https://github.com/MarlinFirmware/Marlin/pull/21261 (thanks @rhapsodyv!)

thisiskeithb, 2021-03-06:
Fix has been merged. Closing.

closed 2021-03-06 by thisiskeithb