MarlinFirmware/Marlin #21262: [FR] Move StallGuard into Endstop

espr14, 2021-03-05:
### Description

StallGuard is turned on and off no matter of endstop state (enabled). This causes more complex code and possible issues in the development.

### Feature Workflow

I can put Stallguard control into Endstop.cpp. However, this would cause different (better) behavior:
1. Caller can pick which endstop to enable and which to disable (per axis).
2. Endstop keeps StallGuard always on if enabled_gloably is true.

This will require additional static RAM space for Endstop.cpp but PROGMEM could be lower. What do you think?

#### Additional Information

qwewer0, 2021-03-05:
Could you make a PR for it, to see it in "action"? Even as draft, if it isn't complete.

espr14, 2021-03-17:
https://github.com/MarlinFirmware/Marlin/pull/21366