MarlinFirmware/Marlin #21264: [BUG] corrupted display with latest BugFix build on Meli board

bill-orange, 2021-03-06:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

Minims, 2021-03-06:
Same behavoir on skr pro with lcd12864

bill-orange, 2021-03-06:
@Minims Did adding the three lines mentioned in my earlier post to your Configuration.h fix the problem for you?

Minims, 2021-03-07:
@bill-orange having those value 
```
// Alter timing for graphical display
#if HAS_MARLINUI_U8GLIB
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1    DELAY_NS(96)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2    DELAY_NS(48)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   DELAY_NS(600)
  #endif
#endif
```

will try with yours

Minims, 2021-03-07:
Updateed with your value but stil have the problem

bill-orange, 2021-03-07:
You can try increasing the delays and see if that helps. Increasing the delay steals CPU time from other, likely more important tasks, however. 

I suspect that, if my suggestion didn’t fix the problem that the root cause lies elsewhere. 

I tested last night on a spare SKR 1.3 board with the generic discount display and it worked fine without the delays. 

Confusing.

XDA-Bam, 2021-03-07:
There have recently been some changes to the delay function. See #20901 for example. Those *should* only affect ARM platforms, but maybe there were small changes affecting AVR, too. I don't know, but if you can reprodrucibly fix the problem with longer delays, it might be worth a look.

bill-orange, 2021-03-07:
@XDA-Bam #20901 sure appears to be a likely candidate for the source of the problem.  

I tried getting rid of the corruption multiple times.  I  did a clean rebuild, double checked that OctoPI was loading an uncorrupted hex file and finally confirmed that the delay fixed the problem.  Clearly, however, others with Melzi boards and the stock Melzi display should try to reproduce this problem. 

I don't know how many Melzi board users are left.  If the delays had not worked, I would probably have thrown up my hands and installed my spare SKR1,3 into the Monoprice Maker Select V2  and left the problem unsolved.

rhapsodyv, 2021-03-07:
@bill-orange makes all sense #20901 change lcd timing behavior. It fixes a issue that some of the delay functions were taking 50% more... and most of these functions are used on lcd timing and neopixel...

The error rate vary, but you can try to increase time by 50% or more to test.

Minims, 2021-03-07:
I have made x2 on the 3 values for SKR PRO and problem is fixed. Here is my values : 

```
#if HAS_MARLINUI_U8GLIB
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1    DELAY_NS(192)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2    DELAY_NS(96)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   DELAY_NS(1200)
  #endif
#endif
```

rhapsodyv, 2021-03-07:
> I have made x2 on the 3 values for SKR PRO and problem is fixed. Here is my values :
>
> ```
> #if HAS_MARLINUI_U8GLIB
>   #ifndef BOARD_ST7920_DELAY_1
>     #define BOARD_ST7920_DELAY_1    DELAY_NS(192) [...]

I think someone need to open a pull request and fix marlin defaults, as they are clearly outdated now.

XDA-Bam, 2021-03-07:
@bill-orange @Minims Could you try if it works with 1.5x or 1.6x the default delays? In that case, we don't have to randomly double everything.

bill-orange, 2021-03-07:
@XDA-Bam  So we are on the same page, what value are the default delays? The values used by @Minims to fix the problem are double mine.

XDA-Bam, 2021-03-07:
Good point. Defaults seem to differ by board. For Creality Melzi, they are
https://github.com/MarlinFirmware/Marlin/blob/b26516b3ac7cd7d15a751af4b299a38ac8ac7a57/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L38-L49

These values (125) are also the "default default" found in `Conditionals_LCD.h`. For SKR Pro, the defaults are
https://github.com/MarlinFirmware/Marlin/blob/b26516b3ac7cd7d15a751af4b299a38ac8ac7a57/Marlin/src/pins/stm32f4/pins_BTT_SKR_PRO_common.h#L363-L374

I do not know if it is now possible to use the same defaults due to the improved delay function, or if your have to still use different values per board.

bill-orange, 2021-03-08:
@XDA-Bam This looks like a mess.  The delays for each board appear to be empirically derived.  I don't know how you can fix this without retesting every affected board.  Maybe #20901 should be redacted for now.

XDA-Bam, 2021-03-08:
That may be the case, but obviously there is a problem and we know how to fix it for at least three of the boards (Melzi, SKR Pro, SKR Mini E3 on my printer). So I suggest we start there.

I see no reason to roll back #20901 because a couple of delays need to be tweaked now - after all, this is the development/bugfix branch and the delays were bugged. As long as none of the main developers or mister thinkyhead himself have a different opinion on this, I would kindly ask you to help me out here with testing the lowest sensible delay defaults for the Melzi and SKR Pro boards. I'll check my SKR Mini and can do the PR.

bill-orange, 2021-03-08:
@XDA-Bam For the Melzi, I would stick with the delays shown above.  I could try shorter values, but what works for me may not work for someone else.  These are the recommended "in case of difficulty" values.  https://marlinfw.org/docs/basics/troubleshooting.html

FanDjango, 2021-03-09:
I stumbled across this problem too. See my comment at the end of this commit: https://github.com/MarlinFirmware/Marlin/commit/56462cf082e8bf2bae6cb288c

I actually thought it was that commit which caused the problem, **but may be mistaken**.

I also had to adjust one or two of the above mentioned three constants in my pins file (BTT SKR PRO V1.1).

bill-orange, 2021-03-09:
@FanDjango Could you post your values for @XDA-Bam ?

FanDjango, 2021-03-09:
Yes, I will. But I am not sure what it will be worth as I spent no time to find the boundary values. I just increased timings a couple of times and I am sure this is not optimal.

If timing is now accurate to the nanosecond I suppose one could now try to adapt to the ST7920 datasheet. I was planning to look into that but was sidetracked by other issues.

XDA-Bam, 2021-03-10:
Would be nice to know the new limits on the SKR Pro. I just tested it on my SKR Mini E3 v1.2 using the stock Creality 12864 display and can report that there is no garbled text with the default timings using bugfix @ 5dc8f5cfab.

I thought about it again and I'm not sure a PR makes sense at this point. We only have two user reports with this specific problem. I think I'll wait and see if there is more people affected.

FanDjango, 2021-03-10:
Ok, this is the result of some bisecting the delay values:
```
#define BOARD_ST7920_DELAY_1    DELAY_NS(120) // 115 fails. 120 works.  96 is std.
#define BOARD_ST7920_DELAY_2    DELAY_NS(85)  //  80 fails.  85 works.  48 is std.
#define BOARD_ST7920_DELAY_3    DELAY_NS(590) // 580 fails. 590 works. 600 is std.
```
Board is `#define BOARD_INFO_NAME "BTT SKR Pro V1.1"`

I cannot say how conservative (i.e. too large by how much) the standard values (in the BTT common.h file) were before the timer commit (see prev. posts). So the relative change needed for these values **after** the commit I mentioned tells us nothing.

Note that the timer change would influence **short** delays, not long ones. So the DELAY_3 value was not compromised by the commit. It was rather the short ones (DELAY_1, DELAY_2) that were finally accurate ones and thus too short.

XDA-Bam, 2021-03-10:
@bill-orange What board is in your printer? Or, even better in case you know: What pins-file is relevant? Does it technically qualify as a Melzi board (Monoprice calls it that) in Marlin? Is `pins_MELZI.h` applicable, or would it be `pins_MELZI_V2.h`?

bill-orange, 2021-03-10:
@XDA-Bam My machine is a Monoprice Maker Select Version 2.  Board and display are stock .  The board is:

` #define MOTHERBOARD BOARD_MELZI_V2`  I am not sure how to tell what pins file is pulled into the build.  If you call tell me an easy way to find out, I will look.

XDA-Bam, 2021-03-10:
OK, thanks, that's exactly what was needed. That means pins are defined in `pins_MELZI_V2.h`, which defaults to `pins_MELZI.h`, which defaults to `pins_SANGUINOLOLU_12.h`, which finally tracks back to `pins_SANGUINOLOLU_11.h`. Oh boy...

As the config defines `LCD_FOR_MELZI`, default delays in your case should be given by 
https://github.com/MarlinFirmware/Marlin/blob/bb6d718e19c7c0e6db743b0bc615b17f121f0073/Marlin/src/pins/sanguino/pins_SANGUINOLOLU_11.h#L168-L176

which would also be set to the same values in `pins_MELZI_V2.h`, if they were not set there. Would you perhaps be willing to test some values? I'm thinking of something like
```c++
#define ST7920_DELAY_1 DELAY_NS(0) // After CLK LOW
#define ST7920_DELAY_2 DELAY_NS(400) // After DAT
#define ST7920_DELAY_3 DELAY_NS(0) // After CLK HIGH
```
in your `Configuration.h`. And if those work, maybe reduce `ST7920_DELAY_2` to 300 and then 250 and check if it's still OK?

bill-orange, 2021-03-10:
@XDA-Bam 250 and 300 did not work.  400 worked fine.  I am presently having success with:

```
#define ST7920_DELAY_1 DELAY_NS(0) // After CLK LOW
#define ST7920_DELAY_2 DELAY_NS(400) // After DAT
#define ST7920_DELAY_3 DELAY_NS(0) // After CLK HIGH
```

FanDjango, 2021-03-10:
Given the ratios I experienced, you might get 340 or 350 to work

FanDjango, 2021-03-10:
If the original values were numbers like "188", one would like to believe it to be a good idea to get as close as possible and then add a little margin. I opted for a step distance of 5 - but, phew, what a load of flashing...

XDA-Bam, 2021-03-10:
> 
> If the original values were numbers like "188", one would like to believe it to be a good idea to get as close as possible and then add a little margin. I opted for a step distance of 5 - but, phew, what a load of flashing...

I can imagine. Slightly off-topic, but when looking for one value in a large range, it can be useful to do a [binary search](https://en.wikipedia.org/wiki/Binary_search_algorithm). The algorithm is fast and easy to do "by hand".

FanDjango, 2021-03-10:
I was binary searching. Terminating at step-size 5 and using a piece of paper. 

Please note that it was needed for three separate values. And there was no real guarantee at first that these values are independant. But thankfully they are.

It seems these values more or less influence the SPI speed.

Sadly the controller datasheet gives no indication of the low level SPI acceptance, only the high level display command minimum timing.

If the code changes causing this are making timing more accurate, well done. So that DELAY_NS actually means nanoseconds! That's fine, but then I would like to know why ARM 600ns is equal to ATMEGA 400ns? (I can imagine why, really, but we are still far away from deterministic timing, IMO).

FanDjango, 2021-03-10:
I have found one quite good datasheet for the ST7920.

It would seem to me that the relationship for DLY1, DLY2 and DLY3 is more or less:

DELAY_3 > 1200ns - DELAY_1 + DELAY_2
DELAY_1 >= 140ns
DELAY_2 use equations above

Delay 1,2,3 are the timing of the software E pulses to the display (I think).

Need to hook up an oscilloscope to the E signal to check if one hundred of our config nanoseconds (on a STM32F4..) is really 100 nanoseconds. The timing change commits are supposed to make these correspond more closely than before - thus I am not surprised the timing values need to be increased a bit. But I cannot use a scope on that printer right now sadly.

If it were **sure** that we have nanosecond values for ATMEGA and AVR delay values we could unify them all. Are we sure? Not yet maybe, but the timer changes should be getting us closer. And until then we need to keep "special" config values for each board, I suppose.

XDA-Bam, 2021-03-10:
I stumbled upon #12003 where people were also having timing issues. Many soved them by increasing delays to 0/400/0 or 0/250/250. I also saw [this](https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller#Issues), which recommends inreasing DELAY_3, in case the LCD is "too slow", without going into any detail. There's also other Marlin issues and forum posts (non-Marlin) which talk about similar issues - timing problems with this display seem to be quite common.

According to the RepRap Wiki, the timing logic is
```
For each bit that is sent over SPI
    Set the clock pin low and wait ST7920_DELAY_1
    Set the data pin and wait ST7920_DELAY_2
    Set the clock pin high and wait ST7920_DELAY_3
```
That doesn't really seem to fit with those datasheet rules, though. At least not at first glance. So oscilloscope: Clean solution. Otherwise: Trial & error 😀

XDA-Bam, 2021-03-14:
As seen above, I've set up the work in progress PR #21346. Before it can get merged, **we need a couple more users to test the new timings on their SKR Pro or Melzi V2 boards**. If you are experiencing garbled display output and own one of those boards, please head over to the PR and help us test the fix.

If your are experiencing garbled output and do own a different board (*not* a SKR Pro or Melzi V2), please let us know in this issue so we can test and add improved display timings to the PR.

ramiropolla, 2021-03-22:
I believe the issue is unrelated to the improvement in the delay functions, but comes instead from the upgrade from gcc 5.4.0 to 7.3.0 in platformio's toolchain-atmelavr. You can check it by disassembling the ST7920_SWSPI_SND_8BIT() function. My board is an Anet V1.0 (sanguino/pins_ANET_10.h) running on 16 MHz and these are the results I get:

This is the sequence for writing each bit using GCC 5.4:
```asm
                cbi     2, 1
                mov     r18, r24
                lsl     r18
                sbrc    r24, 6
                rjmp    loc_9581
                cbi     2, 3

loc_9536:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+5A↓j
                nop
                sbi     2, 1
                nop
                nop
[...]
loc_9581:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+C↑j
                sbi     2, 3
                rjmp    loc_9536
```

The sequence in GCC 7.3 is a bit smarter and doesn't reload `val` for each bit:
```asm
                cbi     2, 1
                sbrc    r24, 5
                rjmp    loc_1A79
                cbi     2, 3

loc_1A44:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+4A↓j
                nop
                sbi     2, 1
                nop
                nop
[...]
loc_1A79:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+12↑j
                sbi     2, 3
                rjmp    loc_1A44
```

Note that there are no `mov` and `lsl` instructions anymore in 7.3, instead just doing `sbrc` directly on the correct bit. The time that would have been taken by `mov` and `lsl` were affecting ST7920_DELAY_1, which was being defined with a value too low.

If I set ST7920_DELAY_1 to 63 ns (which is 1 cycle in 16 Mhz) with this diff:
```diff
diff --git a/Marlin/src/pins/sanguino/pins_ANET_10.h b/Marlin/src/pins/sanguino/pins_ANET_10.h
index 74692a21a3..43d8b52604 100644
--- a/Marlin/src/pins/sanguino/pins_ANET_10.h
+++ b/Marlin/src/pins/sanguino/pins_ANET_10.h
@@ -172,7 +172,7 @@
     #define BTN_EN1                           11
     #define BTN_EN2                           10
     #define BTN_ENC                           16
-    #define BOARD_ST7920_DELAY_1 DELAY_NS(0)
+    #define BOARD_ST7920_DELAY_1 DELAY_NS(63)
     #define BOARD_ST7920_DELAY_2 DELAY_NS(63)
     #define BOARD_ST7920_DELAY_3 DELAY_NS(125)
     #define STD_ENCODER_PULSES_PER_STEP        4
```

I get this sequence:
```asm
                cbi     2, 1
                nop
                sbrc    r24, 4
                rjmp    loc_1A7F
                cbi     2, 3

loc_1A50:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+50↓j
                nop
                sbi     2, 1
                nop
                nop
[...]
loc_1A7F:                               ; CODE XREF: ST7920_SWSPI_SND_8BIT(uchar)+1E↑j
                sbi     2, 3
                rjmp    loc_1A50
```

which just adds one `nop` after `cbi     2, 1`, and my display works again.

FanDjango, 2021-03-22:
I am on STM not AVR. But your discovery is intriguing.

XDA-Bam, 2021-03-23:
That is interesting and pretty impressive detective work, @ramiropolla 😄  Maybe this issue has two separate causes: The switch in the toolchain on AVRs and the change to delay calc on STM chips.

Would somebody be able to check, if adding 63 ns to the first delay also works on other AVRs?

FanDjango, 2021-03-23:
> Maybe this issue has two separate causes: The switch in the toolchain on AVRs and the change to delay calc on STM chips.

My thinking too.

One point, though. I am surprised by the following: @ramiropolla 's display works if the DELAY1 is 63ns. Mine fails at 63ns (but I have no proof yet that my 120ns(STM) really do produce 120ns). But assuming that they do: For safety, in case his display is faster than most others, I would use two **nop**s = 126ns? maybe?

ramiropolla, 2021-03-23:
> That is interesting and pretty impressive detective work, @ramiropolla  Maybe this issue has two separate causes: The switch in the toolchain on AVRs and the change to delay calc on STM chips.
> 
> Would somebody be able to check, if adding 63 ns to the first delay also works on other AVRs?

Indeed, it seems there are two separate issues. I am not sure how to proceed though, should I create a pull request with my diff against `bugfix-2.0.x`? Or should I add a commit to `#21346` (I'm not even sure this is possible)?

XDA-Bam, 2021-03-23:
#21346 is merged, therefore you can't add anymore commits to it AFAIK. Just create a fresh PR.

ramiropolla, 2021-03-23:
> #21346 is merged, therefore you can't add anymore commits to it AFAIK. Just create a fresh PR.

Thanks, done.

@FanDjango I ended up increasing the delay from 63 to 125 because I had some tiny glitches from time to time. It looks like I don't have a faster display after all :)

XDA-Bam, 2021-04-05:
@bill-orange As far as I can see, this issue has been fixed by a PR. If you agree, would you be so kind to close the issue?

bill-orange, 2021-04-05:
Fixed by PR as noted above.

PR #21346 (2021-03-14): Update display timing for SKR Pro
### Description

This adresses timing problems and resulting garbled display output discussed in #21264. Problems were likely caused by improvements to the delay code in #20901, which probably made timing more precise and thereby revealed that the default timings are too tight. This PR is a work in progress because

- Changes to the The Melzi V2 board have only been tested by 1 board owner (@bill-orange) and are quite drastic (now 0/400/0, before 0/188/0). I am not sure that this change fits all Melzi V2 variants "in the wild" and it should be tested on more b [... cut]
closed 2021-04-05 by bill-orange