MarlinFirmware/Marlin #21265: [FR] Please document HEATER_0_MAXTEMP limit offset

ZhuDaHai, 2021-03-06:
### Description
I stumbled upon this undocumented feature after setting HEATER_0_MAXTEMP to 245 and then not being able to set the temperature above 230. Having reset it to 275 and then finding this bug report. I understand now what is going on. However I don't understand why that is not documented in configuration.h for that setting.
https://github.com/MarlinFirmware/Marlin/issues/17693

### Feature Workflow

This is basically just a documentation request, The existing documentation in Marlin for the HEATER_0_MAXTEMP is as follows:
```
// Above this temperature the heater will be switched off.
// This can protect components from overheating, but NOT from shorts and failures.
// (Use MINTEMP for thermistor short/failure protection.)
```
Please add addition documentation to this section, such as:
`//The maximum settable temperature will be 15 degrees below this temperature`

Then we know that with PTFE lined hot-ends, we need to set the HEATER_0_MAXTEMP to 265 so that we can't stupidly set the temperature above 250c and croak from toxic fumes. 

#### Additional Information

Thank you

ellensp, 2021-03-06:
Its not that simple...  the value is not hardcoded to 15. its is what ever HOTEND_OVERSHOOT is defined as, default when not manually defined is 15.

Further the MAXTEMP is also tied to the thermistor your using

`_MIN(thermalManager.heater_maxtemp[e] - HOTEND_OVERSHOOT, ui.material_preset[indh].hotend_temp)`

ie it will use the minimum value between the HEATER_0_MAXTEMP-HOTEND_OVERSHOOT and the highest temperature value in the thermistor table. 

Could all this be documented, sure, would it cause more headaches than it solves, very very likly.

ZhuDaHai, 2021-03-06:
```
// The max settable hot-end temperature will be less, according to the HOTEND_OVERSHOOT parameter, 
// or the default of 15 degresss less, if not defined
```

thinkyhead, 2021-03-06:
> Further the MAXTEMP is also tied to the thermistor your using

Oops! You're looking at the function that starts heating based on a preset, and it is being limited to what is (now) called the `max_target_temp` for the given hotend.

thinkyhead, 2021-03-06:
This should get their attention…
```cpp
/**
 * As an operator of a 3D printers it is VITAL that you understand the concept of "overshoot."
 * If you set the target temperature right at MAXTEMP you must be trying to cause a MAXTEMP error.
 * If you have observed your printer heating, you know that it can "overshoot the target" by a
 * lot, especially before you do your PID tuning. Marlin sets reasonable OVERSHOOT values to
 * prevent accidental MAXTEMP errors. If you like MAXTEMP errors, set HOTEND_OVERSHOOT to 0.
 */
```

thinkyhead, 2021-03-06:
> Then we know that with PTFE lined hot-ends, we need to set the HEATER_0_MAXTEMP to 265 so that we can't stupidly set the temperature above 250c and croak from toxic fumes.

Sure… But the error would be in unintentionally setting the max too low, not too high, right?

thisiskeithb, 2021-03-06:
Breaking out or warning about `HOTEND_OVERSHOOT` & `BED_OVERSHOOT` would be a good idea. I'm starting to add a comment on my configs with the overshoot math factored in:

```cpp
#define HEATER_0_MAXTEMP 310 // 310 - 15 = 295
#define BED_MAXTEMP      132 // 132 - 10 = 122
```

thinkyhead, 2021-03-06:
I must admit, I am slightly annoyed. I am wishing that this was a PR and not just a FR. It would have taken less typing and less of our time. The website lacks a good section about this too, and that is open source. If it's worth complaining about, it should be worth doing…. Ok, rant over. Clearly I need sleep.

thisiskeithb, 2021-03-06:
Extending the options are easy enough. I added the disclaimer from above as well: https://github.com/MarlinFirmware/Marlin/pull/21273

thinkyhead, 2021-03-06:
Maybe I should tone it down … or go the other way: "Listen here ye bastards…!"

thisiskeithb, 2021-03-06:
[image]

closed 2021-03-07 by thinkyhead