MarlinFirmware/Marlin #21269: [FR] homeing required before auto level

tome9111991, 2021-03-06:
Hi.
look for a function that allows auto level to start only after homeing has been carried out.

thinkyhead, 2021-03-06:
By default ABL will throw an error if you send `G29` before homing with `G28`. UBL's behavior is to only home if needed. You can use the undocumented `G29 N` to force homing to occur before leveling. You can also enable `NO_MOTION_BEFORE_HOMING` for more strict limits on motion before homing.

tome9111991, 2021-03-06:
If NO_MOTION_BEFORE_HOMING enable
Still work to move any axis manuel with lcd pannel?

borland1, 2021-03-07:
With manual bed leveling (MBL), mesh bed leveling is disabled prior to Homing being performed.   Why would this be any different with UBL?

enderextender, 2021-03-17:
On the Ender 3V2 the auto level menu is on the top level. Thus users will just hit that instead of homing first.  Ill have to tweak the menu or something to force it to G28 first. This should be default behavior imho

tome9111991, 2021-03-17:
I have ender3v2.
accidentally started auto level on lcd before homeing. the nozzles crash into bed

enderextender, 2021-03-18:
I think this is a bug

```
#if HAS_ONESTEP_LEVELING

  /* Leveling */
  void HMI_Leveling() {
    Popup_Window_Leveling();
    DWIN_UpdateLCD();
    queue.inject_P(PSTR("G28O\nG29"));
  }

#endif
```

the PSTR says G280 but it should be G28

closed 2021-03-31 by tome9111991