MarlinFirmware/Marlin #21271: [BUG] LCD 20x4 screen 3rd line is not completly updated when using SHOW_REMAINING_TIME

valeriu-ene, 2021-03-06:
### Before Reporting a Bug
Test with the `import-2.0.x` branch to see whether the issue still exists. This will require testing with the Marlin `bugfix-2.0.x` branch.
Tested with Marlin "Nightly" Builds, bugfix-2.0.x.zip, downloaded yesterday 2021-03-05 at 19:42UTC.

### Bug Description

I want to say first that I'm glad this feature was implemented for 20x4 LCD and is very useful even with this low severity bug.

Activating the following flags displays the elapsed time for 1s and the remaining time for another 1s.
LCD_SET_PROGRESS_MANUALLY
SHOW_REMAINING_TIME
USE_M73_REMAINING_TIME
ROTATE_PROGRESS_DISPLAY

During printing from SD card, while Bed Heating, the 3rd line is disturbed. The remaining time increaseas from 07:38 to 463.50 then to 1030:2 (just an example from the images attached). After printing starts the 3rd line of the LCD has two unwanted letters: S and R. 

### Configuration Files

[Marlin_Config_Files.zip](https://github.com/MarlinFirmware/Configurations/files/6095241/Marlin_Config_Files.zip)

### Steps to Reproduce

1. Slice the stl file with PrusaSlicer. The gcode contains M73 commands:
M73 P0 R458
M73 P0 R457
M73 P0 R456, etc, one for each minute
2. Save the gcode on the SD Card.
3. Start the print from the SD Card.

**Expected behavior:**

Remaining time should be read from gcode file and converted into hours and minutes.
During the Bed Heating the remaining time should not change if no new M73 command is processed.
LCD 3rd line should be: 
"[FR]100%  SD---% R07:33" or
"[FR]100%  SD  1% R07:33" or
"[FR]100%  SD  1% o00:12"

**Actual behavior:**

During the bed heating the remaining time increases every second until it has 3 digit hours and 2 digit minutes and then 4 digit hours and one digit minutes. 

LCD 3rd line is:
"[FR]100% SD---%**%R463**:50" or
"[FR]100% SD---%**%R1030:2**" or
"[FR]100% **S**SD  1%**R**R07:33" or
"[FR]100% **S**SD  1%**R**o00:12"

#### Additional Information

I don't know the behavior during Bed Heating but after it started printing, pressing the knob and going into Menu and out of Menu the screen refreshes and everything looks fine.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ellensp, 2021-03-07:
Line 2 is also corrupted, the custom H character is overwritten

ellensp, 2021-03-07:
Can you provide gcode file that is guaranteed to do this?

valeriu-ene, 2021-03-07:
Didn't notice line 2 until now. It seems to be related to line 3, is probably the last digit of R1030:2**7** overflowing.

I noticed that this happens when using a bigger gcode file. Keeping only the first 30 lines of gcode file does not have the same effect. Is not reproducible by sending the commands one by one from Repetier.

The archive the complete gcode files and the serial output captured with timestamps from puttyextra during the first 6 min.
[suport dus polita tara sapun p1_0.2mm_PETG.zip](https://github.com/MarlinFirmware/Marlin/files/6096930/suport.dus.polita.tara.sapun.p1_0.2mm_PETG.zip)

I also uploaded a 6 min video here: https://youtu.be/-6uuFbK1M0k
The video contains end of bed leveling with correct remaining time (first second). Then the Bed is Heating with continuous increasing of remaing time and actual printing with decreasing remaing time, until a new M73 command is reached that updated the remaining time (last 4 seconds)

deram, 2021-04-28:
Looks like there is at least two different problems leading to this behavior. 

First problem is the extrapolated estimate (not M73) values going very high (over 1000h), to values never seen as actual print time. This extrapolation should only happen if USE_M73_REMAINING_TIME is not enabled, and it only uses persentage of SD print done and time it has taken so far. It is very crude estimate, but gets progressively better when the print progresses.

Second problem, which is the cause of the extra letters on line 3, is that the lcd update function has never intended to show values go shorter (3 char hours to 2 chars) while printing, and just go longer, like the elapsed time.

The first one can be fixed by setting hard limit like 999h to the printout. With this, there should always be enough space to print the value without causing corruption on other lines than line 3.

Second part is to start printing blanking characters in between the readouts. Current implementation is optimized to not blank the characters that are intended to never change back to blank after being set to something else before. This clearly is not the case when display rotation is activated as elapsed and remaining times can have different length in characters.

deram, 2021-04-28:
Now to think this further, the remaining estimate could also show 0 until at least X percent of the print has finished. At the moment it starts showing estimate values after progress counter shows 1%.

deram, 2021-04-28:
This needs to be tried in bugfix-2.0.x, as this part of the code has seen reordering and the position where the value is printed is no longer calculated like it used to in previous stable release.

deram, 2021-04-28:
Finally found the newer addition in Marlin/src/lcd/marlinui.h Line 256 ( #20549 ) which adds the use of estimation instead of M73 set values when remaining_time has not been set (or has been reset after setting).

closed 2021-07-10 by github-actions[bot]