MarlinFirmware/Marlin #21274: Ender 6 config does not compile

Majestix1970, 2021-03-06:
Hi
thank you for creating an Ender 6 setup....

Unfortunately I can not get it compiled with VS Code 1.54.1 and PlatformIO Core 5.1.0
and latest Marlin 2.0-bugfix nighly build (06 March 2021) and your original configuration.h and Configuration_adv.h (No changes at all)
I would appriciate your help.

Best regards
Eike

I get the following error:

<details><summary>Complete Build Output, enjoy!</summary>

```
Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL_MinSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL_SPI.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\MarlinSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\SPI.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\Servo.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_bl24cxx.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_flash.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_if_iic.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_sdcard.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_wired.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\fast_pwm.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\msc_sd.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\onboard_sd.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\sdio.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_fsmc.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_spi.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\xpt2046.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\timers.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\watchdog.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\shared\HAL_MinSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\shared\esp_wifi.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\shared\servo.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\MarlinCore.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\core\serial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\core\utility.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\feature\babystep.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\feature\bedlevel\bedlevel.cpp.o
[... 109 lines cut ...]
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\extui\lib\dgus\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::HandlePIDAutotune(DGUS_VP_Variable&, void*)':
C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:29:59: error: expected primary-expression before ')' token
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
                                                           ^
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:564:13: note: in expansion of macro 'sprintf_P'
             sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"));
             ^~~~~~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp.o] Error 1
===================================================== [FAILED] Took 25.94 seconds =====================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:25.936
================================================ 1 failed, 0 succeeded in 00:00:25.936 ================================================ 
The terminal process "C:\Users\eiker\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
</details>

_Originally posted by @Majestix1970 in https://github.com/MarlinFirmware/Configurations/issues/375#issuecomment-791979621_

thinkyhead, 2021-03-06:
The issue is now patched! Thanks for the report.

Majestix1970, 2021-03-06:
Thanks thinkyhead!
please help me to understand. what does "... is now patched" mean? 
Can I already download a new build? Or can you please tell me what neets to be patch so I can continue my work.
Best Regards

ellensp, 2021-03-07:
patched means that the issue has been fixed in bugfix. Download latest version and try again. 
Details on what was changed 
https://github.com/MarlinFirmware/Marlin/commit/87bef13a4c9c1c7f3a670f906c106bc868ab12f2

Majestix1970, 2021-03-07:
Thank you! I will check

87bef13<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FMarlinFirmware%2FMarlin%2Fcommit%2F87bef13a4c9c1c7f3a670f906c106bc868ab12f2&data=04%7C01%7C%7C1920c32a369442d1958708d8e10c0bb2%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637506789131979186%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=XDKpbRgboO7CHNkgY1p95PYXMzWzTdC52aUbFGr650c%3D&reserved=0>

-

Majestix1970, 2021-03-07:
Hi together,

thanks for fixing the issue with include path fpr powerloss.h.
Unfortunately that is not all.
After downloading the nightly build bugfix again and compiling it (no changes at all to your example config for Ender 6) I get now the following error:
Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)       
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\pwr.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\rcc.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\rcc_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\spi.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\spi_f1.c.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\stm32f1\performance\isrs.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\stm32f1\performance\vector_table.S.o
Compiling .pio\build\STM32F103RET6_creality\FrameworkArduino\libmaple\systick.c.o
[... 156 lines cut ...]
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\extui\lib\dgus\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::HandlePIDAutotune(DGUS_VP_Variable&, void*)':        
[... cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp.o] Error 1
=================================================================== [FAILED] Took 29.67 seconds ===================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:29.671
============================================================== 1 failed, 0 succeeded in 00:00:29.671 ==============================================================

Majestix1970, 2021-03-07:
looks lik one include path was still wrong.
I added one more ../ to line 40 in DGUSScreenHandler.cpp
39: #if ENABLED(POWER_LOSS_RECOVERY)
40: #include "../../../../../feature/powerloss.h"
41: #endif

BUT now I get the following errors:
Terminal will be reused by tasks, press any key to close it.

> Executing task in folder Marlin-bugfix-2.0.x_Ender6: C:\Users\user\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\BL24CXX.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\buzzer.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\crc16.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\nozzle.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\numtostr.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\stopwatch.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\vector_3.cpp.o
In file included from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29:0,
                 from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
                 from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/./STM32F1/../shared/Marduino.h:36,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/./STM32F1/HAL.h:32,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\extui\lib\dgus\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::HandlePIDAutotune(DGUS_VP_Variable&, void*)':        
C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:29:59: error: expected primary-expression before ')' token    
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
                                                           ^
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:561:13: note: in expansion of macro 'sprintf_P'
             sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"));
             ^~~~~~~~~
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_FileSelected(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp:47:22: error: 'filelist' was not declared in this scope
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp:47:22: note: suggested alternative: 'filepos_t'
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
                      filepos_t
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp:48:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(touched_nr)) return;
[... 13 lines cut ...]
       dgusdisplay.WriteVariable(VP_SD_Print_Filename, filelist.filename(), VP_SD_FileName_LEN, true);
                                                       ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp:59:55: note: suggested alternative: 'filepos_t'
       dgusdisplay.WriteVariable(VP_SD_Print_Filename, filelist.filename(), VP_SD_FileName_LEN, true);
                                                       ^~~~~~~~
                                                       filepos_t
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_StartPrint(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\lib\dgus\fysetc\DGUSScreenHandler.cpp:69:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
[... cut ...]
==================================================================== [FAILED] Took 8.51 seconds ====================================================================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:08.507
============================================================== 1 failed, 0 succeeded in 00:00:08.507 ============================================================== 
The terminal process "C:\Users\user\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

ellensp, 2021-03-07:
I can confirm, Current bugfix and bugfix Ender 6 Example Config files will not compile.

ellensp, 2021-03-07:
The first sprintf_P bug is easy, in Marlin/src/lcd/extui/lib/dgus/DGUSScreenHandler.cpp 

find sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"));
replace with sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"), ExtUI::heater_t::BED);

The second filelist bug is in Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp
just add   static ExtUI::FileList filelist; after #if ENABLED(SDSUPPORT)

the last bug however has me stumped. Looks like incomplete code
It is in Marlin/src/lcd/extui/lib/dgus/origin/DGUSScreenHandler.cpp 
the line is         GotoScreen(MKSLCD_SCREEN_PAUSE);

There is no screen of this name in dgus/origin this screen is from dgus/mks and I cant see an equivalent. 
You can comment out this line and it compiles, but that leaves a broken SD_ResumePauseAbort menu.

Majestix1970, 2021-03-07:
Thank You
I will see if I find the time to get into the Marlin Code Univers to see if I can help.... But I might need help to better understand the Marlin Code structure. Would you be willing to help me if I have questions? Or is there a great place to go to understand the overall code better?

Thanks for the other tips I will try it and provide feedback

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>

Majestix1970, 2021-03-07:
Hi ellensp,
looks like this code is somehow jumbled....
now I get the following errors.....
Processing STM32F103RET6_creality (platform: ststm32@~12.0; board: genericSTM32F103RE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (12.0.0) > STM32F103RE (64k RAM. 512k Flash)       
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\sd\cardreader.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib369\SoftwareSerialM\HAL_softserial_LPC1768.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib369\SoftwareSerialM\HAL_softserial_SAMD51.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib369\SoftwareSerialM\HAL_softserial_STM32.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib369\SoftwareSerialM\HAL_softserial_STM32F1.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib369\SoftwareSerialM\SoftwareSerial.cpp.o
Compiling .pio\build\STM32F103RET6_creality\libe80\STM32ADC\STM32ADC.cpp.o
Compiling .pio\build\STM32F103RET6_creality\libe80\STM32ADC\utility\util_adc.c.o
Compiling .pio\build\STM32F103RET6_creality\lib300\EEPROM\EEPROM.cpp.o
Compiling .pio\build\STM32F103RET6_creality\lib300\EEPROM\flash_stm32.c.o
Compiling .pio\build\STM32F103RET6_creality\libfa1\USBComposite\AbsMouse.cpp.o
Compiling .pio\build\STM32F103RET6_creality\libfa1\USBComposite\Consumer.cpp.o
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_FileSelected(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:47:22: error: 'filelist' was not declared in this scope
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:47:22: note: suggested alternative: 'filepos_t'
     if (touched_nr > filelist.count()) return;
                      ^~~~~~~~
                      filepos_t
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:48:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:48:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(touched_nr)) return;
          ^~~~~~~~
          filepos_t
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:50:9: error: 'filelist' was not declared in this scope
     if (filelist.isDir()) {
         ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:50:9: note: suggested alternative: 'filepos_t'
     if (filelist.isDir()) {
         ^~~~~~~~
         filepos_t
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:65:87: error: 'filelist' was not declared in this scope
     HandleUserConfirmationPopUp(VP_SD_FileSelectConfirm, nullptr, PSTR("Print file"), filelist.filename(), PSTR("from SD Card?"), true, 
true, false, true);
                                                                                       ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:65:87: note: suggested alternative: 'filepos_t'
     HandleUserConfirmationPopUp(VP_SD_FileSelectConfirm, nullptr, PSTR("Print file"), filelist.filename(), PSTR("from SD Card?"), true, 
true, false, true);
                                                                                       ^~~~~~~~
                                                                                       filepos_t
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_StartPrint(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:69:10: error: 'filelist' was not declared in this scope
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp:69:10: note: suggested alternative: 'filepos_t'
     if (!filelist.seek(file_to_print)) return;
          ^~~~~~~~
          filepos_t
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\dgus\origin\DGUSScreenHandler.cpp.o] Error 1
====================================================== [FAILED] Took 8.22 seconds ======================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:08.222
================================================= 1 failed, 0 succeeded in 00:00:08.222 =================================================

Majestix1970, 2021-03-07:
Sorry ellensp! My mistake
I did overlook you second correction....
After following your suggestion I have now the same last error regarding'MKSLCD_SCREEN_PAUSE'

Now I can look into this.

Thanks again

Majestix1970, 2021-03-07:
I did the following.... Now it compiles successfully! Is that a valid fix?
in DGUSScreenHandler.cpp
I replaced line 86
    GotoScreen(MKSLCD_SCREEN_PAUSE);
with
    GotoScreen(DGUSLCD_SCREEN_POPUP); 

Where can I find information on how this screen handling/setup is structured?
Where should MKSLCD_SCREEN_PAUSE definded and implemented?

Thanks for your help

Majestix1970, 2021-03-07:
Also it looks like the macro defind in pgmspace.h:29:59: 
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
thoughts a warning now: too many arguments for format [-Wformat-extra-args]
and refers to your suggested change in DGUSScreenHandler.cpp:564:13: 
note: in expansion of macro 'sprintf_P'
             sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"), ExtUI::heater_t::BED);

Any advice?
Thank you

here the warning :
In file included from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29:0,
                 from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\marlinui.cpp.o
                 from C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from c:\users\eiker\desktop\marlin-bugfix-2.0.x\marlin\src\hal\shared\marduino.h:36,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/./STM32F1/HAL.h:32,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/../HAL/HAL.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\../../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\extui\lib\dgus\../../ui_api.h:45,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.h:30,
                 from Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::HandlePIDAutotune(DGUS_VP_Variable&, void*)':
C:\Users\eiker\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:29:59: warning: too many arguments for format [-Wformat-extra-args]
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
                                                           ^
Marlin\src\lcd\extui\lib\dgus\DGUSScreenHandler.cpp:564:13: note: in expansion of macro 'sprintf_P'
             sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"), ExtUI::heater_t::BED);
             ^~~~~~~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\libs\BL24CXX.cpp.o

Majestix1970, 2021-03-07:
sorry for the type... "throws" not "thoughts"  ;-)

Majestix1970, 2021-03-07:
sprintf is expacting a parameter e.g %d in the format string. Thats why it throws this warning...
Would it be correct to replace your suggestion in DGUSScreenHandler.cpp line 564
    sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"), ExtUI::heater_t::BED); 
with
   sprintf_P(buf, PSTR("M303 E%d C5 S70 U1"), ExtUI::heater_t::BED); 

If I do so the compiler is happy and no warnings any more.

Majestix1970, 2021-03-09:
Hi, any idea what this could be.....
showed up after setting Extruders =2 => after fixing some other missing #defines this error showed up which I am not deep enought into the code (yet)....
Can you help?

....
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\motion.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\planner.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\planner_bezier.cpp.o
Compiling .pio\build\BIGTREE_SKR_PRO\src\src\module\probe.cpp.o
Marlin\src\lcd\extui\lib\dgus\origin\DGUSDisplayDef.cpp:219:75: error: 'DGUSLCD_SendFloatAsLongValueToDisplay' was not declared in this scope
  219 |     VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, DGUSLCD_SendFloatAsLongValueToDisplay<1>),
      |                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSDisplayDef.cpp:153:64: note: in definition of macro 'VPHELPER'
  153 |   .set_by_display_handler = RXFPTR, .send_to_display_handler = TXFPTR }
      |                                                                ^~~~~~
Marlin\src\lcd\extui\lib\dgus\origin\DGUSDisplayDef.cpp:153:71: error: expected primary-expression before '}' token
  153 |   .set_by_display_handler = RXFPTR, .send_to_display_handler = TXFPTR }
      |                                                                       ^
Marlin\src\lcd\extui\lib\dgus\origin\DGUSDisplayDef.cpp:219:5: note: in expansion of macro 'VPHELPER'
  219 |     VPHELPER(VP_T_E1_Is, &thermalManager.temp_hotend[1].celsius, nullptr, DGUSLCD_SendFloatAsLongValueToDisplay<1>),
      |     ^~~~~~~~
*** [.pio\build\BIGTREE_SKR_PRO\src\src\lcd\extui\lib\dgus\origin\DGUSDisplayDef.cpp.o] Error 1
============================================================= [FAILED] Took 11.47 seconds =============================================================

Majestix1970, 2021-03-09:
Hm..... just figured out that the whole definition of the funktion DGUSLCD_SendFloatAsLongValueToDisplay is missing. 
after adding the code below  to "lcd/extui/lib/dgus/DGUSDisplay.h" it compiles..... 
I wounder if the DGUS lib corrently included in the nightly build is complete at all. 
If I compare it with 
[https://gitlab.d-net.ovh/dusakus/marlin-ferk/-/blob/5f27f7de47191c5cd2a6f4c2f0b65748ef13c268/Marlin/src/lcd/extensible_ui/lib/dgus/DGUSDisplay.h] it looks not complete to me. Any idea whats wrong here?

/// Send a float value to the display.
  /// Display will get a 4-byte integer scaled to the number of digits:
  /// Tell the display the number of digits and it cheats by displaying a dot between...
  template<unsigned int decimals>
  static void DGUSLCD_SendFloatAsLongValueToDisplay(DGUS_VP_Variable &var) {
    if (var.memadr) {
      float f = *(float *)var.memadr;
      f *= cpow(10, decimals);
      union { long l; char lb[4]; } endian;

      char tmp[4];
      endian.l = f;
      tmp[0] = endian.lb[3];
      tmp[1] = endian.lb[2];
      tmp[2] = endian.lb[1];
      tmp[3] = endian.lb[0];
      dgusdisplay.WriteVariable(var.VP, tmp, 4);
    }
  }

Majestix1970, 2021-03-14:
just a short update.... did compile, but after flashing firmware, the display remains black.....
need to dig further....

any help is more than welcome...

Majestix1970, 2021-03-21:
Hi out there....
can someone guide me in order to understand how displays are added into Marlin. I have no detailed knowledge about how this is done in Marlin 2.0
I currently try reverse engineering on the original Ender-6 how the Creality Firmware communicates with the DGUS screen and then to write a working DGUS original display driver for Marlin 2.0.
But I need help with the Marlin code.
Thank you.

vipertech17, 2021-04-21:
tried to compile firmware for ender 6 tonight and still fails

thinkyhead, 2021-05-02:
Sorry, we are needing fixes from @makerbase-mks and maybe @solawc …. The ExtUI code for DGUS displays was never very stable to begin with, and it was "approved" by the submitters apparently before really being tested.

msedlak77, 2021-05-02:
Is there config files available for the 12864 LCD Display (Ender 3 screen) instead. Or is it easy to change it in the current files and will it then compile?

ellensp, 2021-05-02:
@msedlak77 In theory all you need to do that is disable DGUS_LCD_UI_ORIGIN and enable the new type of screen CR10_STOCKDISPLAY

solawc, 2021-05-05:
> Sorry, we are needing fixes from @makerbase-mks and maybe @solawc …. The ExtUI code for DGUS displays was never very stable to begin with, and it was "approved" by the submitters apparently before really being tested.

Hi, I'm sorry for the problem. Colleagues, I also have doubts. If dgus is turned DGUS_LCD_UI_MKS, the compilation error path should not appear in /dgus/origin/dgusscreenhandler.cpp, but should be in /dgus/mks/dgusscreenhandler.cpp, because it has been separated and independent. @Majestix1970

Majestix1970, 2021-06-26:
no unfortunately not.... I> started to dig into this issue and was sofar able to understand and already change the screen configuration of the DGUS II Ender6 display. What needs to be done now is to understand how the Marlin screen handler is programmed (I did not do this before...)  Maybe someone could help me explain how to programm your own screen handler in Marlin. The DGUS II DIsplay ist completely different in working as all the other displays....

Any help is appreceated!  I do not have to time currently to learn it by my self from scratch.
>Thanks

sparksplur, 2021-07-02:
Has anyone been able to actually flash to the ender 6? i can compile the bin but every time try to flash it just stays on loading screen...

kudagaci, 2021-07-07:
I tried to cheat by changing the mkslcd_screen_pause with a proper one from DGUSDisplayDef.h (DGUSLCD_SCREEN_WAITING to be exact) just to make it compile at least and work until i hit the pause in screen. It compiled. But it just stucks at the opening screen with the loading bar at %0. 

With the latest config file, the Marlin tries to use DGUS lcd in the extui folder but the problem is, as stated above, there is no screen defined for pause. But i've no idea why it doesn't flash even though another screen is set to be pop up when pressed pause.

sassysander, 2021-07-19:
Same issue here. tried both the regular and bugfix versions and get the following error:

In file included from C:\Users\ssass\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/WString.h:29:0,
                 from C:\Users\ssass\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/wirish.h:47,
                 from C:\Users\ssass\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/Arduino.h:30,
                 from d:\ender_6\marlin-2.0.x\marlin\src\hal\shared\marduino.h:36,
                 from d:\ender_6\marlin-2.0.x\marlin\src\hal\stm32f1\hal.h:32,
                 from d:\ender_6\marlin-2.0.x\marlin\src\hal\hal.h:30,
                 from Marlin\src\lcd\extui\dgus\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\extui\dgus\../ui_api.h:45,
                 from Marlin\src\lcd\extui\dgus\DGUSScreenHandler.h:30,
                 from Marlin\src\lcd\extui\dgus\DGUSScreenHandler.cpp:27:
Marlin\src\lcd\extui\dgus\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::HandlePIDAutotune(DGUS_VP_Variable&, void*)':
C:\Users\ssass\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple/avr/pgmspace.h:29:59: error: expected primary-expression before ')' token
 #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
                                                           ^
Marlin\src\lcd\extui\dgus\DGUSScreenHandler.cpp:562:13: note: in expansion of macro 'sprintf_P'
             sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"));
             ^~~~~~~~~
Marlin\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_ResumePauseAbort(DGUS_VP_Variable&, void*)':
Marlin\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp:88:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope
         GotoScreen(MKSLCD_SCREEN_PAUSE);
                    ^~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp:88:20: note: suggested alternative: 'DGUSLCD_SCREEN_UNUSED'
                    ^~~~~~~~~~~~~~~~~~~
                    DGUSLCD_SCREEN_UNUSED
*** [.pio\build\STM32F103RET6_creality_maple\src\src\lcd\extui\dgus\DGUSScreenHandler.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality_maple\src\src\lcd\extui\dgus\origin\DGUSScreenHandler.cpp.o] Error 1

ellensp, 2021-07-31:
After PR https://github.com/MarlinFirmware/Marlin/pull/22464 was added to fix this LCD type on the ender 5 plus
The Ender 6 example config also compiles when using building env STM32F103RET6_creality
But not with the older soon to be removed anyways STM32F103RET6_creality_maple

ellensp, 2021-07-31:
If you really want to use te maple env, apply this diff

```DIFF
diff --git a/Marlin/src/lcd/extui/dgus/DGUSScreenHandler.cpp b/Marlin/src/lcd/extui/dgus/DGUSScreenHandler.cpp
index 57ecc4bb65..5b91b7a880 100644
--- a/Marlin/src/lcd/extui/dgus/DGUSScreenHandler.cpp
+++ b/Marlin/src/lcd/extui/dgus/DGUSScreenHandler.cpp
@@ -559,7 +559,7 @@ void DGUSScreenHandler::HandleStepPerMMExtruderChanged(DGUS_VP_Variable &var, vo
         #endif
         #if ENABLED(PIDTEMPBED)
           case VP_PID_AUTOTUNE_BED:
-            sprintf_P(buf, PSTR("M303 E-1 C5 S70 U1"));
+            sprintf_P(buf, PSTR("M303 E%d C5 S70 U1"),-1);
             break;
         #endif
     }
```

ellensp, 2021-07-31:
@thinkyhead do we want to patch for maple env or should we just start removing maple env's as they fail?

thinkyhead, 2021-07-31:
Are you saying that Maple platforms are still being updated by "Maple, Inc." and breaking our builds?

ellensp, 2021-07-31:
@thinkyhead no, that there is a bug in maple env. that marlin triggers  In this case sprintf_P without a veritable will not compile 
You can get around it, but is it worth it? or should we start removing maple envs

thinkyhead, 2021-07-31:
Are we confident that STM32 now has everything covered￼ and that there are no boards or situations that need Maple to work best?￼

ellensp, 2021-07-31:
not fully for all boards, and I don't recommend removing all maple in one go at all.  For now I am just talking about potentially removing maple build option from the pins.h for this one board.

thinkyhead, 2021-08-01:
It's fine if all `sprintf(a, "b")` instances are replaced with `strcpy(a, "b")` as that is more correct, and `sprintf` should really only be used when there are substitution parameters.

closed 2021-08-01 by thinkyhead

Majestix1970, 2021-08-08:
Hi all!
I was quite inactive the last weeks due to too much work.....

Can someone of you explain/tell me please where in the Marlin code / files I find the source code to map the DGUS screen Id and variable IDs to the normal Marlin commands.

so far I maneged to change and compile the DGUS original Ender 6 screens. Now I need to understand how the marlin code die read/send DGUS screen command to from/to the display.  If I understood that, I could provide a complete tutorial to change/udapt and compile the DGSU screen for the Ender6 and provide a correct Marlin display driver.

Thanks

sassysander, 2021-08-09:
I've just ran the compile using the Ender 6 configs again, but I still 
get the same sort of errors? Has anyone successfully compiled Ender 6 
firmware using the latest config?

Sander Sassen
CinePilot | FPV.drone.cinematic
FPV droneshots - Filmmaking
Web: http:\\www.cinepilot.nl
Call\Whatsapp: +31(0)651015475

ellensp, 2021-08-09:
@sassysander you need to use current bugfix or current 2.0.x (from today) and you probably should delete your Marlin/.pio directory 
It builds fine on both build environments and both versions of marlin using these config files
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-6

sassysander, 2021-08-09:
@ellensp would you perhaps be available to help me debug this perhaps? I'd be happy to make a donation, or buy you a coffee?

ellensp, 2021-08-09:
@sassysander come to the discord https://discord.gg/n5NJ59y and post in general "Looking for ellensp"  ill start a DM with you.   But be quick its really late here

sassysander, 2021-08-09:
@ellensp same here, I'm also about to call it a night! Can I propose a different approach perhaps? I have a document prepared with some questions etc. illustrated with a few videos, can I forward that to you so you can have a look? Perhaps email is better suited for this? FYI, I can be reached at github(at)sassiness(dot)nl.

ellensp, 2021-08-09:
Will think about it tomorrow then