MarlinFirmware/Marlin #21275: [BUG] Repeated warnings during compile about `warning: "DEBUG_ECHOPGM_P" redefined`

lightmaster, 2021-03-06:
### Bug Description

Sometime in the last week I've started getting a lot of warnings during compile about `warning: "DEBUG_ECHOPGM_P" redefined`, possibly related to #21231. It still compiles fine, but figured since it shows up hundreds of times I'd mention it.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Marlin SKR 1.4 Turbo configs 2021-03-06.zip](https://github.com/MarlinFirmware/Marlin/files/6095977/Marlin.SKR.1.4.Turbo.configs.2021-03-06.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

Marlin should compile without a huge amount of warnings

**Actual behavior:**

Lots of repeated warnings

#### Additional Information

Complete log file with compile warnings: [https://pastebin.com/4f3WVaE6](https://pastebin.com/4f3WVaE6)

```
Compiling .pio\build\LPC1769\src\src\core\utility.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\babystep.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\bedlevel.cpp.o
In file included from Marlin\src\feature/bedlevel/ubl/ubl.h:328,
                 from Marlin\src\feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\MarlinCore.cpp:175:
Marlin\src\feature/bedlevel/ubl/../../../core/debug_out.h:92: warning: "DEBUG_ECHOPGM_P" redefined
   92 |   #define DEBUG_ECHOPGM_P(P)          NOOP
      |
In file included from Marlin\src\feature/bedlevel/ubl/ubl.h:29,
                 from Marlin\src\feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\MarlinCore.cpp:175:
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\ubl\ubl.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
Marlin\src\feature/bedlevel/ubl/../../../core/debug_out.h:62: note: this is the location of the previous definition
Compiling .pio\build\LPC1769\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o
   62 |   #define DEBUG_ECHOPGM_P(P)      SERIAL_ECHOPGM_P(P)
Compiling .pio\build\LPC1769\src\src\feature\fwretract.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\leds\leds.cpp.o
      | 
Compiling .pio\build\LPC1769\src\src\feature\leds\neopixel.cpp.o
Compiling .pio\build\LPC1769\src\src\feature\leds\printer_event_leds.cpp.o
In file included from Marlin\src\core\../feature/bedlevel/ubl/ubl.h:328,
Compiling .pio\build\LPC1769\src\src\feature\pause.cpp.o
                 from Marlin\src\core\../feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\core\utility.cpp:56:
Marlin\src\core\../feature/bedlevel/ubl/../../../core/debug_out.h:92: warning: "DEBUG_ECHOPGM_P" redefined
   92 |   #define DEBUG_ECHOPGM_P(P)          NOOP
      | 
In file included from Marlin\src\core\../feature/bedlevel/ubl/ubl.h:29,
                 from Marlin\src\core\../feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\core\utility.cpp:56:
Compiling .pio\build\LPC1769\src\src\feature\runout.cpp.o
Marlin\src\core\../feature/bedlevel/ubl/../../../core/debug_out.h:62: note: this is the location of the previous definition
   62 |   #define DEBUG_ECHOPGM_P(P)      SERIAL_ECHOPGM_P(P)
      | 
Compiling .pio\build\LPC1769\src\src\feature\tmc_util.cpp.o
Compiling .pio\build\LPC1769\src\src\gcode\bedlevel\G42.cpp.o
In file included from Marlin\src\feature\bedlevel\ubl/ubl.h:328,
Compiling .pio\build\LPC1769\src\src\gcode\bedlevel\M420.cpp.o
[... 103 lines cut ...]
      |
In file included from Marlin\src\gcode\bedlevel\ubl\../../../feature/bedlevel/ubl/ubl.h:29,
Compiling .pio\build\LPC1769\src\src\gcode\config\M309.cpp.o
                 from Marlin\src\gcode\bedlevel\ubl\../../../feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\gcode\bedlevel\ubl\M421.cpp:32:
c:\users\light\onedrive\documents\github\marlin_q5\marlin\src\core\debug_out.h:62: note: this is the location of the previous definition
   62 |   #define DEBUG_ECHOPGM_P(P)      SERIAL_ECHOPGM_P(P)
      | 
In file included from Marlin\src\gcode\bedlevel\../../feature/bedlevel/ubl/ubl.h:29,
                 from Marlin\src\gcode\bedlevel\../../feature/bedlevel/bedlevel.h:67,
                 from Marlin\src\gcode\bedlevel\M420.cpp:28:
Marlin\src\gcode\bedlevel\../../feature/bedlevel/ubl/../../../core/debug_out.h:62: note: this is the location of the previous definition
   62 |   #define DEBUG_ECHOPGM_P(P)      SERIAL_ECHOPGM_P(P)
      |
In file included from Marlin\src\gcode\bedlevel\ubl\../../../feature/bedlevel/ubl/ubl.h:328,
                 from Marlin\src\gcode\bedlevel\ubl\../../../feature/bedlevel/bedlevel.h:67,
Compiling .pio\build\LPC1769\src\src\gcode\config\M43.cpp.o
                 from Marlin\src\gcode\bedlevel\ubl\G29.cpp:32:
c:\users\light\onedrive\documents\github\marlin_q5\marlin\src\core\debug_out.h:92: warning: "DEBUG_ECHOPGM_P" redefined
   92 |   #define DEBUG_ECHOPGM_P(P)          NOOP
Compiling .pio\build\LPC1769\src\src\gcode\config\M92.cpp.o
```

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-03-07:
trigger is DEBUG_LEVELING_FEATURE

ellensp, 2021-03-07:
your assertion that its in https://github.com/MarlinFirmware/Marlin/pull/21231 is correct

In Marlin/src/core/debug_out.h  is an old #undef DEBUG_PRINT_P 
this should be updated to #undef DEBUG_ECHOPGM_P

lightmaster, 2021-03-07:
Apparently a month ago I enabled DEBUG_LEVELING_FEATURE, however I have no idea why I would have done that nor do I remember doing that... Is there any particular reason to have that enabled?

ellensp, 2021-03-07:
no, for regular use DEBUG_LEVELING_FEATURE should be disabled.
A PR has been created to fix these warnings.

lightmaster, 2021-03-07:
Well, glad I brought it up, and I'll now disable that option, lol

ellensp, 2021-03-07:
PR has been merged. Closing

closed 2021-03-07 by ellensp